bluez/bluez #86: bluez-5.55: undefined references when using slibtool

pacho2, 2021-01-18:
As reported downstream at
https://bugs.gentoo.org/765715
When building with (slibtool - https://git.foss21.org/slibtool ) it fails with undefined references:
```
/usr/lib/gcc/x86_64-pc-linux-gnu/9.3.0/../../../../x86_64-pc-linux-gnu/bin/ld: plugins/.libs/sixaxis_la-sixaxis.o: in function `agent_auth_cb':
sixaxis.c:(.text+0x80d): undefined reference to `g_hash_table_iter_init'
/usr/lib/gcc/x86_64-pc-linux-gnu/9.3.0/../../../../x86_64-pc-linux-gnu/bin/ld: sixaxis.c:(.text+0x828): undefined reference to `g_hash_table_iter_next'
/usr/lib/gcc/x86_64-pc-linux-gnu/9.3.0/../../../../x86_64-pc-linux-gnu/bin/ld: sixaxis.c:(.text+0x889): undefined reference to `g_hash_table_steal'
/usr/lib/gcc/x86_64-pc-linux-gnu/9.3.0/../../../../x86_64-pc-linux-gnu/bin/ld: sixaxis.c:(.text+0x893): undefined reference to `g_malloc0'
/usr/lib/gcc/x86_64-pc-linux-gnu/9.3.0/../../../../x86_64-pc-linux-gnu/bin/ld: sixaxis.c:(.text+0x8a9): undefined reference to `g_idle_add'
/usr/lib/gcc/x86_64-pc-linux-gnu/9.3.0/../../../../x86_64-pc-linux-gnu/bin/ld: sixaxis.c:(.text+0x914): undefined reference to `btd_adapter_get_address'
/usr/lib/gcc/x86_64-pc-linux-gnu/9.3.0/../../../../x86_64-pc-linux-gnu/bin/ld: sixaxis.c:(.text+0x992): undefined reference to `btd_device_set_trusted'
/usr/lib/gcc/x86_64-pc-linux-gnu/9.3.0/../../../../x86_64-pc-linux-gnu/bin/ld: sixaxis.c:(.text+0x99d): undefined reference to `btd_device_set_temporary'
/usr/lib/gcc/x86_64-pc-linux-gnu/9.3.0/../../../../x86_64-pc-linux-gnu/bin/ld: sixaxis.c:(.text+0x9b8): undefined reference to `ba2str'
/usr/lib/gcc/x86_64-pc-linux-gnu/9.3.0/../../../../x86_64-pc-linux-gnu/bin/ld: sixaxis.c:(.text+0x9ca): undefined reference to `ba2str'
/usr/lib/gcc/x86_64-pc-linux-gnu/9.3.0/../../../../x86_64-pc-linux-gnu/bin/ld: sixaxis.c:(.text+0x9d7): undefined reference to `ba2str'
/usr/lib/gcc/x86_64-pc-linux-gnu/9.3.0/../../../../x86_64-pc-linux-gnu/bin/ld: sixaxis.c:(.text+0xa11): undefined reference to `btd_debug'
/usr/lib/gcc/x86_64-pc-linux-gnu/9.3.0/../../../../x86_64-pc-linux-gnu/bin/ld: sixaxis.c:(.text+0xa3d): undefined reference to `btd_debug'
/usr/lib/gcc/x86_64-pc-linux-gnu/9.3.0/../../../../x86_64-pc-linux-gnu/bin/ld: sixaxis.c:(.text+0xa81): undefined reference to `btd_error'
/usr/lib/gcc/x86_64-pc-linux-gnu/9.3.0/../../../../x86_64-pc-linux-gnu/bin/ld: sixaxis.c:(.text+0xafa): undefined reference to `baswap'
/usr/lib/gcc/x86_64-pc-linux-gnu/9.3.0/../../../../x86_64-pc-linux-gnu/bin/ld: sixaxis.c:(.text+0xb1b): undefined reference to `btd_device_set_record'
/usr/lib/gcc/x86_64-pc-linux-gnu/9.3.0/../../../../x86_64-pc-linux-gnu/bin/ld: sixaxis.c:(.text+0xb54): undefined reference to `btd_error'
/usr/lib/gcc/x86_64-pc-linux-gnu/9.3.0/../../../../x86_64-pc-linux-gnu/bin/ld: sixaxis.c:(.text+0xb89): undefined reference to `baswap'
/usr/lib/gcc/x86_64-pc-linux-gnu/9.3.0/../../../../x86_64-pc-linux-gnu/bin/ld: sixaxis.c:(.text+0xbd6): undefined reference to `btd_error'
/usr/lib/gcc/x86_64-pc-linux-gnu/9.3.0/../../../../x86_64-pc-linux-gnu/bin/ld: sixaxis.c:(.text+0xc10): undefined reference to `btd_error'
/usr/lib/gcc/x86_64-pc-linux-gnu/9.3.0/../../../../x86_64-pc-linux-gnu/bin/ld: plugins/.libs/sixaxis_la-sixaxis.o: in function `auth_closure_destroy':
sixaxis.c:(.text+0x30): undefined reference to `g_free'
collect2: error: ld returned 1 exit status
rdlibtool: exec error upon slbt_exec_link_create_library(), line 1446: (see child process error messages).
rdlibtool: < returned to > slbt_exec_link(), line 1843.
make[1]: *** [Makefile:4505: plugins/sixaxis.la] Error 2
make[1]: *** Waiting for unfinished jobs....
rdlibtool  --tag=CC   --mode=link x86_64-pc-linux-gnu-gcc  -fPIC   -O2 -pipe -pie -Wl,-z,now -Wl,-O1 -Wl,--as-needed -o tools/hex2hcd tools/hex2hcd.o  
rdlibtool: link: ln -s hid2hci tools/hid2hci.exe.wrapper

rdlibtool --tag=CC --mode=link x86_64-pc-linux-gnu-gcc -fPIC -O2 -pipe -pie -Wl,-z,now -Wl,-O1 -Wl,--as-needed -o tools/hex2hcd tools/hex2hcd.o

rdlibtool: lconf: {.name="libtool"}.
rdlibtool: fdcwd: {.fdcwd=AT_FDCWD, .realpath="/var/tmp/portage/net-wireless/bluez-5.55/work/bluez-5.55-abi_x86_64.amd64"}.
rdlibtool: lconf: fstatat(AT_FDCWD,".",...) = 0 {.st_dev = 65026, .st_ino = 10770856}.
rdlibtool: lconf: openat(AT_FDCWD,"libtool",O_RDONLY,0) = 3.
rdlibtool: lconf: found "/var/tmp/portage/net-wireless/bluez-5.55/work/bluez-5.55-abi_x86_64.amd64/libtool".
rdlibtool: link: x86_64-pc-linux-gnu-gcc tools/hex2hcd.o -fPIC -O2 -pipe -pie -Wl,-z,now -Wl,-O1 -Wl,--as-needed -o tools/.libs/hex2hcd
rdlibtool: link: ln -s bluemoon tools/bluemoon.exe.wrapper
rdlibtool: link: ln -s hex2hcd tools/hex2hcd.exe.wrapper
```
Reporter thinks that this is because in Makefile.plugins plugins_sixaxis_la_LDFLAGS contains -no-undefined which GNU libtool silently ignores and the suggestion is to simply drop the offending flag with:
```diff --git a/Makefile.plugins b/Makefile.plugins
index 98eee9ec8..0ce51f120 100644
--- a/Makefile.plugins
+++ b/Makefile.plugins
@@ -107,8 +107,7 @@ builtin_sources += profiles/battery/battery.c
 if SIXAXIS
 plugin_LTLIBRARIES += plugins/sixaxis.la
 plugins_sixaxis_la_SOURCES = plugins/sixaxis.c
-plugins_sixaxis_la_LDFLAGS = $(AM_LDFLAGS) -module -avoid-version \
-						-no-undefined
+plugins_sixaxis_la_LDFLAGS = $(AM_LDFLAGS) -module -avoid-version
 plugins_sixaxis_la_LIBADD = $(UDEV_LIBS)
 plugins_sixaxis_la_CFLAGS = $(AM_CFLAGS) -fvisibility=hidden
 endif
```

Thanks a lot

Vudentz, 2021-01-19:
It seems sixaxis is indeed the only plugin to be using such flags, @sjanc do you have any idea why we needed that in the first place?

sjanc, 2021-01-26:
I think at time of adding sixaxis support (2013,) it was needed due to sixaxis being external plugin, but if it is no longer required (ie sixaxis plugin is properly linked and loaded without that option) and fix the issue we should just remove it

closed 2021-01-27 by Vudentz