bluez/bluez #88: not connecting to android after reboot of raspberry

dr-ni, 2021-01-22:
Dear all,
I've found some strange behavior of bluez on raspberry pi zero w (raspbian buster) when connecting  to a paired android headset/hfp device (all done in bluetoothctl).
- Connecting will not be working anymore after reboot.
- Connecting to an ios headset/hfp device is still working correct after reboot.
- I have found a possible workaround:
When comparing /var/lib/bluetooth/<raspi-mac>/<android-mac>/info and /var/lib/bluetooth/<raspi-mac>/<ios-mac>/info I have found an additional key section in the android info file which is not present in the ios info file:

[SlaveLongTermKey]
Key=9BC6821CB1AE51995F237A7E15474032
Authenticated=2
EncSize=16
EDiv=0
Rand=0

If this section is removed and the pi is rebooted again, all is working correct

Is this a bug?

$ uname -a
Linux pilot1 5.4.83+ #1379 Mon Dec 14 13:06:05 GMT 2020 armv6l GNU/Linux

$ sudo apt list bluez 
Auflistung... Fertig
bluez/testing,now 5.50-1.2~deb10u1+rpt2 armhf  [Installiert,automatisch]

Vudentz, 2021-01-25:
Please attach the HCI traces, you can collect them using btmon.

dr-ni, 2021-01-26:
[btlogandroidpair.txt](https://github.com/bluez/bluez/files/5873120/btlogandroidpair.txt)
[btlogiospair.txt](https://github.com/bluez/bluez/files/5873121/btlogiospair.txt)

dr-ni, 2021-01-26:
[btlogandroidconnect.txt](https://github.com/bluez/bluez/files/5873128/btlogandroidconnect.txt)
[btlogiosconnect.txt](https://github.com/bluez/bluez/files/5873129/btlogiosconnect.txt)

dr-ni, 2021-01-28:
do you need some more information?

dr-ni, 2021-01-30:
after receiving a new version of bluez-firmware the problem seems now have been gone:

Start-Date: 2021-01-29  14:54:29
Commandline: apt upgrade
Requested-By: pi (1000)
Upgrade: rpi-eeprom:armhf (11.5-1, 11.6-1), sudo:armhf (1.8.27-1+deb10u2, 1.8.27-1+deb10u3), bluez-firmware:armhf (1.2-4+rpt7, 1.2-4+rpt8),$
End-Date: 2021-01-29  14:57:14

dr-ni, 2021-02-07:
don't know why but again same problems

closed 2025-08-30 by github-actions[bot]