bluez/bluez #89: Device connection reference counting

dlech, 2021-01-23:
I came across https://crbug.com/630586 today where 5 years ago @Vudentz made some comments about adding device connection reference counting or possibly adding an `org.bluez.Service1` API to handle things on a service level rather than on the device level.

Has there been any progress or discussion on this topic since then? Or any change of opinion on how the issue is best solved?

In addition to the problems mentioned in the link, one problem I think this could solve nicely is devices not disconnecting when the app (dbus client) using the device crashes.

Vudentz, 2021-01-25:
Interesting, I recall my comments regarding this but the Chrome OS folks never mentioned this problem still exits, perhaps @alainmic can comment on it.

This could also be fixed in the browser afaik, since it can implement a ref count if it had started the connection, or better yet have a dialog to disconnect when leaving the webpage that had the device connected.

dlech, 2021-01-26:
From what I read in the Chromium issues was that there were cases where the OS was using a device as a mouse and the browser was connected to it too. So even if there was reference counting in the browser, it would still disconnect the mouse when the browser was closed.

I also contribute to [Bleak](https://github.com/hbldh/bleak/) which is a cross-platform Python library for BLE. One issue that many users face is that if their Python script crashes, the BLE device remains connected. Other OSes will disconnect the device if the Python script crashes. So this is what interests me in the reference counting idea. If BlueZ had a reference count that was decremented when a D-Bus client disappeared (i.e. when the Python script crashes) then it could behave much more consistent with other OSes.

alainmic, 2021-01-26:
+1 on this and this is still a problem for Chrome with the browser and
ARC++ Apis today.  The mitigations put in place today simply will avoid
disconnecting if the device is paired which is pretty weak but the best we
got without lower level reference counting.

Vudentz, 2021-01-26:
Ok, a ref count sounds like a good idea then, the only problem I see is how to maintain the connection in case it was triggered by the system settings and that exits, so we probably need to maintain 2 differen APIs. We could perhaps take advantage of the upcoming bearer specific APIs to introduce ref counting that way we don't break existing APIs that do leave the device connected when exiting.

Bunagana, 2021-03-05:
@Vudentz, curious if the bearer specific APIs mentioned in your last comment have been released in 5.56. We have also run into the issue mentioned by David in the first comment and would really appreciate if you could let us know of any info regarding the addition of a ref counting. Thanks.

Bunagana, 2021-06-17:
Hi @Vudentz, any update on this issue would be very much appreciated. Thank you.

closed 2025-10-06 by github-actions[bot]