bluez/bluez #90: [Bluez 5.55-3] Debian bluetoothd crashes when a connection is established

mrcnpp, 2021-01-24:
Hi i'm using  Bluez 5.55-3 on Debian testing/sid .
Whenever i connect my headphones to the pc they connects to pc for 1-2 seconds then bluethootd crashes
systemctl status blutooth output:
```
● bluetooth.service - Bluetooth service
Loaded: loaded (/lib/systemd/system/bluetooth.service; enabled; vendor preset: enabled)
Active: failed (Result: signal) since Sun 2021-01-24 18:57:44 CET; 22min ago
Docs: man:bluetoothd
Process: 4820 ExecStart=/usr/local/libexec/bluetooth/bluetoothd -d (code=killed, signal=ABRT)
Main PID: 4820 (code=killed, signal=ABRT)        CPU: 37ms

```
juornalctl -fu bluetooth output:
```
-- Journal begins at Sat 2021-01-23 12:08:03 CET. --
Jan 24 18:57:44 Marco-PC bluetoothd[4820]: profiles/audio/sink.c:sink_connect() stream creation in progress
Jan 24 18:57:44 Marco-PC bluetoothd[4820]: src/service.c:change_state() 0x55ad43f35d20: device 4C:87:5D:55:F2:25 profile a2dp-sink state changed: disconnected -> connecting (0)
Jan 24 18:57:44 Marco-PC bluetoothd[4820]: src/service.c:btd_service_ref() 0x55ad43f48ec0: ref=3
Jan 24 18:57:44 Marco-PC bluetoothd[4820]: plugins/policy.c:service_cb() Added Headset Voice gateway reconnect 0
Jan 24 18:57:44 Marco-PC bluetoothd[4820]: profiles/audio/avdtp.c:avdtp_connect_cb() AVDTP: connected signaling channel to 4C:87:5D:55:F2:25
Jan 24 18:57:44 Marco-PC bluetoothd[4820]: profiles/audio/avdtp.c:avdtp_connect_cb() AVDTP imtu=672, omtu=895
Jan 24 18:57:44 Marco-PC bluetoothd[4820]: free(): double free detected in tcache 2
Jan 24 18:57:44 Marco-PC bluetoothd[4820]: Unable to load Endpoint: seid 3
Jan 24 18:57:44 Marco-PC systemd[1]: bluetooth.service: Main process exited, code=killed, status=6/ABRT
Jan 24 18:57:44 Marco-PC systemd[1]: bluetooth.service: Failed with result 'signal'.

```

Vudentz, 2021-01-26:
This is likely a duplicated of:

https://github.com/bluez/bluez/issues/57

mrcnpp, 2021-01-26:
Ok , thanks 
So it's likely to be fixed in the next release? If so when the new release will be out ?

mrcnpp, 2021-01-28:
Using the master branch of the repo it works

closed 2021-01-28 by mrcnpp