bluez/bluez #93: Re-pairing of a already known Android mobile phone fails on Raspberry PI

mibcat, 2021-02-06:
Hi !

I'm having trouble to pair an already known device again.

If the pairing process with an Android mobile phone is done for the very first time everything is working fine. But if the bluetooth connection stored on the mobile phone is deleted and the pairing process is started again I get these errors:

```bash
sudo systemctl status bluetooth.service 
● bluetooth.service - Bluetooth service
   Loaded: loaded (/lib/systemd/system/bluetooth.service; enabled; vendor preset: enabled)
   Active: active (running) since Sat 2021-02-06 12:19:31 CET; 40min ago
     Docs: man:bluetoothd(8)
  Process: 687 ExecStartPost=/usr/bin/sdptool add SP (code=exited, status=0/SUCCESS)
 Main PID: 684 (bluetoothd)
   Status: "Running"
    Tasks: 1 (limit: 3819)
   CGroup: /system.slice/bluetooth.service
           └─684 /usr/lib/bluetooth/bluetoothd -C --noplugin=sap

Feb 06 12:19:31 dc00014 systemd[1]: Starting Bluetooth service...
Feb 06 12:19:31 dc00014 bluetoothd[684]: Bluetooth daemon 5.50
Feb 06 12:19:31 dc00014 bluetoothd[684]: Starting SDP server
Feb 06 12:19:31 dc00014 bluetoothd[684]: Excluding (cli) sap
Feb 06 12:19:31 dc00014 bluetoothd[684]: Bluetooth management interface 1.18 initialized
Feb 06 12:19:31 dc00014 sdptool[687]: Serial Port service registered
Feb 06 12:19:31 dc00014 systemd[1]: Started Bluetooth service.
Feb 06 12:19:31 dc00014 bluetoothd[684]: Failed to set privacy: Rejected (0x0b)
Feb 06 12:25:19 dc00014 bluetoothd[684]: No agent available for request type 2
Feb 06 12:25:19 dc00014 bluetoothd[684]: device_confirm_passkey: Operation not permitted
```

After some investigation I've found out two workarounds (but both of them are not handy)

1) delete the stored Android device on the Raspberry e.g. with
`bt-device -r <device name>`

2) or start an agent manually (control via bluetoothctrl does **not** work)
`bt-agent --capability=NoInputNoOutput`
(and during the pairing process "yes" must be entered)

But both of the solution would need an user interaction on the Raspberry which is therefore not an option.

Don't know if it is a matter of the last Raspberry OS update (apt upgrade) but this issue was not present in December 2020 before a system update was done a few days ago.

Curiously the bluez version did not changed and is still the same (5.50) - so what could be gone wrong here ?

```bash
apt list --installed |grep blue

WARNING: apt does not have a stable CLI interface. Use with caution in scripts.

bluez-firmware/testing,now 1.2-4+rpt8 all [installed]
bluez-obexd/testing,now 5.50-1.2~deb10u1+rpt2 armhf [installed,automatic]
bluez-tools/stable,now 2.0~20170911.0.7cb788c-2 armhf [installed]
bluez/testing,now 5.50-1.2~deb10u1+rpt2 armhf [installed]
libbluetooth3/testing,now 5.50-1.2~deb10u1+rpt2 armhf [installed,automatic]
lxplug-bluetooth/testing,now 0.19 armhf [installed,automatic]
pi-bluetooth/testing,now 0.1.15 all [installed]
python3-bluez/stable,now 0.22+really0.22-1 armhf [installed]
```

```bash
uname -a
Linux dc00014 5.10.11-v7l+ #1399 SMP Thu Jan 28 12:09:48 GMT 2021 armv7l GNU/Linux
```

```bash
hciconfig -a
hci0:	Type: Primary  Bus: UART
	BD Address: DC:A6:32:C6:5A:85  ACL MTU: 1021:8  SCO MTU: 64:1
	UP RUNNING PSCAN 
	RX bytes:10630 acl:157 sco:0 events:426 errors:0
	TX bytes:256534 acl:383 sco:0 commands:133 errors:0
	Features: 0xbf 0xfe 0xcf 0xfe 0xdb 0xff 0x7b 0x87
	Packet type: DM1 DM3 DM5 DH1 DH3 DH5 HV1 HV2 HV3 
	Link policy: RSWITCH SNIFF 
	Link mode: SLAVE ACCEPT 
	Name: 'dc00014'
	Class: 0x000000
	Service Classes: Unspecified
	Device Class: Miscellaneous, 
	HCI Version: 5.0 (0x9)  Revision: 0x156
	LMP Version: 5.0 (0x9)  Subversion: 0x6119
	Manufacturer: Cypress Semiconductor Corporation (305)
```

mkarhumaa, 2021-02-06:
Hi @mibcat ,

I just moments ago sent kernel patch to linux-bluetooth mailing list. It should fix Just-Works re-pairing problem. Any chance you could test if it solves your problem?

https://marc.info/?l=linux-bluetooth&m=161262093311177&w=2

mibcat, 2021-02-06:
Hi Matias,

thanks for your feedback. If you provide a guide of how to test it I'll try it. 
Hopefully this not beyond my capabilities - never build the kernel so far ...

mkarhumaa, 2021-02-06:
Compiling kernel requires some work. Here are instructions for building Raspberry Pi kernel: https://www.raspberrypi.org/documentation/linux/kernel/building.md

Make sure to include this patch when compiling: https://marc.info/?l=linux-bluetooth&m=161262095811187&w=2

If you want to be sure that you are running into exactly same problem than what my patch fixes, you could first take HCI capture of failing pairing attempt and post it here. You can use `btmon` for taking HCI capture.

mkarhumaa, 2021-02-06:
@mibcat Check also this issue, might be related: https://github.com/bluez/bluez/issues/88

mibcat, 2021-02-06:
I've attached the output of btmon ...

[bt_pairing_first_time.txt](https://github.com/bluez/bluez/files/5937181/bt_pairing_first_time.txt)
[bt_pairing_second_time.txt](https://github.com/bluez/bluez/files/5937182/bt_pairing_second_time.txt)

this screenshot shows the difference during the passkey request
left side: first time
right side: second time

[image]

So I wait for you first review before starting to build the kernel ;-)

mkarhumaa, 2021-02-06:
Unfortunately this looks like different case and my patch won't help here.

mibcat, 2021-02-06:
ok - thanks for your feedback - I'll have a look on the issue #88 you mentioned above ...

Vudentz, 2021-02-18:
This is the result of:

https://git.kernel.org/pub/scm/linux/kernel/git/bluetooth/bluetooth-next.git/commit/?id=ffee202a78c2980688bc5d2f7d56480e69a5e0c9

So one can no longer repair without the use of an agent, @sonnysasaka Im just not sure why it works for the first time though? Should it always be requesting user confirmation?

Vudentz, 2021-02-18:
@mibcat If you want to just works repairing to work without an agent you need to set JustWorksRepairing = always:

https://git.kernel.org/pub/scm/bluetooth/bluez.git/tree/src/main.conf#n78

closed 2021-02-18 by Vudentz

mibcat, 2021-02-22:
@Vudentz Thanks for your feedback.

On the raspi I had to build bluez from source because the mentioned configuration parameter **JustWorksRepairing** is not available for the Debian installed version **5.50**.

Therefore I built the latest version **5.55** and the re-pairing seems to work - thanks again !!

pierre-pretorius, 2021-03-15:
I tried using bluez 5.54 from buster-backports but got errors all over. I then reverted the RPI kernel which solved the problem wasn't a good option.

I then realized the simplest solution is to spam the "y" to the command:
```
yes | bt-agent -c NoInputNoOutput
```

If you need to use it in a systemd service, you can use the example below. I noticed bt-agent has to be stopped with SIGINT:
```
[Unit]
Description=Bluetooth Agent

[Service]
ExecStart=/bin/sh -c '/usr/bin/yes | /usr/bin/bt-agent -c NoInputNoOutput'
ExecStop=/bin/kill -s SIGINT $MAINPID
Type=simple

[Install]
WantedBy=multi-user.target
```

skast96, 2021-11-10:
This is still a issue does anybody have a good solution for this ?

CabralPro, 2022-03-15:
> @mibcat If you want to just works repairing to work without an agent you need to set JustWorksRepairing = always:
> 
> https://git.kernel.org/pub/scm/bluetooth/bluez.git/tree/src/main.conf#n78

It work for me, thanks

JJK96, 2025-07-30:
> I tried using bluez 5.54 from buster-backports but got errors all over. I then reverted the RPI kernel which solved the problem wasn't a good option.
>
> I then realized the simplest solution is to [...]

I had to use this, but with 'DisplayOnly' instead.