bluez/bluez #96: bluez-5.54-2 regression

NullSense, 2021-02-13:
Hi, I noticed releases later than bluez-5.54-1 are affected by a bug (I didn't look at which commit specifically), which doesn't let me use bluetooth no my TP-Link Archer TX50E AX3000 wifi + bluetooth card (ax200 chipset).

https://www.reddit.com/r/linux4noobs/comments/lhx6zz/tplink_archer_tx50e_ax3000_wifi_bluetooth_card/ there are some logs here.

njdom24, 2021-02-15:
I have a somewhat related issue. Same headphones, same WiFi/Bluetooth adapter. Headphones work fine with a cheap USB adapter, and had been working since I got this card, but it suddenly keeps dropping connections. In my case, downgrading BlueZ to the version you listed does not help.

Vudentz, 2021-02-16:
@NullSense Can you try to reproduce with master? Also the daemon logs and btmont traces would help to nail down what is exactly is the problem.

NullSense, 2021-02-19:
@Vudentz I can try once I have a bit more time, currently swamped...

closed 2025-08-29 by github-actions[bot]