bluez/bluez #99: adapter: Toggling boolean properties is not always correctly handled

BenKaufmann, 2021-03-04:
Toggling (boolean) adapter properties like `Discoverable` fails to work correctly if a request is issued while an opposing request is still in progress resulting in a situation where the final value is not the expected value.

This is most easily reproducible with subsequent asynchronous dbus calls but it can also (often) be reproduced with the following simple script:
```
#! /bin/bash
# start in state A
busctl set-property org.bluez /org/bluez/hci0 org.bluez.Adapter1 Discoverable b false
busctl get-property org.bluez /org/bluez/hci0 org.bluez.Adapter1 Discoverable

# request change to state B
busctl set-property org.bluez /org/bluez/hci0 org.bluez.Adapter1 Discoverable b true &
# and immediately toggle back to state A
busctl set-property org.bluez /org/bluez/hci0 org.bluez.Adapter1 Discoverable b false

sleep 2
# should now be in state A but is in B
busctl get-property org.bluez /org/bluez/hci0 org.bluez.Adapter1 Discoverable
```
The problem is in https://github.com/bluez/bluez/blob/ad05009dd992dee229634e033a007f002a6da4d5/src/adapter.c#L2888
because when the second request (back to A) enters, while the first (to B) is still in flight, the original value is still the old value (i.e. A). Furthermore, `pending_settings` is oblivious to the pending value and therefore doesn't realize that the new request should be either passed on, queued, or rejected. Instead, the call returns successfully but the property value is never changed accordingly. 

[The attached patch fixes the issue for me](https://github.com/bluez/bluez/files/6086472/0001-Adapter-Track-pending-settings.txt). 

As a side note: The switch in https://github.com/bluez/bluez/blob/ad05009dd992dee229634e033a007f002a6da4d5/src/adapter.c#L634 looks wrong to me as it should probably switch over `opcode` instead of `mode`.

Vudentz, 2021-03-04:
It is probably easier to just return Busy otherwise if there are multiple clients this becomes way too complicated.

BenKaufmann, 2021-03-05:
@Vudentz Thank you for looking into this.

Out of curiosity: could you please elaborate on why the code would become more complicated with multiple clients and why it is important to do the busy check on entry to `property_set_mode()`? If I understand it correctly, the lower level code already fails with MGMT_STATUS_BUSY if it is really busy. This error would then be propagated up to `property_set_mode_complete()`, where it could be mapped back to `g_dbus_pending_property_error(id, ERROR_INTERFACE ".Busy", NULL);`

BenKaufmann, 2021-03-05:
Regarding https://github.com/BluezTestBot/bluez/commit/ea299ab3763f2f0cd6684c10146d5a11b7c2128a
It's probably unlikely but it looks like this fix could get "stuck" on error.

Assume `property_set_mode()` is called with some setting, e.g. MGMT_SETTING_BONDABLE, and the corresponding setting bit is newly set in `pending_settings`. Further assume that the following call to `mgmt_send()` fails (either directly or indirectly later). In that case, `settings_changed()` is not called and therefore `pending_settings` is not updated. Any subsequent call to `property_set_mode()` with MGMT_SETTING_BONDABLE will therefore now immediately fail with "Busy".

BenKaufmann, 2021-03-05:
I simplified and cleaned up [my patch](https://github.com/bluez/bluez/files/6090950/0001-Adapter-Track-pending-settings.txt) a little so that it should now apply cleanly against master. If you are interested, I could open a PR.

Vudentz, 2021-03-06:
The problem is this assumption:

```
# request change to state B
busctl set-property org.bluez /org/bluez/hci0 org.bluez.Adapter1 Discoverable b true &
# and immediately toggle back to state A
busctl set-property org.bluez /org/bluez/hci0 org.bluez.Adapter1 Discoverable b false
```

If there are another application doing the same the whole thing becomes unpredictable since app 1 may expect it to be false but in the meantime app 2 set it to true so this will alway be time sensetive and queuing doesn't really solve anything.

closed 2021-03-06 by Vudentz