bluez/bluez #101: PnP ID Device Information Gatt Server

inonoob, 2021-03-05:
Hey, 

when I use the default installation for a raspberry pi with Bluez 5.50 and start the Gatt python server, I only have the following default services: 

- Generic Access
- Generic Attribute 
- (My custom services) 

But now if I try to compile from source a newer version for Bluez, I get the following: 

- Generic Access
- Generic Attribute
- Device Information ==> PnP ID 

the problem here is if with the Gatt python server, I define my own Device Information. I have the following: 

- Generic Access
- Generic Attribute
- Device Information ==> PnP ID 
- Device Information 

Question How can I set to not use PnP ID as it is for bluez 5.50 on the default installation of raspbian? Or how can I add it to Device Information without creating a second on with the Gatt python server ? 

Best Regards

ryanteck, 2021-03-05:
This issue is also affecting me causing issues as well.

closed 2021-03-06 by Vudentz

dhruvelectro, 2023-05-23:
I have found same issue for our ble peripheral code. I have done following:
`sudo nano /etc/bluetooth/main.conf`

Added following line (Enabled the DeviceID and set) followings
`DeviceID = false`

restarted the blutooth deamon
```
sudo systemctl daemon-reload
sudo systemctl restart bluetooth
sudo systemctl status bluetooth
```

then if I re-run my BLE Peripheral python code

rpm-projects, 2023-12-21:
> I have found same issue for our ble peripheral code. I have done following: `sudo nano /etc/bluetooth/main.conf`
> 
> Added following line (Enabled the DeviceID and set) followings `DeviceID = false`
> 
> restarted the blutooth deamon
> 
> ```
> sudo systemctl daemon-reload
> sudo systemctl restart bluetooth
> sudo systemctl status bluetooth
> ```
> 
> then if I re-run my BLE Peripheral python code it worked like we expected and there in no other Device Information services and PnP ID Characteristic.

Hello, do you know by chance how to disable **Generic Access** and **Generic Attribute** ?
Thank you

Vudentz, 2023-12-21:
> Hello, do you know by chance how to disable **Generic Access** and **Generic Attribute** ? Thank you

This will probably break qualification if we allow upper layer to overwrite these services since they are mandatory and exposes details of GATT/ATT implementation, we could expose some new API so application can register their own values, well in case they cannot be entered via main.conf, etc.

lingxd, 2025-07-31:
> > I have found same issue for our ble peripheral code. I have done following: `sudo nano /etc/bluetooth/main.conf`我发现我们的 BLE 外设代码也存在同样的问题。我已采取以下措施： `sudo nano /etc/bluetooth/main.conf` 
> > Added following line (Enabled the DeviceID and set) followings `DeviceID = false`添加以下行（启用 DeviceID 并设置）以下 `DeviceID = false`
> > restarted the blutooth deamon重启蓝牙守护进程
> > ```
> > sudo systemctl daemon-reload
> > sudo systemctl restart bluetooth
> > sudo systemctl status bluetooth
> > ```
> > 
> > 
> >     
> >       
> >     
> > 
> >       
> >     
> > 
> >     
> >   
> > then if I re-run my BLE Peripheral python code it worked like we expected and there in no other Device Information services and PnP ID Characteristic.然后，如果我重新运行我的 BLE 外设 python 代码，它会像我们预期的那样工作，并且没有其他设备信息服务和 PnP ID 特性。
> 
> Hello, do you know by chance how to disable **Generic Access** and **Generic Attribute** ?您好，您知道如何禁用**通用访问**和**通用属性**吗？ Thank you  谢谢

I want to disable **Generic Access** and **Generic Attribute** , now are you find ways?