bluez/bluez #102: Not working right after updating from 5.55 to 5.56. Immediate disconnects and forced HSP mode

JaneSmith, 2021-03-07:
I'm running Fedora Silverblue 33. Recently, it was updated so that the `bluez`, `bluez-cups`, `bluez-libs`, and `bluez-obexd` packages changed from versions `5.55-1.fc33` to `5.56-1.fc33`. Ever since then, my Bluetooth experience has been very bad and frustrating.

With 5.56, my Mpow Flame Pro IEMs "true wireless earbuds" immediately disconnect after connecting. I'll power them on, hear "connected" and then immediately after "disconnected". This happens consistently, every single time. The only way I can successfully connect it is if I remove the pairing, put the Mpow Flame Pro into pairing mode, and re-pair. After re-pairing, it immediately connects and stays connected. This last until I turn the Mpow Flame Pro off.

With 5.56, my Mpow Cheetah headset connects as expected when powered on. However, after a while (maybe an hour), the connection drops out. A few seconds later it comes back, but is locked into HSP mode with very poor audio quality. I am forced to turn the Mpow Cheetah off and back on again, to reconnect in the proper mode.

With 5.55, neither of the above problems occur. My Mpow Flame Pro connects perfectly fine when it's turned on. My Mpow Cheetah stays connected and in the correct mode for the entire session.

As a result, I've been forced to roll back to Fedora Silverblue 33.20210227.0 from the 27th of February, where `bluez` is still at version 5.55 and still working properly.

Let me know if there's any more information I can provide to help resolve this.

inonoob, 2021-03-07:
Hey, 

I'm not a dev of bluez but I think it would help if you could use ```btmon``` to check what is happening during the session. And maybe even post the dump. This would help to see the Bluetooth is seeing and doing.

casta, 2021-03-09:
I think I get the same issue: after updating to 5.56 I only get the HSP profile. A2DP profile is not available anymore.

I get this in logs that seems relevant:
```
mars 09 13:14:47 bespin bluetoothd[377]: profiles/audio/a2dp.c:a2dp_select_capabilities() Unable to select SEP
```

Everything else is the same as with 5.55. And downgrading to 5.55 fixes the issue

The relevant btmon part should be in there:
```
> HCI Event: Role Change (0x12) plen 8                                                                                                                                                                                   #190 [hci0] 13.235506 
        Status: Success (0x00)                                                                                                                                                                                                                 
        Address: 5C:C6:E9:B6:94:F3 (Edifier International)                                                                                                                                                                                     
        Role: Slave (0x01)                                                                                                                                                                                                                     
> ACL Data RX: Handle 256 flags 0x02 dlen 30                                                                                                                                                                             #191 [hci0] 13.247567 
      Channel: 64 len 26 [PSM 3 mode Basic (0x00)] {chan 0}                                                                                                                                                                                    
      RFCOMM: Unnumbered Info with Header Check (UIH) (0xef)                                                                                                                                                                                   
         Address: 0x6b cr 1 dlci 0x1a                                                                                                                                                                                                          
         Control: 0xef poll/final 0                                                                                                                                                                                                            
         Length: 22                                                                                                                                                                                                                            
         FCS: 0xe4                                                                                                                                                                                                                             
        41 54 2b 49 50 48 4f 4e 45 41 43 43 45 56 3d 31  AT+IPHONEACCEV=1                                                                                                                                                                      
        2c 31 2c 31 30 0d e4                             ,1,10..                                                                                                                                                                               
< ACL Data TX: Handle 256 flags 0x00 dlen 17                                                                                                                                                                             #192 [hci0] 13.247871 
      Channel: 708 len 13 [PSM 3 mode Basic (0x00)] {chan 0}                                                                                                                                                                                   
      RFCOMM: Unnumbered Info with Header Check (UIH) (0xef)                                                                                                                                                                                   
         Address: 0x69 cr 0 dlci 0x1a                                                                                                                                                                                                          
         Control: 0xef poll/final 0                                                                                                                                                                                                            
         Length: 9                                                                                                                                                                                                                             
         FCS: 0x3e                                                                                                                                                                                                                             
        0d 0a 45 52 52 4f 52 0d 0a 3e                    ..ERROR..>                                                                                                                                                                            
[... cut ...]
= bluetoothd: profiles/audio/a2dp.c:a2dp_select_capabilities() Unable to select SEP                                                                                                                                                  14.095704
< ACL Data TX: Handle 256 flags 0x00 dlen 12                                                                                                                                                                             #196 [hci0] 15.093710
      L2CAP: Disconnection Request (0x06) ident 10 len 4
        Destination CID: 769
        Source CID: 65
```

wunderf1tz, 2021-03-09:
Same here, I can provide the syslog if it helps:
```
bluetoothd[3956]: profiles/audio/avdtp.c:handle_unanswered_req() No reply to Suspend request
bluetoothd[3956]: src/profile.c:ext_io_disconnected() Unable to get io data for Headset Voice gateway: getpeername: Transport endpoint is not connected (107)
bluetoothd[3956]: Device is already marked as connected
bluetoothd[3956]: src/service.c:btd_service_connect() a2dp-sink profile connect failed for 38:18:4C:96:74:03: Device or resource busy
bluetoothd[3956]: /org/bluez/hci0/dev_38_18_4C_96_74_03/fd5: fd(63) ready
```

Fr0stykiller, 2021-03-09:
I can confirm similar behavior on my side:
```
bluetoothd[49831]: src/profile.c:ext_io_disconnected() Unable to get io data for Headset Voice gateway: getpeername: Transport endpoint is not connected (107)
src/profile.c:record_cb() Unable to get Headset Voice gateway SDP record: Host is down
bluetoothd[49831]: profiles/audio/a2dp.c:a2dp_select_capabilities() Unable to select SEP
bluetoothd[49831]: src/service.c:btd_service_connect() a2dp-sink profile connect failed for xx:xx:xx:xx:xx: Device or resource busy
bluetoothd[49831]: plugins/policy.c:reconnect_timeout() Reconnecting services failed: Device or resource busy (16)```

albertoantunes, 2021-03-09:
Hi, I've recently updated my system and noticed this behaviour as well. When playing music through Spotify/Youtube (tried both the desktop app and web app in chromium) i get a disconnection about every 10 minutes, and then it quickly reconnects. 

I don't get however the behavior @casta mentioned, I still get HSP, A2DP profiles everytime after it reconnects.

Sony WH-XM4 headphones

Here are the logs of btmon when the issue happend (disconnected at around line 181)

[btmon.log](https://github.com/bluez/bluez/files/6110820/btmon.log)

I tried replacing pulseaudio with pipewire, but the issue still happens the same way, so I tought it might be bluez related since as far as I understand (not very much sorry) pipewire also uses pipewire-pulse which in turn uses bluez right?

I will try downgrading to a prior bluez version later to see if it fixes the issue

Vudentz, 2021-03-11:
Run with bluetooth -dn:

```
bluetoothd[916489]: profiles/audio/avdtp.c:avdtp_register_remote_sep() seid 1 type 1 media 0 delay_reporting true
bluetoothd[916489]: profiles/audio/a2dp.c:register_remote_sep() Found remote SEP: /org/bluez/hci0/dev_CC_98_8B_49_51_EE/sep1
bluetoothd[916489]: profiles/audio/avdtp.c:avdtp_register_remote_sep() seid 6 type 1 media 0 delay_reporting true
bluetoothd[916489]: profiles/audio/a2dp.c:register_remote_sep() Found remote SEP: /org/bluez/hci0/dev_CC_98_8B_49_51_EE/sep6
bluetoothd[916489]: profiles/audio/avdtp.c:avdtp_register_remote_sep() seid 3 type 1 media 0 delay_reporting true
bluetoothd[916489]: profiles/audio/a2dp.c:register_remote_sep() Found remote SEP: /org/bluez/hci0/dev_CC_98_8B_49_51_EE/sep3
bluetoothd[916489]: profiles/audio/avdtp.c:avdtp_register_remote_sep() seid 2 type 1 media 0 delay_reporting true
bluetoothd[916489]: profiles/audio/a2dp.c:register_remote_sep() Found remote SEP: /org/bluez/hci0/dev_CC_98_8B_49_51_EE/sep2
bluetoothd[916489]: profiles/audio/avdtp.c:session_cb() 
bluetoothd[916489]: profiles/audio/avdtp.c:avdtp_parse_resp() DISCOVER request succeeded
bluetoothd[916489]: profiles/audio/avdtp.c:avdtp_discover_resp() seid 1 type 1 media 0 in use 0
bluetoothd[916489]: profiles/audio/avdtp.c:avdtp_discover_resp() seid 6 type 1 media 0 in use 0
bluetoothd[916489]: profiles/audio/avdtp.c:avdtp_discover_resp() seid 3 type 1 media 0 in use 0
bluetoothd[916489]: profiles/audio/avdtp.c:avdtp_discover_resp() seid 2 type 1 media 0 in use 0
bluetoothd[916489]: profiles/audio/a2dp.c:discover_cb() version 0x0103 err (nil)
bluetoothd[916489]: profiles/audio/sink.c:discovery_complete() Discovery complete
bluetoothd[916489]: profiles/audio/a2dp.c:setup_ref() 0x6afec80: ref=2
bluetoothd[916489]: profiles/audio/a2dp.c:setup_ref() 0x6afec80: ref=3
bluetoothd[916489]: profiles/audio/media.c:media_endpoint_async_call() Calling SelectConfiguration: name = :1.94 path = /MediaEndpoint/A2DPSoc

```

So my headset return exactly the same endpoint I have in cache but perhaps that is not the case depending on the model, or perhaps we are crashing when it detects an endpoint has been remove.

Vudentz, 2021-03-11:
`bluetoothd[49831]: profiles/audio/a2dp.c:a2dp_select_capabilities() Unable to select SEP`

This could indicate there are no local endpoints registered which would manifest if e.g. PulseAudio have crashed which perhaps is the case when we detect that an endpoint is no longer available it disconnects.

i-garrison, 2021-03-11:
This actually looks like a race somewhere. I'm sure below pulseaudio is not crashing anywhere, and it's MediaEndpoint/* entries are exactly the same.

with 5.55
```
bluetoothd: profiles/audio/avdtp.c:avdtp_register_remote_sep() seid 8 type 1 media 0 delay_reporting false
bluetoothd: profiles/audio/a2dp.c:register_remote_sep() Found remote SEP: /org/bluez/hci0/dev_00_1D_43_A0_A9_B7/sep8
bluetoothd: profiles/audio/avdtp.c:avdtp_register_remote_sep() seid 6 type 1 media 0 delay_reporting false
bluetoothd: profiles/audio/a2dp.c:register_remote_sep() Found remote SEP: /org/bluez/hci0/dev_00_1D_43_A0_A9_B7/sep6
bluetoothd: profiles/audio/avdtp.c:avdtp_register_remote_sep() seid 7 type 1 media 0 delay_reporting true
bluetoothd: profiles/audio/a2dp.c:register_remote_sep() Found remote SEP: /org/bluez/hci0/dev_00_1D_43_A0_A9_B7/sep7
bluetoothd: profiles/audio/avdtp.c:avdtp_register_remote_sep() seid 5 type 1 media 0 delay_reporting true
bluetoothd: profiles/audio/a2dp.c:register_remote_sep() Found remote SEP: /org/bluez/hci0/dev_00_1D_43_A0_A9_B7/sep5
bluetoothd: profiles/audio/avdtp.c:avdtp_register_remote_sep() seid 4 type 1 media 0 delay_reporting false
bluetoothd: profiles/audio/a2dp.c:register_remote_sep() Found remote SEP: /org/bluez/hci0/dev_00_1D_43_A0_A9_B7/sep4
bluetoothd: profiles/audio/avdtp.c:avdtp_register_remote_sep() seid 1 type 1 media 0 delay_reporting true
bluetoothd: profiles/audio/a2dp.c:register_remote_sep() Found remote SEP: /org/bluez/hci0/dev_00_1D_43_A0_A9_B7/sep1
bluetoothd: profiles/audio/avdtp.c:session_cb()
bluetoothd: profiles/audio/avdtp.c:avdtp_parse_resp() DISCOVER request succeeded
bluetoothd: profiles/audio/avdtp.c:avdtp_discover_resp() seid 8 type 1 media 0 in use 0
bluetoothd: profiles/audio/avdtp.c:avdtp_discover_resp() seid 6 type 1 media 0 in use 0
bluetoothd: profiles/audio/avdtp.c:avdtp_discover_resp() seid 7 type 1 media 0 in use 0
bluetoothd: profiles/audio/avdtp.c:avdtp_discover_resp() seid 5 type 1 media 0 in use 0
bluetoothd: profiles/audio/avdtp.c:avdtp_discover_resp() seid 4 type 1 media 0 in use 0
bluetoothd: profiles/audio/avdtp.c:avdtp_discover_resp() seid 1 type 1 media 0 in use 0
bluetoothd: profiles/audio/a2dp.c:discover_cb() version 0x0103 err (nil)
bluetoothd: profiles/audio/sink.c:discovery_complete() Discovery complete
bluetoothd: profiles/audio/a2dp.c:endpoint_match_codec_ind() vendor 0x000000d7 codec 0x0024
bluetoothd: profiles/audio/a2dp.c:endpoint_match_codec_ind() vendor 0x0000004f codec 0x0001
bluetoothd: profiles/audio/a2dp.c:setup_ref() 0x55ed4f5eddd0: ref=2
bluetoothd: profiles/audio/a2dp.c:setup_ref() 0x55ed4f5eddd0: ref=3
bluetoothd: profiles/audio/media.c:media_endpoint_async_call() Calling SelectConfiguration: name = :1.293 path = /MediaEndpoint/A2DPSource/sbc_xq_552
bluetoothd: profiles/audio/a2dp.c:setup_unref() 0x55ed4f5eddd0: ref=2
bluetoothd: profiles/audio/a2dp.c:a2dp_config() a2dp_config: selected SEP 0x55ed4f5f6b10
bluetoothd: profiles/audio/a2dp.c:setup_ref() 0x55ed4f5eddd0: ref=3
bluetoothd: profiles/audio/avdtp.c:avdtp_set_configuration() 0x55ed4f605cf0: int_seid=12, acp_seid=1
bluetoothd: profiles/audio/a2dp.c:setup_unref() 0x55ed4f5eddd0: ref=2
bluetoothd: profiles/audio/a2dp.c:setup_unref() 0x55ed4f5eddd0: ref=1
bluetoothd: profiles/audio/avdtp.c:session_cb()
bluetoothd: profiles/audio/avdtp.c:avdtp_parse_resp() SET_CONFIGURATION request succeeded
```

with 5.56
```
bluetoothd: profiles/audio/avdtp.c:avdtp_register_remote_sep() seid 8 type 1 media 0 delay_reporting false
bluetoothd: profiles/audio/a2dp.c:register_remote_sep() Found remote SEP: /org/bluez/hci0/dev_00_1D_43_A0_A9_B7/sep8
bluetoothd: profiles/audio/avdtp.c:avdtp_register_remote_sep() seid 6 type 1 media 0 delay_reporting false
bluetoothd: profiles/audio/a2dp.c:register_remote_sep() Found remote SEP: /org/bluez/hci0/dev_00_1D_43_A0_A9_B7/sep6
bluetoothd: profiles/audio/avdtp.c:avdtp_register_remote_sep() seid 7 type 1 media 0 delay_reporting true
bluetoothd: profiles/audio/a2dp.c:register_remote_sep() Found remote SEP: /org/bluez/hci0/dev_00_1D_43_A0_A9_B7/sep7
bluetoothd: profiles/audio/avdtp.c:avdtp_register_remote_sep() seid 5 type 1 media 0 delay_reporting true
bluetoothd: profiles/audio/a2dp.c:register_remote_sep() Found remote SEP: /org/bluez/hci0/dev_00_1D_43_A0_A9_B7/sep5
bluetoothd: profiles/audio/avdtp.c:avdtp_register_remote_sep() seid 4 type 1 media 0 delay_reporting false
bluetoothd: profiles/audio/a2dp.c:register_remote_sep() Found remote SEP: /org/bluez/hci0/dev_00_1D_43_A0_A9_B7/sep4
bluetoothd: profiles/audio/avdtp.c:avdtp_register_remote_sep() seid 1 type 1 media 0 delay_reporting true
bluetoothd: profiles/audio/a2dp.c:register_remote_sep() Found remote SEP: /org/bluez/hci0/dev_00_1D_43_A0_A9_B7/sep1
bluetoothd: profiles/audio/avdtp.c:avdtp_ref() 0x55b772301c80: ref=1
bluetoothd: profiles/audio/avdtp.c:set_disconnect_timer() timeout 1
bluetoothd: profiles/audio/avdtp.c:avdtp_ref() 0x55b772301c80: ref=2
bluetoothd: profiles/audio/avdtp.c:avdtp_unref() 0x55b772301c80: ref=1
bluetoothd: profiles/audio/avdtp.c:avdtp_ref() 0x55b772301c80: ref=2
bluetoothd: profiles/audio/a2dp.c:setup_ref() 0x55b7722e3200: ref=1
bluetoothd: profiles/audio/sink.c:sink_connect() stream creation in progress
bluetoothd: src/service.c:change_state() 0x55b7722d0030: device 00:1D:43:A0:A9:B7 profile a2dp-sink state changed: disconnected -> connecting (0)
bluetoothd: plugins/policy.c:policy_connect() /org/bluez/hci0/dev_00_1D_43_A0_A9_B7 profile a2dp-sink
[... cut ...]
bluetoothd: profiles/audio/avdtp.c:remove_disappeared() seid 7 disappeared
bluetoothd: profiles/audio/avdtp.c:remove_disappeared() seid 5 disappeared
bluetoothd: profiles/audio/avdtp.c:remove_disappeared() seid 4 disappeared
bluetoothd: profiles/audio/avdtp.c:remove_disappeared() seid 1 disappeared
bluetoothd: profiles/audio/a2dp.c:discover_cb() version 0x0103 err (nil)
bluetoothd: profiles/audio/sink.c:discovery_complete() Discovery complete
bluetoothd: profiles/audio/a2dp.c:a2dp_select_capabilities() Unable to select SEP
```

i-garrison, 2021-03-11:
Need to add that I frequently see this 'Unable to select SEP' error with 5.56 version, but *sometimes* it succeeds (do not have a log for success case yet.)

As far as I remember I have not seen this issue with 5.55 yet.

i-garrison, 2021-03-11:
With 5.56 when it succeeds there is a `discover_cb` in the log and otherwise that log fragment looks the same as with 5.55 version. So I'd guess there is either a race with sending related DISCOVER with 5.56?

Vudentz, 2021-03-11:
Looks like the problem in on avdtp_discover that don't check if the remote SEPs have been discovered yet so it completes the discover directly which in turn causes all the endpoints to be removed. I send a patch for this which should appear here once CI has complete testing it.

casta, 2021-03-11:
Great, with this patch it’s ok for me now

Portisch, 2021-03-11:
Does this solve also my issue?
https://github.com/bluez/bluez/issues/85#issuecomment-796502844

Vudentz, 2021-03-11:
> Does this solve also my issue?
> [#85 (comment)](https://github.com/bluez/bluez/issues/85#issuecomment-796502844)

It should, let me know if you are able to test it. We should probably make it duplicated of this one.

closed 2021-03-12 by BluezTestBot

Portisch, 2021-03-12:
I will do shortly and come back with the results, thank you!

Portisch, 2021-03-12:
Thank you very much!

I can just confirm this fix does solve the issue!

bukajsytlos, 2021-04-11:
Not sure if it is related to this, but I am having trouble with reconnects of my Sony XM4.
I installed bluez 5.58 but still get this behavior. This is in journal
```
apr 11 14:14:23 millipede bluetoothd[1236]: /org/bluez/hci0/dev_94_DB_56_C0_40_4A/sep1/fd2: fd(41) ready
apr 11 14:14:23 millipede rtkit-daemon[2180]: Supervising 10 threads of 4 processes of 1 users.
apr 11 14:14:23 millipede rtkit-daemon[2180]: Successfully made thread 114077 of process 2142 owned by '1000' RT at priority 5.
apr 11 14:14:23 millipede rtkit-daemon[2180]: Supervising 11 threads of 4 processes of 1 users.
apr 11 14:14:23 millipede kernel: input: WH-1000XM4 (AVRCP) as /devices/virtual/input/input45
apr 11 14:14:23 millipede systemd-logind[873]: Watching system buttons on /dev/input/event30 (WH-1000XM4 (AVRCP))
apr 11 14:18:48 millipede bluetoothd[1236]: src/profile.c:ext_io_disconnected() Unable to get io data for Headset Voice gateway: getpeername: Transport endpoint is not connected (107)
apr 11 14:18:48 millipede dbus-daemon[863]: [system] Rejected send message, 0 matched rules; type="method_return", sender=":1.78" (uid=1000 pid=2142 comm="/usr/bin/pulseaudio --daemonize=no --log-target=jo") interface="(unset)" member="(unset)" error name="(unset)" requested_reply="0" destination=":1.16" (uid=0 pid=1236 comm="/usr/lib/bluetooth/bluetoothd ")
apr 11 14:18:48 millipede plasmashell[2101]: org.kde.plasma.pulseaudio: No object for name "bluez_sink.94_DB_56_C0_40_4A.a2dp_sink"
apr 11 14:18:51 millipede bluetoothd[1236]: /org/bluez/hci0/dev_94_DB_56_C0_40_4A/sep1/fd3: fd(42) ready
apr 11 14:18:51 millipede rtkit-daemon[2180]: Supervising 10 threads of 4 processes of 1 users.
apr 11 14:18:51 millipede rtkit-daemon[2180]: Successfully made thread 114930 of process 2142 owned by '1000' RT at priority 5.
apr 11 14:18:51 millipede rtkit-daemon[2180]: Supervising 11 threads of 4 processes of 1 users.
apr 11 14:18:51 millipede kernel: input: WH-1000XM4 (AVRCP) as /devices/virtual/input/input46
apr 11 14:18:51 millipede systemd-logind[873]: Watching system buttons on /dev/input/event30 (WH-1000XM4 (AVRCP))
```