bluez/bluez #103: Dual GATT server running on different hci

inonoob, 2021-03-07:
Hey, 

so I try to run 2 GATT python servers at the same time on two different dongles (hci0, hci1). I have a main python script in which I start both GATT server in their own thread. But as soon as the second GATT server needs to be launched. I always get the following error message: 

```
Exception in thread Thread-3:
Traceback (most recent call last):
  File "/usr/lib/python3.7/threading.py", line 917, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.7/threading.py", line 865, in run
    self._target(*self._args, **self._kwargs)
  File "waterrowerthreads.py", line 55, in BleService2
    bleService2 = smartrowerble.main(out_q, ble_in_q_sr)
  File "/home/pi/pirowflo/src/smartrowerble.py", line 268, in main
    agent_manager.RegisterAgent(AGENT_PATH, "NoInputNoOutput") # register the bluetooth agent with no input and output which should avoid asking for pairing
  File "/usr/local/lib/python3.7/dist-packages/dbus/proxies.py", line 72, in __call__
    return self._proxy_method(*args, **keywords)
  File "/usr/local/lib/python3.7/dist-packages/dbus/proxies.py", line 147, in __call__
    **keywords)
  File "/usr/local/lib/python3.7/dist-packages/dbus/connection.py", line 653, in call_blocking
    message, timeout)
dbus.exceptions.DBusException: org.bluez.Error.AlreadyExists: Already Exists
```
I check the agent manager to be different: 

```
<Interface <ProxyObject wrapping <dbus._dbus.SystemBus (system) at 0xb3df5c90> :1.5 /org/bluez at 0xb3deae70> implementing 'org.bluez.AgentManager1' at 0xb3deab70>
<Interface <ProxyObject wrapping <dbus._dbus.SystemBus (system) at 0xb3df5c90> :1.5 /org/bluez at 0xb3e2b710> implementing 'org.bluez.AgentManager1' at 0xb358c030>
``` 

Why is it still saying `Already Exists`

Has someone an idea ?

Vudentz, 2021-03-08:
I guess you are saying you have 2 different services rather than 2 different servers, the server is implemented by bluetoothd, also for bluetoothd all it matter is that the D-Bus connection and path are unique, if it is not the it will assume it is the same instance, so this suggests the problem is the application side.

inonoob, 2021-03-09:
So, 

currently it is not possible to run two Gatt python server example in separate threads in the same python script ? The wired thing here is if I'm quick enough then sometime it works and sometime it doesn't.

Vudentz, 2021-03-10:
> So,
> 
> currently it is not possible to run two Gatt python server example in separate threads in the same python script ? The wired thing here is if I'm quick enough then sometime it works and sometime it doesn't.

You may need to create different connection if you are running just different threads since for the daemon identify based on the connection id if python just reuses the same connection it won't work, or run as separate process, anyway this is on the client side and there is nothing we can do if the connection ID and path clashes with othe instances.

closed 2021-03-10 by Vudentz