bluez/bluez #107: bluez needs a workaround for HCIs with the HCI_QUIRK_INVALID_BDADDR quirk

jwrdegoede, 2021-03-14:
I have been debugging a regression where Bluetooth has stopped working on some broadcom Wifi/BT chipsets. This is caused by the BT parts of these chips not having a unique MAC programmed into them. Instead they all use the same factory-default address.

This used to work as long as only one device with this factory-default address was in the room. But a while ago the kernel learned to recognize the factory-default address and now it (rightly) refuses to use this:
https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/commit/?id=5035726128cd2e38

Instead it expects userspace to configure a valid MAC address. So it looks like we need to have a workaround for this in bluez.

I'm willing to spend some time implementing a workaround for this myself, but first I would like to get some input from you (the bluez devs) to hear if I'm on the right track. Also note that this will not a spare-time project for me, so I don't know when I will get around to working on this. If someone who knows the bluez code better then me wants to write a workaround instead, I would be happy to test it.

Below are my current ideas on implementing a workaround for this:

This workaround should do something like the following:

1. Check if the HCI needs to have its MAC address set, if it already has a MAC set goto 8.
2. readlink("/sys/class/bluetooth/hci#")
3. Change the path returned to readlink into a filename (replacing '/' with e.g. '_')
4. Try to read the MAC from /var/lib/bluetooth/<name>-btaddr exists, on success goto 7.
5. Generate a static random MAC (last 2 bits are both 1, e.g. last byte is 0x03.)
6. Write the MAC to /var/lib/bluetooth/<name>-btaddr
7. Set the MAC address
8. Done

ATM I'm using a workaround with a udev rule, which activates a systemd service (to escape the "jail" for udev rules which disallows using bluetooth netlink) which just statically sets the address through the btmgmt util:

```
ACTION=="add", SUBSYSTEM=="bluetooth", ENV{DEVTYPE}=="host" \
  ENV{DEVPATH}=="*/serial[0-9]*/serial[0-9]*/bluetooth/hci[0-9]*", \
  TAG+="systemd", ENV{SYSTEMD_WANTS}="hci-btaddress@%k.service"
```

and the service looks like this:

```
[Unit]
Description=HCI btaddress check/fixup

[Service]
Type=simple
ExecStart=/usr/sbin/btmgmt -i %I public-addr 'B0:F1:EC:82:1D:B3'

```

My first idea for implementing a workaround for this is to have the steps which I described above be done by a dedicated tool, which would then get run from the ExecStart line.

As you can see I only let the udev rule trigger on uart attached HCIs since only those seem to need this, the idea here is to not cause any extra work to be done at boot in other cases.

Alternatively the whole udev/systemd-service dance could be avoided by having bluetoothd detect these HCIs itself and to the workaround inside bluetoothd.

So 2 questions for you:

1. Does the step-by-step plan for the wa which I gave above sound good, or do you see any problems there?
2. Should I go for the udev + systemd-service + separate tool route, or should this be integrated into bluetoothd?

jwrdegoede, 2021-04-13:
Ping? As mentioned I can make some time to try and fix this, but could really use some input (see the 2 questions at the end of my initial comment) on the best direction to take for fixing this.

jwrdegoede, 2025-09-28:
This issue is still relevant, so not stale.

jwrdegoede, 2025-10-31:
This still an issue (adding a comment to avoid auto-close).

jglathe, 2026-01-02:
No its not. Still an issue.