bluez/bluez #109: segfault in service_disconnect ... queue_remove_if / match_endpoint_by_path

martinetd, 2021-03-17:
downstream bug: https://bugzilla.redhat.com/show_bug.cgi?id=1921254 has been inactive for a while so bringing it here..
Appologizes if this has already been fixed, didn't see any issue like this but I didn't test/could have been fixed otherwise.

Here's a full backtrace if that helps:
```
(gdb) bt full
#0  0x00007f03f11a6afe in __strcmp_avx2 () from /lib64/libc.so.6
No symbol table info available.
#1  0x000055e5d7e6ef71 in match_endpoint_by_path (a=<optimized out>, b=<optimized out>) at profiles/audio/media.c:2188
        endpoint = <optimized out>
        path = <optimized out>
#2  0x000055e5d7eec504 in queue_remove_if (queue=0x55e5d98d2460, function=0x55e5d7e6ef60 <match_endpoint_by_path>, user_data=0x55e5d98da6b0) at src/shared/queue.c:279
        entry = 0x55e5d98e7220
        prev = 0x55e5d98f6980
#3  0x000055e5d7e74f40 in proxy_removed_cb (proxy=<optimized out>, user_data=0x55e5d98d91c0) at profiles/audio/media.c:2210
        app = 0x55e5d98d91c0
        endpoint = <optimized out>
        player = <optimized out>
        iface = <optimized out>
        path = <optimized out>
        __btd_debug_desc = {
          file = 0x55e5d7f074d2 "profiles/audio/media.c",
          flags = 0
        }
        __func__ = "proxy_removed_cb"
        __btd_debug_desc = {
          file = 0x55e5d7f074d2 "profiles/audio/media.c",
          flags = 0
        }
#4  0x000055e5d7eeaffd in proxy_free (data=0x55e5d98f3690) at gdbus/client.c:568
        client = <optimized out>
        proxy = 0x55e5d98f3690
#5  0x00007f03f12afe40 in g_list_foreach () from /lib64/libglib-2.0.so.0
No symbol table info available.
#6  0x00007f03f12ba88f in g_list_free_full () from /lib64/libglib-2.0.so.0
No symbol table info available.
#7  0x000055e5d7ee5cca in service_disconnect (conn=0x55e5d98d3fc0, user_data=0x55e5d98f31f0) at gdbus/client.c:1300
        client = 0x55e5d98f31f0
#8  0x000055e5d7ee84f8 in service_filter (connection=0x55e5d98d3fc0, message=<optimized out>, user_data=0x55e5d98da7e0) at gdbus/watch.c:471
        data = 0x55e5d98da7e0
        cb = 0x55e5d98f32c0
        name = 0x55e5d9905e34 ":1.12855"
        old = 0x55e5d9905e44 ":1.12855"
        new = 0x55e5d9905e54 ""
#9  0x000055e5d7ee7983 in message_filter (connection=0x55e5d98d3fc0, message=0x55e5d9920670, user_data=<optimized out>) at gdbus/watch.c:544
        data = 0x55e5d98da7e0
        sender = 0x55e5d98e3278 "org.freedesktop.DBus"
        path = 0x55e5d98e3298 "/org/freedesktop/DBus"
        iface = 0x55e5d98e32b8 "org.freedesktop.DBus"
        member = 0x55e5d98e32d8 "NameOwnerChanged"
        arg = 0x55e5d9905e34 ":1.12855"
        current = 0x55e5d98e1ca0 = {0x55e5d98da7e0, 0x55e5d98f3560, 0x55e5d98d6d70, 0x55e5d98f37c0, 0x55e5d98f4200, 0x55e5d98f4a00, 0x55e5d98f5120, 0x55e5d98f5950, 
          0x55e5d98f6070}
        delete_listener = 0x0
#10 0x00007f03f1234221 in dbus_connection_dispatch () from /lib64/libdbus-1.so.3
No symbol table info available.
#11 0x000055e5d7ee3958 in message_dispatch (data=0x55e5d98d3fc0) at gdbus/mainloop.c:59
        conn = 0x55e5d98d3fc0
#12 0x00007f03f12beb5b in g_idle_dispatch () from /lib64/libglib-2.0.so.0
No symbol table info available.
#13 0x00007f03f12bfa9f in g_main_context_dispatch () from /lib64/libglib-2.0.so.0
No symbol table info available.
#14 0x00007f03f1311a98 in g_main_context_iterate.constprop () from /lib64/libglib-2.0.so.0
No symbol table info available.
#15 0x00007f03f12bf163 in g_main_loop_run () from /lib64/libglib-2.0.so.0
No symbol table info available.
#16 0x000055e5d7e5f1a9 in mainloop_run () at src/shared/mainloop-glib.c:66
No locals.
#17 mainloop_run_with_signal (func=0x55e5d7ea0c20 <signal_callback>, user_data=0x0) at src/shared/mainloop-notify.c:188
        data = 0x55e5d98dbd20
        io = 0x55e5d98dbd40
        ret = <optimized out>
        data = <optimized out>
        io = <optimized out>
        ret = <optimized out>
        __n = <optimized out>
        __s = <optimized out>
        __p = <optimized out>
#18 main (argc=<optimized out>, argv=<optimized out>) at src/main.c:1084
        context = <optimized out>
        err = 0x0
        sdp_mtu = 0
        sdp_flags = <optimized out>
        gdbus_flags = <optimized out>
        __func__ = "main"
        __btd_debug_desc = {
          file = 0x55e5d7f113a4 "src/main.c",
          flags = 0
        }
(gdb) up 2
#2  0x000055e5d7eec504 in queue_remove_if (queue=0x55e5d98d2460, function=0x55e5d7e6ef60 <match_endpoint_by_path>, user_data=0x55e5d98da6b0) at src/shared/queue.c:279
279			if (function(entry->data, user_data)) {
(gdb) p *((struct media_endpoint*)(entry->data))
$8 = {
  sep = 0x0,
  sender = 0x0,
  path = 0x0,
  uuid = 0x0,
  codec = 0 '\000',
  capabilities = 0xa000100240418 <error: Cannot access memory at address 0xa000100240418>,
  size = 4294967295,
  hs_watch = 3649991104,
  ag_watch = 21989,
  watch = 3650096560,
  requests = 0x55e5d98ef170 = {0x55e5d98dee60, 0x55e5d98dee20, 0x55e5d98dd810},
  adapter = 0x0,
  transports = 0x0
}
```

I can reproduce this pretty reliably by killing pipewire processes with an audio device previously connected at some point, looking at the entry data it looks like it's been freed somewhere else first?

I'll wait a few days see if someone has an idea, if no answer/idea then will try to reproduce on master with ASAN to check for use-after-free that it looks like it is, should pinpoint what went wrong.

Thanks.

pv, 2021-03-30:
Address sanitizer trace below of a crash at same location, seen on bluetoothd 9d493f8060ee1, when shutting down pipewire-media-session:

<details>

```
maalis 30 22:26:18 bluetoothd[23952]: Endpoint unregistered: sender=:1.1026 path=/MediaEndpoint/A2DPSource/ldac
maalis 30 22:26:18 bluetoothd[23952]: Endpoint unregistered: sender=:1.1026 path=/MediaEndpoint/A2DPSink/aptx_hd
maalis 30 22:26:18 bluetoothd[23952]: Endpoint unregistered: sender=:1.1026 path=/MediaEndpoint/A2DPSource/aptx_hd
maalis 30 22:26:18 bluetoothd[23952]: Endpoint unregistered: sender=:1.1026 path=/MediaEndpoint/A2DPSink/aptx
maalis 30 22:26:18 bluetoothd[23952]: Endpoint unregistered: sender=:1.1026 path=/MediaEndpoint/A2DPSource/aptx
maalis 30 22:26:18 bluetoothd[23952]: Endpoint unregistered: sender=:1.1026 path=/MediaEndpoint/A2DPSource/aac
maalis 30 22:26:18 bluetoothd[23952]: Endpoint unregistered: sender=:1.1026 path=/MediaEndpoint/A2DPSink/sbc_xq
maalis 30 22:26:18 bluetoothd[23952]: Endpoint unregistered: sender=:1.1026 path=/MediaEndpoint/A2DPSource/sbc_xq
maalis 30 22:26:18 bluetoothd[23952]: Endpoint unregistered: sender=:1.1026 path=/MediaEndpoint/A2DPSink/sbc
maalis 30 22:26:18 bluetoothd[23952]: Endpoint unregistered: sender=:1.1026 path=/MediaEndpoint/A2DPSource/sbc
maalis 30 22:26:18 bluetoothd[23952]: Destroy Adv Monitor Manager
maalis 30 22:26:18 bluetoothd[23952]: Battery Provider Manager destroyed
maalis 30 23:08:03 bluetoothd[23952]: =================================================================
maalis 30 23:08:03 bluetoothd[23952]: ==23952==ERROR: AddressSanitizer: heap-use-after-free on address 0x60800004ac30 at pc 0x555935395fd9 bp 0x7ffd729b25b0 sp 0x7ffd729b25a8
maalis 30 23:08:03 bluetoothd[23952]: READ of size 8 at 0x60800004ac30 thread T0
maalis 30 23:08:03 bluetoothd[23952]:     #0 0x555935395fd8 in match_endpoint_by_path profiles/audio/media.c:2188
maalis 30 23:08:03 bluetoothd[23952]:     #1 0x555935557fa6 in queue_remove_if src/shared/queue.c:279
maalis 30 23:08:03 bluetoothd[23952]:     #2 0x5559353960ec in proxy_removed_cb profiles/audio/media.c:2210
maalis 30 23:08:03 bluetoothd[23952]:     #3 0x5559355515c1 in proxy_free gdbus/client.c:568
maalis 30 23:08:03 bluetoothd[23952]:     #4 0x7fab7d9d1e3f in g_list_foreach (/lib64/libglib-2.0.so.0+0x43e3f)
maalis 30 23:08:03 bluetoothd[23952]:     #5 0x7fab7d9dc88e in g_list_free_full (/lib64/libglib-2.0.so.0+0x4e88e)
maalis 30 23:08:03 bluetoothd[23952]:     #6 0x555935554f2a in service_disconnect gdbus/client.c:1300
maalis 30 23:08:03 bluetoothd[23952]:     #7 0x555935544723 in service_filter gdbus/watch.c:468
maalis 30 23:08:03 bluetoothd[23952]:     #8 0x555935544efb in message_filter gdbus/watch.c:544
maalis 30 23:08:03 bluetoothd[23952]:     #9 0x7fab7d956220 in dbus_connection_dispatch (/lib64/libdbus-1.so.3+0x19220)
maalis 30 23:08:03 bluetoothd[23952]:     #10 0x5559355419c1 in message_dispatch gdbus/mainloop.c:59
maalis 30 23:08:03 bluetoothd[23952]:     #11 0x7fab7d9e0b5a  (/lib64/libglib-2.0.so.0+0x52b5a)
maalis 30 23:08:03 bluetoothd[23952]:     #12 0x7fab7d9e1a9e in g_main_context_dispatch (/lib64/libglib-2.0.so.0+0x53a9e)
maalis 30 23:08:03 bluetoothd[23952]:     #13 0x7fab7da33a97  (/lib64/libglib-2.0.so.0+0xa5a97)
maalis 30 23:08:03 bluetoothd[23952]:     #14 0x7fab7d9e1162 in g_main_loop_run (/lib64/libglib-2.0.so.0+0x53162)
maalis 30 23:08:03 bluetoothd[23952]:     #15 0x55593559724b in mainloop_run src/shared/mainloop-glib.c:66
maalis 30 23:08:03 bluetoothd[23952]:     #16 0x555935597ce0 in mainloop_run_with_signal src/shared/mainloop-notify.c:188
maalis 30 23:08:03 bluetoothd[23952]:     #17 0x55593543e0b0 in main src/main.c:1091
maalis 30 23:08:03 bluetoothd[23952]:     #18 0x7fab7d6891e1 in __libc_start_main (/lib64/libc.so.6+0x281e1)
maalis 30 23:08:03 bluetoothd[23952]:     #19 0x55593534fbad in _start (/usr/local/stow/bluez-9d493f8060ee1/libexec/bluetooth/bluetoothd+0xe5bad)
maalis 30 23:08:03 bluetoothd[23952]: 0x60800004ac30 is located 16 bytes inside of 96-byte region [0x60800004ac20,0x60800004ac80)
maalis 30 23:08:03 bluetoothd[23952]: freed by thread T0 here:
maalis 30 23:08:03 bluetoothd[23952]:     #0 0x7fab7dc080c7 in __interceptor_free (/lib64/libasan.so.6+0xab0c7)
maalis 30 23:08:03 bluetoothd[23952]:     #1 0x7fab7d9e670c in g_free (/lib64/libglib-2.0.so.0+0x5870c)
maalis 30 23:08:03 bluetoothd[23952]:     #2 0x55593538bdc5 in media_endpoint_remove profiles/audio/media.c:219
maalis 30 23:08:03 bluetoothd[23952]:     #3 0x55593538d741 in release_endpoint profiles/audio/media.c:544
maalis 30 23:08:04 bluetoothd[23952]:     #4 0x55593538dcb8 in a2dp_destroy_endpoint profiles/audio/media.c:664
maalis 30 23:08:04 bluetoothd[23952]:     #5 0x555935371459 in a2dp_unregister_sep profiles/audio/a2dp.c:2414
maalis 30 23:08:04 bluetoothd[23952]:     #6 0x555935372304 in a2dp_remove_sep profiles/audio/a2dp.c:2574
maalis 30 23:08:04 bluetoothd[23952]:     #7 0x55593538bc56 in media_endpoint_remove profiles/audio/media.c:205
maalis 30 23:08:04 bluetoothd[23952]:     #8 0x55593538d741 in release_endpoint profiles/audio/media.c:544
maalis 30 23:08:04 bluetoothd[23952]:     #9 0x555935396f2a in path_free profiles/audio/media.c:2388
maalis 30 23:08:04 bluetoothd[23952]:     #10 0x555935548a20 in remove_interface gdbus/object.c:654
maalis 30 23:08:04 bluetoothd[23952]:     #11 0x55593554c900 in g_dbus_unregister_interface gdbus/object.c:1388
[... cut ...]
maalis 30 23:08:04 bluetoothd[23952]:   Intra object redzone:    bb
maalis 30 23:08:04 bluetoothd[23952]:   ASan internal:           fe
maalis 30 23:08:04 bluetoothd[23952]:   Left alloca redzone:     ca
maalis 30 23:08:04 bluetoothd[23952]:   Right alloca redzone:    cb
maalis 30 23:08:04 bluetoothd[23952]:   Shadow gap:              cc
maalis 30 23:08:04 bluetoothd[23952]: ==23952==ABORTING
maalis 30 23:08:04 systemd[1]: bluetooth.service: Main process exited, code=dumped, status=6/ABRT
```

</details>

a2leexx, 2021-06-23:
I also have this issue (bluez 5.59-1, Fedora 34). This happens every time on resuming from sleep.

Downstream bug (Fedora 34): https://bugzilla.redhat.com/show_bug.cgi?id=1936141. 

Full backtrace:
```
#0  __strcmp_avx2 () at ../sysdeps/x86_64/multiarch/strcmp-avx2.S:101
No locals.
#1  0x000055a649b5e411 in match_endpoint_by_path (a=<optimized out>, b=<optimized out>) at profiles/audio/media.c:2188
        endpoint = <optimized out>
        path = <optimized out>
#2  0x000055a649bdfe54 in queue_remove_if (queue=0x55a64b945060, function=0x55a649b5e400 <match_endpoint_by_path>, user_data=0x55a64b94ee30)
    at src/shared/queue.c:279
        entry = 0x55a64b938a40
        prev = 0x55a64b9384b0
#3  0x000055a649b67023 in proxy_removed_cb (proxy=<optimized out>, user_data=0x55a64b92e790) at profiles/audio/media.c:2210
        app = 0x55a64b92e790
        endpoint = <optimized out>
        player = <optimized out>
        iface = <optimized out>
        path = 0x55a64b94ee30 "/MediaEndpoint/A2DPSource/ldac"
        __btd_debug_desc = {file = 0x55a649bfce91 "profiles/audio/media.c", flags = 0}
        __func__ = "proxy_removed_cb"
        __btd_debug_desc = {file = 0x55a649bfce91 "profiles/audio/media.c", flags = 0}
#4  0x000055a649bdec6d in proxy_free (data=0x55a64b94b7d0) at gdbus/client.c:568
        client = <optimized out>
        proxy = 0x55a64b94b7d0
#5  0x00007f2fcfa53bf0 in g_list_foreach (list=<optimized out>, func=0x55a649bdec20 <proxy_free>, user_data=0x0) at ../glib/glist.c:1090
        next = 0x55a64b942ce0
#6  0x00007f2fcfa535af in g_list_free_full (list=0x55a64b92de80, free_func=<optimized out>) at ../glib/glist.c:244
--Type <RET> for more, q to quit, c to continue without paging--c
No locals.
#7  0x000055a649bdc58a in service_disconnect (conn=0x55a64b931d50, user_data=0x55a64b9452d0) at gdbus/client.c:1300
        client = 0x55a64b9452d0
#8  0x000055a649bd5b68 in service_filter (connection=0x55a64b931d50, message=<optimized out>, user_data=0x55a64b9382a0) at gdbus/watch.c:471
        data = 0x55a64b9382a0
        cb = 0x55a64b9453a0
        name = 0x55a64b953254 ":1.100"
        old = 0x55a64b953260 ":1.100"
        new = 0x55a64b95326c ""
#9  0x000055a649bd5093 in message_filter (connection=0x55a64b931d50, message=0x55a64b947370, user_data=<optimized out>) at gdbus/watch.c:544
        data = 0x55a64b9382a0
        sender = 0x55a64b944a68 "org.freedesktop.DBus"
        path = 0x55a64b944a88 "/org/freedesktop/DBus"
        iface = 0x55a64b944aa8 "org.freedesktop.DBus"
        member = 0x55a64b944ac8 "NameOwnerChanged"
        arg = 0x55a64b953254 ":1.100"
        current = 0x55a64b93fb00
        delete_listener = 0x0
#10 0x00007f2fcf9cd191 in dbus_connection_dispatch (connection=0x55a64b931d50) at ../../dbus/dbus-connection.c:4704
        filter = <optimized out>
        next = 0x55a64b936e70
        message = <optimized out>
        link = <optimized out>
        reply_serial = <optimized out>
        filter_list_copy = 0x55a64b936e40
        status = <optimized out>
        found_object = <optimized out>
        message_link = 0x55a64b930230
        result = DBUS_HANDLER_RESULT_NOT_YET_HANDLED
        pending = <optimized out>
        message = <optimized out>
        link = <optimized out>
        filter_list_copy = <optimized out>
        message_link = <optimized out>
        result = <optimized out>
        pending = <optimized out>
        reply_serial = <optimized out>
        status = <optimized out>
        found_object = <optimized out>
[... 53 lines cut ...]
        begin_time_nsec = 2877438328221
#17 0x00007f2fcfa58a93 in g_main_loop_run (loop=0x55a64b92e520) at ../glib/gmain.c:4329
        self = <optimized out>
        __func__ = "g_main_loop_run"
#18 0x000055a649b51240 in mainloop_run () at src/shared/mainloop-glib.c:66
No locals.
#19 mainloop_run_with_signal (func=0x55a649b92430 <signal_callback>, user_data=0x0) at src/shared/mainloop-notify.c:188
        data = 0x55a64b939db0
        io = 0x55a64b939dd0
        ret = <optimized out>
        data = <optimized out>
        io = <optimized out>
        ret = <optimized out>
        __n = <optimized out>
        __s = <optimized out>
        __p = <optimized out>
#20 main (argc=<optimized out>, argv=<optimized out>) at src/main.c:1120
        context = <optimized out>
        err = 0x0
        sdp_mtu = 0
        sdp_flags = <optimized out>
        gdbus_flags = <optimized out>
        __func__ = "main"
        __btd_debug_desc = {file = 0x55a649c05490 "src/main.c", flags = 0}
```

a2leexx, 2021-06-30:
I want to clarify that bluetoothd crashes on resuming from suspend-to-RAM every time when Bluetooth is disabled in GNOME settings. When Bluetooth is enabled, it works without issues.

jirislaby, 2021-07-22:
`app->endpoints` in my `bluez-5.59` seems to contain garbage:
```
(gdb) set $x = app->endpoints->head
(gdb) while $x
 >x/s $x->data
 >set $x = $x->next
 >end
0x561315f80520: "\020\016\370\025\023V"
0x561315f81780: "\201\267\305t\026V"
0x561315f82310: "\031"
0x561315f82510: ""
0x561315f827a0: "\001"
0x561315f82a30: "\f"
0x561315f82cc0: ""
0x561315f82f50: "\240\316\367\025\023V"
```
Matching against (looks like correct) path:
```
(gdb) p path
$24 = 0x561315f7a130 "/MediaEndpoint/A2DPSource/ldac"
```

a2leexx, 2021-07-26:
On my computer issue disappeared after updating to bluez 5.60.

closed 2025-08-29 by github-actions[bot]