bluez/bluez #113: WH-1000XM3: intermittently reconnects

ymatsiuk, 2021-03-22:
Running [this commit](https://github.com/bluez/bluez/commit/2af9ef2fa0a3aee7bbeb7fd2c10205dd5fcfd2d3) which is supposed to fix profile caching, but overall experience remains the same. Below you can see how often headphones reconnect:

```
Mar 22 08:42:53 nixps bluetoothd[1494]: Bluetooth daemon 5.56
Mar 22 08:42:53 nixps bluetoothd[1494]: src/main.c:check_config() Unknown group Controller in /etc/bluetooth/main.conf
Mar 22 08:42:53 nixps bluetoothd[1494]: Starting SDP server
Mar 22 08:42:53 nixps bluetoothd[1494]: Bluetooth management interface 1.19 initialized
Mar 22 08:42:54 nixps bluetoothd[1494]: profiles/sap/server.c:sap_server_register() Sap driver initialization failed.
Mar 22 08:42:54 nixps bluetoothd[1494]: sap-server: Operation not permitted (1)
Mar 22 08:42:54 nixps bluetoothd[1494]: Endpoint registered: sender=:1.28 path=/MediaEndpoint/A2DPSource/ldac
Mar 22 08:42:54 nixps bluetoothd[1494]: Endpoint registered: sender=:1.28 path=/MediaEndpoint/A2DPSink/aptx_hd
Mar 22 08:42:54 nixps bluetoothd[1494]: Endpoint registered: sender=:1.28 path=/MediaEndpoint/A2DPSource/aptx_hd
Mar 22 08:42:54 nixps bluetoothd[1494]: Endpoint registered: sender=:1.28 path=/MediaEndpoint/A2DPSink/aptx
Mar 22 08:42:54 nixps bluetoothd[1494]: Endpoint registered: sender=:1.28 path=/MediaEndpoint/A2DPSource/aptx
Mar 22 08:42:54 nixps bluetoothd[1494]: Endpoint registered: sender=:1.28 path=/MediaEndpoint/A2DPSource/aac
Mar 22 08:42:54 nixps bluetoothd[1494]: Endpoint registered: sender=:1.28 path=/MediaEndpoint/A2DPSink/sbc_xq
Mar 22 08:42:54 nixps bluetoothd[1494]: Endpoint registered: sender=:1.28 path=/MediaEndpoint/A2DPSource/sbc_xq
Mar 22 08:42:54 nixps bluetoothd[1494]: Endpoint registered: sender=:1.28 path=/MediaEndpoint/A2DPSink/sbc
Mar 22 08:42:54 nixps bluetoothd[1494]: Endpoint registered: sender=:1.28 path=/MediaEndpoint/A2DPSource/sbc
Mar 22 13:33:54 nixps bluetoothd[1494]: /org/bluez/hci0/dev_CC_98_8B_93_08_1F/sep6/fd0: fd(31) ready
Mar 22 13:46:43 nixps bluetoothd[1494]: src/profile.c:ext_io_disconnected() Unable to get io data for Hands-Free Voice gateway: getpeername: Transport endpoint is not connected (107)
Mar 22 13:51:46 nixps bluetoothd[1494]: /org/bluez/hci0/dev_CC_98_8B_93_08_1F/sep6/fd1: fd(32) ready
Mar 22 14:03:21 nixps bluetoothd[1494]: src/profile.c:ext_io_disconnected() Unable to get io data for Hands-Free Voice gateway: getpeername: Transport endpoint is not connected (107)
Mar 22 14:03:23 nixps bluetoothd[1494]: src/profile.c:record_cb() Unable to get Hands-Free Voice gateway SDP record: Host is down
Mar 22 14:03:25 nixps bluetoothd[1494]: /org/bluez/hci0/dev_CC_98_8B_93_08_1F/fd2: fd(30) ready
Mar 22 14:03:56 nixps bluetoothd[1494]: src/profile.c:ext_io_disconnected() Unable to get io data for Hands-Free Voice gateway: getpeername: Transport endpoint is not connected (107)
Mar 22 14:08:30 nixps bluetoothd[1494]: src/profile.c:ext_io_disconnected() Unable to get io data for Hands-Free Voice gateway: getpeername: Transport endpoint is not connected (107)
Mar 22 15:18:31 nixps bluetoothd[1494]: /org/bluez/hci0/dev_CC_98_8B_93_08_1F/sep6/fd4: fd(32) ready
Mar 22 15:22:44 nixps bluetoothd[1494]: Player registered: sender=:1.130 path=/_1_65
Mar 22 15:33:40 nixps bluetoothd[1494]: src/profile.c:ext_io_disconnected() Unable to get io data for Hands-Free Voice gateway: getpeername: Transport endpoint is not connected (107)
Mar 22 15:33:42 nixps bluetoothd[1494]: src/profile.c:record_cb() Unable to get Hands-Free Voice gateway SDP record: Host is down
Mar 22 15:33:44 nixps bluetoothd[1494]: /org/bluez/hci0/dev_CC_98_8B_93_08_1F/fd5: fd(30) ready
Mar 22 15:34:41 nixps bluetoothd[1494]: profiles/audio/avdtp.c:handle_unanswered_req() No reply to Suspend request
Mar 22 15:34:41 nixps bluetoothd[1494]: src/profile.c:ext_io_disconnected() Unable to get io data for Hands-Free Voice gateway: getpeername: Transport endpoint is not connected (107)
Mar 22 15:34:43 nixps bluetoothd[1494]: src/profile.c:record_cb() Unable to get Hands-Free Voice gateway SDP record: Host is down
```

Meanwhile I see this during `btmon` (let me know if you need full dump to analyze):

```
Bluetooth monitor ver 5.56
= Note: Linux version 5.11.7 (x86_64)                                                                                                                                                                                                  0.135024
= Note: Bluetooth subsystem version 2.22                                                                                                                                                                                               0.135027
= New Index: 04:6C:59:F6:42:A7 (Primary,USB,hci0)                                                                                                                                                                               [hci0] 0.135028
= Open Index: 04:6C:59:F6:42:A7                                                                                                                                                                                                 [hci0] 0.135029
= Index Info: 04:6C:59:F6:42:A7 (Intel Corp.)                                                                                                                                                                                   [hci0] 0.135029
@ MGMT Open: bluetoothd (privileged) version 1.19                                                                                                                                                                             {0x0001} 0.135031
> HCI Event: Disconnect Complete (0x05) plen 4                                                                                                                                                                               #1 [hci0] 5.971821
        Status: Success (0x00)
        Handle: 512
        Reason: Connection Timeout (0x08)
@ MGMT Event: Device Disconnected (0x000c) plen 8                                                                                                                                                                      {0x0001} [hci0] 5.971848
        BR/EDR Address: CC:98:8B:93:08:1F (SONY Visual Products Inc.)
        Reason: Connection timeout (0x01)
= bluetoothd: profiles/audio/avdtp.c:handle_unanswered_req() No reply to Suspend request                                                                                                                                               5.972371
< HCI Command: Write Scan Enable (0x03|0x001a) plen 1                                                                                                                                                                        #2 [hci0] 5.986090
        Scan enable: Page Scan (0x02)
> HCI Event: Command Complete (0x0e) plen 4                                                                                                                                                                                  #3 [hci0] 5.987854
      Write Scan Enable (0x03|0x001a) ncmd 2
        Status: Success (0x00)
= bluetoothd: src/profile.c:ext_io_disconnected() Unable to get io data for Hands-Free Voice gateway: getpeername: Transport endpoint is not connected (107)                                                                           5.994355
< HCI Command: Create Connection (0x01|0x0005) plen 13                                                                                                                                                                       #4 [hci0] 6.931751
        Address: CC:98:8B:93:08:1F (SONY Visual Products Inc.)
        Packet type: 0xcc18
          DM1 may be used
          DH1 may be used
          DM3 may be used
          DH3 may be used
          DM5 may be used
          DH5 may be used
[... 13 lines cut ...]
@ MGMT Event: Connect Failed (0x000d) plen 8                                                                                                                                                                           {0x0001} [hci0] 7.273109
        BR/EDR Address: CC:98:8B:93:08:1F (SONY Visual Products Inc.)
        Status: Connect Failed (0x04)
= bluetoothd: src/profile.c:record_cb() Unable to get Hands-Free Voice gateway SDP record: Host is down                                                                                                                                7.273660
[... cut ...]
        Status: Success (0x00)
> HCI Event: Role Change (0x12) plen 8                                                                                                                                                                                      #10 [hci0] 7.425957
        Status: Success (0x00)
        Address: CC:98:8B:93:08:1F (SONY Visual Products Inc.)
        Role: Master (0x00)
```

Not sure where to report this either here or to kernel firmware people. Thanks in advance for the help.

Vudentz, 2021-03-22:
> Running [this commit](https://github.com/bluez/bluez/commit/2af9ef2fa0a3aee7bbeb7fd2c10205dd5fcfd2d3) which is supposed to fix profile caching, but overall experience remains the same. Below you can see how often headphones reconnect:

Try with the latest tree, the commit that fixed it was https://github.com/bluez/bluez/commit/28ddec8d6b829e002fa268c07b71e4c564ba9e16

ymatsiuk, 2021-03-22:
@Vudentz `28ddec8` is actually included into `2af9ef2`. Nonetheless I think I solved the issue downgrading the firmware. More details [here](https://gitlab.freedesktop.org/pipewire/pipewire/-/issues/732#note_849446). I'm going to report this to Intel developers in the kernel bugzilla. Thanks.

closed 2021-03-22 by ymatsiuk