bluez/bluez #114: Bluez segfaults without obvious reason

Meister1593, 2021-03-22:
I'm having a problem with bluez with random segfaults for quite a while
I'm not quite sure why they happen, but they are sudden and i don't do anything special while they happen.

I'm using bluetooth headphones with A2DP profile, 
**Headset Model**: Bluedio T-Monitor
**Adapter**: CSR 4.0 (lsusb: `Bus 003 Device 003: ID 0a12:0001 Cambridge Silicon Radio, Ltd Bluetooth Dongle (HCI mode)`)
**Distrubition**: Archlinux
**DE/WM**: Awesomewm
**Bluez version**: 5.56
**Pulseaudio version**: 14.2

One possible connection is pulseaudio, but it's a long throw.

Journalctl when this happened: 
[bluetooth.log](https://github.com/bluez/bluez/files/6184023/bluetooth.log)

I have coredump but not sure if it's safe to share it here, so if it's safe and it would help, let me know.

Vudentz, 2021-03-22:
Are you able to reproduce with the latest git? It would be good to have access to the bluetoothd and btmon/HCI traces so we can at least tell what part of the code is triggering this.

Meister1593, 2021-03-24:
I'm currently on bluez-git, i will test this for a week to see if it will still segfault
Usually i get 3 segfaults every 2-3 days (even though i use headset daily), so this can take a while.

Meister1593, 2021-03-24:
> Are you able to reproduce with the latest git? It would be good to have access to the bluetoothd and btmon/HCI traces so we can at least tell what part of the code is triggering this.

I'm getting even more segfaults now, the same error related to dbus...

What do i need to do to capture bluetoothd and btmon/HCI traces?

Meister1593, 2021-03-27:
I got some info with gdb and coredump, with non-stripped bluez
```
#0  0x00007fcef3a74f19 in __strcasecmp_l_avx () from /usr/lib/libc.so.6
#1  0x000055e97be973f1 in play_status_to_val ()
#2  0x000055e97be97ece in avrcp_player_event.part ()
#3  0x000055e97be97f66 in notify_addressed_player_changed ()
#4  0x000055e97be9b6bd in avrcp_unregister_player ()
#5  0x000055e97be8e108 in unregister_player ()
#6  0x000055e97bf10139 in process_message ()
#7  0x00007fcef3bf17d6 in ?? () from /usr/lib/libdbus-1.so.3
#8  0x00007fcef3be13bd in dbus_connection_dispatch () from /usr/lib/libdbus-1.so.3
#9  0x000055e97bf0c7d1 in message_dispatch ()
#10 0x00007fcef3c72ea0 in g_main_context_dispatch () from /usr/lib/libglib-2.0.so.0
#11 0x00007fcef3cc6a49 in ?? () from /usr/lib/libglib-2.0.so.0
#12 0x00007fcef3c72503 in g_main_loop_run () from /usr/lib/libglib-2.0.so.0
#13 0x000055e97bf25f96 in mainloop_run ()
#14 0x000055e97bf26418 in mainloop_run_with_signal ()
#15 0x000055e97be7aac0 in main ()
```

Meister1593, 2021-04-01:
This issue is reproduced even with different adapter and different headphones

Meister1593, 2021-04-02:
It seems like this issue introduced in 5.56 bluez, in 5.55 i don't experience such segfaults (at first glance)

fbellet, 2021-04-20:
I see the same problem, when running ``mpris-proxy``. The problem is that function ``avrcp_player_event()`` is called by ``notify_addressed_player_changed()`` _always_ with null data arg, while some events (``AVRCP_EVENT_STATUS_CHANGED``,``AVRCP_EVENT_TRACK_CHANGED``) require non-null data.

Meister1593, 2021-04-20:
> I see the same problem, when running `mpris-proxy`. The problem is that function `avrcp_player_event()` is called by `notify_addressed_player_changed()` _always_ with null data arg, while some events (`AVRCP_EVENT_STATUS_CHANGED`,`AVRCP_EVENT_TRACK_CHANGED`) require non-null data.

I run mpris-proxy always, so this might be the explanation for segfaults

closed 2025-08-28 by github-actions[bot]