bluez/bluez #117: Can't use bluetooth

zouyonghe, 2021-03-25:
I'm using arch rolling and i can't start my bluetooth.
Here is the sysinfo and log:

lspci
00:00.0 Host bridge: Intel Corporation Xeon E3-1200 v6/7th Gen Core Processor Host Bridge/DRAM Registers (rev 08)
00:02.0 VGA compatible controller: Intel Corporation UHD Graphics 620 (rev 07)
00:04.0 Signal processing controller: Intel Corporation Xeon E3-1200 v5/E3-1500 v5/6th Gen Core Processor Thermal Subsystem (rev 08)
00:14.0 USB controller: Intel Corporation Sunrise Point-LP USB 3.0 xHCI Controller (rev 21)
00:14.2 Signal processing controller: Intel Corporation Sunrise Point-LP Thermal subsystem (rev 21)
00:15.0 Signal processing controller: Intel Corporation Sunrise Point-LP Serial IO I2C Controller #0 (rev 21)
00:15.1 Signal processing controller: Intel Corporation Sunrise Point-LP Serial IO I2C Controller #1 (rev 21)
00:16.0 Communication controller: Intel Corporation Sunrise Point-LP CSME HECI #1 (rev 21)
00:17.0 SATA controller: Intel Corporation Sunrise Point-LP SATA Controller [AHCI mode] (rev 21)
00:1c.0 PCI bridge: Intel Corporation Sunrise Point-LP PCI Express Root Port #1 (rev f1)
00:1c.4 PCI bridge: Intel Corporation Sunrise Point-LP PCI Express Root Port #5 (rev f1)
00:1c.5 PCI bridge: Intel Corporation Sunrise Point-LP PCI Express Root Port #6 (rev f1)
00:1d.0 PCI bridge: Intel Corporation Sunrise Point-LP PCI Express Root Port #9 (rev f1)
00:1d.3 PCI bridge: Intel Corporation Device 9d1b (rev f1)
00:1f.0 ISA bridge: Intel Corporation Sunrise Point LPC Controller/eSPI Controller (rev 21)
00:1f.2 Memory controller: Intel Corporation Sunrise Point-LP PMC (rev 21)
00:1f.3 Audio device: Intel Corporation Sunrise Point-LP HD Audio (rev 21)
00:1f.4 SMBus: Intel Corporation Sunrise Point-LP SMBus (rev 21)
01:00.0 3D controller: NVIDIA Corporation GP108M [GeForce MX150] (rev ff)
02:00.0 Ethernet controller: Realtek Semiconductor Co., Ltd. RTL8111/8168/8411 PCI Express Gigabit Ethernet Controller (rev 15)
03:00.0 Network controller: Intel Corporation Wireless 8265 / 8275 (rev 78)
04:00.0 Non-Volatile memory controller: Intel Corporation SSD 600P Series (rev 03)
05:00.0 Unassigned class [ff00]: Realtek Semiconductor Co., Ltd. RTS522A PCI Express Card Reader (rev 01)

lsusb
Bus 002 Device 002: ID 05e3:0626 Genesys Logic, Inc. USB3.1 Hub
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 007: ID 138a:003f Validity Sensors, Inc. VFS495 Fingerprint Reader
Bus 001 Device 006: ID 8087:0a2b Intel Corp. Bluetooth wireless interface
Bus 001 Device 005: ID 05c8:03b1 Cheng Uei Precision Industry Co., Ltd (Foxlink) Webcam
Bus 001 Device 003: ID 1a2c:9df4 China Resource Semico Co., Ltd USB Gaming Keyboard 
Bus 001 Device 004: ID 1ea7:0064 SHARKOON Technologies GmbH 2.4GHz Wireless rechargeable vertical mouse [More&Better]
Bus 001 Device 002: ID 05e3:0610 Genesys Logic, Inc. Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

sudo systemctl status bluetooth.service 
● bluetooth.service - Bluetooth service
     Loaded: loaded (/usr/lib/systemd/system/bluetooth.service; enabled; vendor preset: disabled)
     Active: active (running) since Fri 2021-03-26 01:09:54 CST; 19min ago
       Docs: man:bluetoothd(8)
   Main PID: 496 (bluetoothd)
     Status: "Running"
      Tasks: 1 (limit: 19061)
     Memory: 2.6M
     CGroup: /system.slice/bluetooth.service
             └─496 /usr/lib/bluetooth/bluetoothd

3月 26 01:09:54 blackarch systemd[1]: Starting Bluetooth service...
3月 26 01:09:54 blackarch bluetoothd[496]: Bluetooth daemon 5.56
3月 26 01:09:54 blackarch systemd[1]: Started Bluetooth service.
3月 26 01:09:54 blackarch bluetoothd[496]: Starting SDP server
3月 26 01:09:54 blackarch bluetoothd[496]: Bluetooth management interface 1.19 initialized

blueberry
**
ERROR:lib/bluez/adapter.c:165:adapter_get_dbus_object_path: assertion failed: (ADAPTER_IS(self))
timeout: 监视的命令已核心转储

(blueberry.py:10737): com.linuxmint.blueberry-WARNING **: 01:31:25.488: Could not retrieve the BT adapter name with 'bt-adapter -i': Command '['timeout', '2s', 'bt-adapter', '-i']' died with <Signals.SIGABRT: 6>.
**
ERROR:lib/bluez/adapter.c:165:adapter_get_dbus_object_path: assertion failed: (ADAPTER_IS(self))
timeout: 监视的命令已核心转储

(blueberry.py:10737): com.linuxmint.blueberry-WARNING **: 01:31:25.836: Could not retrieve the BT adapter name with 'bt-adapter -i': Command '['timeout', '2s', 'bt-adapter', '-i']' died with <Signals.SIGABRT: 6>.

How can I solve it?

tedd-an, 2021-04-05:
Could you check a few commands like:
$ lsmod | grep btusb
$ hciconfig

Also, if you can share "dmesg" log?

zouyonghe, 2021-04-08:
lsmod | grep btusb                                                                                                                                                                                                                     
btusb                  69632  0
btrtl                  28672  1 btusb
btbcm                  20480  1 btusb
btintel                32768  1 btusb
bluetooth             749568  16 btrtl,btintel,btbcm,bnep,btusb

 hciconfig                                                                                                                                                                                                                               
hci0:	Type: Primary  Bus: USB
	BD Address: 88:B1:11:46:21:97  ACL MTU: 1021:4  SCO MTU: 96:6
	UP RUNNING 
	RX bytes:813 acl:0 sco:0 events:43 errors:0
	TX bytes:480 acl:0 sco:0 commands:50 errors:0

The dmesg log is too long.Can you give me some suggestions?