bluez/bluez #120: PS3 controller, bluetoothctl asking for PIN code, need help debugging this

aguaviva, 2021-03-26:
I just downloaded and compiled Bluez 5.56, this is what happens when I try to pair the controller

```
$ bluetoothctl 
Agent registered
[CHG] Controller 5C:AC:4C:CA:C5:FF Pairable: yes

**** I press the PS button, the 4 leds start blinking, bluez connects to the PS3 controller ****

[CHG] Device 00:26:5C:BC:07:01 Connected: yes      
[PLAYSTATION(R)3 Controller]# info
Device 00:26:5C:BC:07:01 (public)
        Name: PLAYSTATION(R)3 Controller
        Alias: PLAYSTATION(R)3 Controller
        Class: 0x00000540
        Icon: input-keyboard
        Paired: no
        Trusted: yes
        Blocked: no
        Connected: yes
        LegacyPairing: no
[PLAYSTATION(R)3 Controller]# pair
Attempting to pair with 00:26:5C:BC:07:01
[agent] PIN code: 591731
Failed to pair: org.bluez.Error.AuthenticationFailed
[CHG] Device 00:26:5C:BC:07:01 Connected: no
[bluetooth]# 
```
Why is it asking for a PIN? Any idea on how to debug/fix this?  Thanks in advance.

Vudentz, 2021-03-26:
```
        Paired: no
[agent] PIN code: 591731   <----------------------------------- This shouldn't happen, it should just pair it!
```

Looks like you are attempting to pair and it is requesting a PIN code confirmation, doesn't seem there is anything wrong with that exept if you expect it to use just works pairing but by default bluetoothctl does register with "" which correspond to KeyboardDisplay:

https://git.kernel.org/pub/scm/bluetooth/bluez.git/tree/doc/agent-api.txt#n33

If you don't want do input anything then you shall use "NoInputNoOutput".

aguaviva, 2021-03-26:
Thanks for the quick response :)

It looks like this helped! But still not there

`[bluetooth]# agent off
Agent unregistered
[CHG] Controller 5C:AC:4C:CA:C5:FF Pairable: no
[bluetooth]# agent NoInputNoOutput
Agent registered
[CHG] Controller 5C:AC:4C:CA:C5:FF Pairable: yes
[CHG] Device 00:26:5C:BC:07:01 Name: Sony PLAYSTATION(R)3 Controller
[CHG] Device 00:26:5C:BC:07:01 Alias: Sony PLAYSTATION(R)3 Controller
[CHG] Device 00:26:5C:BC:07:01 Modalias: usb:v054Cp0268d0000
[CHG] Device 00:26:5C:BC:07:01 Trusted: no
Authorize service
[agent] Authorize service 00001124-0000-1000-8000-00805f9b34fb (yes/no): yes
[CHG] Device 00:26:5C:BC:07:01 Trusted: yes
[CHG] Device 00:26:5C:BC:07:01 UUIDs: 00001124-0000-1000-8000-00805f9b34fb
[bluetooth]# pair
Missing device address argument
[bluetooth]# info 00:26:5C:BC:07:01 
Device 00:26:5C:BC:07:01 (public)
        Name: Sony PLAYSTATION(R)3 Controller
        Alias: Sony PLAYSTATION(R)3 Controller
        Class: 0x00000540
        Icon: input-keyboard
        Paired: no
        Trusted: yes
        Blocked: no
        Connected: no
        LegacyPairing: no
        UUID: Human Interface Device... (00001124-0000-1000-8000-00805f9b34fb)
        Modalias: usb:v054Cp0268d0000
[bluetooth]# pair 00:26:5C:BC:07:01 
Attempting to pair with 00:26:5C:BC:07:01
Failed to pair: org.bluez.Error.AuthenticationFailed  <------------------- authentication fails
`

bluetoothd output:
`$ sudo bluetoothd -n --plugin=sixaxis
bluetoothd[3167]: Bluetooth daemon 5.53
bluetoothd[3167]: Starting SDP server
bluetoothd[3167]: Ignoring (cli) hostname
bluetoothd[3167]: Ignoring (cli) wiimote
bluetoothd[3167]: Ignoring (cli) autopair
bluetoothd[3167]: Ignoring (cli) policy
bluetoothd[3167]: Ignoring (cli) a2dp
bluetoothd[3167]: Ignoring (cli) avrcp
bluetoothd[3167]: Ignoring (cli) network
bluetoothd[3167]: Ignoring (cli) input
bluetoothd[3167]: Ignoring (cli) hog
bluetoothd[3167]: Ignoring (cli) gap
bluetoothd[3167]: Ignoring (cli) scanparam
bluetoothd[3167]: Ignoring (cli) deviceinfo
bluetoothd[3167]: Ignoring (cli) battery
bluetoothd[3167]: Bluetooth management interface 1.17 initialized
bluetoothd[3167]: sixaxis: compatible device connected: Sony PLAYSTATION(R)3 Controller (054C:0268 /sys/devices/pci0000:00/0000:00:12.0/usb4/4-1/4-1:1.0/0003:054C:0268.0006/hidraw/hidraw0)
bluetoothd[3167]: sixaxis: setting up new device
`

aguaviva, 2021-03-27:
1) start bluetoothctl 

```
$ bluetoothctl
[CHG] Controller 5C:AC:4C:CA:C5:FF Class: 0x00200000
[bluetoothctl]# power on
Changing power on succeeded
[bluetoothctl]# pairable on
Changing pairable on succeeded
[bluetoothctl]# discoverable on
Changing discoverable on succeeded
[bluetoothctl]# agent off
No agent is registered
[bluetoothctl]# agent NoInputNoOutput
Agent registered
[CHG] Controller 5C:AC:4C:CA:C5:FF Pairable: yes
[bluetoothctl]#
```
bluetooth output:
```
bluetoothd[1685]: src/adapter.c:property_set_mode() sending Set Powered command for index 0
bluetoothd[1685]: src/adapter.c:dev_class_changed_callback() Class: 0x200000
bluetoothd[1685]: src/adapter.c:property_set_mode_complete() Success (0x00)
bluetoothd[1685]: src/adapter.c:new_settings_callback() Settings: 0x000000d1
bluetoothd[1685]: src/adapter.c:settings_changed() Changed settings: 0x00000001
bluetoothd[1685]: src/adapter.c:settings_changed() Pending settings: 0x00000000
bluetoothd[1685]: src/adapter.c:adapter_start() adapter /org/bluez/hci0 has been enabled
bluetoothd[1685]: src/adapter.c:set_mode() sending set mode command for index 0
bluetoothd[1685]: src/adapter.c:property_set_mode() sending Set Discoverable command for index 0
bluetoothd[1685]: src/adapter.c:new_settings_callback() Settings: 0x000000d3
bluetoothd[1685]: src/adapter.c:settings_changed() Changed settings: 0x00000002
bluetoothd[1685]: src/adapter.c:settings_changed() Pending settings: 0x00000008
bluetoothd[1685]: src/adapter.c:property_set_mode_complete() Success (0x00)
bluetoothd[1685]: src/adapter.c:new_settings_callback() Settings: 0x000000db
bluetoothd[1685]: src/adapter.c:settings_changed() Changed settings: 0x00000008
bluetoothd[1685]: src/adapter.c:settings_changed() Pending settings: 0x00000000
bluetoothd[1685]: src/agent.c:add_default_agent() Default agent set to :1.108 /org/bluez/agent
bluetoothd[1685]: src/agent.c:agent_ref() 0x55a1e06c5dc0: ref=1
bluetoothd[1685]: src/agent.c:register_agent() agent :1.108
bluetoothd[1685]: src/agent.c:agent_disconnect() Agent :1.108 disconnected
bluetoothd[1685]: src/agent.c:remove_default_agent() Default agent cleared
bluetoothd[1685]: src/adapter.c:set_mode() sending set mode command for index 0
bluetoothd[1685]: src/agent.c:agent_destroy() agent :1.108
bluetoothd[1685]: src/agent.c:agent_unref() 0x55a1e06c5dc0: ref=0
bluetoothd[1685]: src/adapter.c:new_settings_callback() Settings: 0x000000cb
bluetoothd[1685]: src/adapter.c:settings_changed() Changed settings: 0x00000010
bluetoothd[1685]: src/adapter.c:settings_changed() Pending settings: 0x00000000
bluetoothd[1685]: src/agent.c:add_default_agent() Default agent set to :1.109 /org/bluez/agent
bluetoothd[1685]: src/adapter.c:set_mode() sending set mode command for index 0
bluetoothd[1685]: src/agent.c:agent_ref() 0x55a1e06c8680: ref=1
bluetoothd[1685]: src/agent.c:register_agent() agent :1.109
bluetoothd[1685]: src/adapter.c:new_settings_callback() Settings: 0x000000db
bluetoothd[1685]: src/adapter.c:settings_changed() Changed settings: 0x00000010
bluetoothd[1685]: src/adapter.c:settings_changed() Pending settings: 0x00000000
```
2) press PS in the PS3 controller
```
[CHG] Device 00:26:5C:BC:07:01 Connected: yes
[Sony PLAYSTATION(R)3 Controller]# 
```

bluetoothd output:
```
bluetoothd[1685]: src/adapter.c:connected_callback() hci0 device 00:26:5C:BC:07:01 connected eir_len 5
```
3) issue the pair command

```
[Sony PLAYSTATION(R)3 Controller]# pair
Attempting to pair with 00:26:5C:BC:07:01
Request PIN code
[agent] Enter PIN code: 
```

bluetoothd output:
```
bluetoothd[1876]: src/agent.c:agent_ref() 0x55836971d880: ref=2
bluetoothd[1876]: src/device.c:bonding_request_new() Requesting bonding for 00:26:5C:BC:07:01
bluetoothd[1876]: src/agent.c:agent_ref() 0x55836971d880: ref=3
bluetoothd[1876]: src/agent.c:agent_unref() 0x55836971d880: ref=2
bluetoothd[1876]: src/adapter.c:suspend_discovery() 
bluetoothd[1876]: src/adapter.c:adapter_bonding_attempt() hci0 bdaddr 00:26:5C:BC:07:01 type 0 io_cap 0x04
bluetoothd[1876]: src/adapter.c:pin_code_request_callback() hci0 00:26:5C:BC:07:01
bluetoothd[1876]: src/device.c:new_auth() Requesting agent authentication for 00:26:5C:BC:07:01
bluetoothd[1876]: src/agent.c:agent_ref() 0x55836971d880: ref=3
```
4) typed 000000, and also tried with random numbers

bluetoothd output:
```
bluetoothd[1898]: src/agent.c:agent_ref() 0x562e03a30a40: ref=4
bluetoothd[1898]: src/adapter.c:btd_adapter_pincode_reply() hci0 addr 00:26:5C:BC:07:01 pinlen 5
bluetoothd[1898]: src/agent.c:agent_unref() 0x562e03a30a40: ref=3
bluetoothd[1898]: src/agent.c:agent_unref() 0x562e03a30a40: ref=2
bluetoothd[1898]: src/adapter.c:pair_device_complete() Authentication Failed (0x05)
bluetoothd[1898]: src/adapter.c:bonding_attempt_complete() hci0 bdaddr 00:26:5C:BC:07:01 type 0 status 0x5
bluetoothd[1898]: src/device.c:device_bonding_complete() bonding 0x562e03a2ab50 status 0x05
bluetoothd[1898]: src/device.c:device_cancel_authentication() Canceling authentication request for 00:26:5C:BC:07:01
bluetoothd[1898]: src/device.c:device_bonding_failed() status 5
bluetoothd[1898]: src/agent.c:agent_unref() 0x562e03a30a40: ref=1
bluetoothd[1898]: src/adapter.c:resume_discovery() 
bluetoothd[1898]: src/adapter.c:dev_disconnected() Device 00:26:5C:BC:07:01 disconnected, reason 0
bluetoothd[1898]: src/adapter.c:adapter_remove_connection() 
bluetoothd[1898]: src/adapter.c:bonding_attempt_complete() hci0 bdaddr 00:26:5C:BC:07:01 type 0 status 0xe
bluetoothd[1898]: src/device.c:device_bonding_complete() bonding (nil) status 0x0e
bluetoothd[1898]: src/device.c:device_bonding_failed() status 14
bluetoothd[1898]: src/adapter.c:resume_discovery() 
```

aguaviva, 2021-03-30:
Updates: 
1- After taking apart the controller I noticed it is a chinese clone. 
2- I tried talking to the controller using a esp32 microcontroller, and it seems to return an error. So all points out that the controller might be misbehaving, but this is still WIP.

Vudentz, 2021-03-31:
@aguaviva if it is trying to emulate a keyboad maybe you have to enter something on it and then repeat it on the BlueZ, though with NoInputNoOutput that should have never triggered the PIN code method, try checking what is its IO capabilities using btmon.

aguaviva, 2021-03-31:
It is a ps3 controller so it is not trying to emulate a keyboard. You can look at the sixaxis plugin for more details.  Anyway, could you point me to some documentarion/details on how to do what you suggest?

dlech, 2021-04-02:
PS3 controllers are weird. You don't actually "pair" them like most Bluetooth devices. Instead after you press and hold the button to initiate the new Bluetooth connection, you have to plug in the USB cable and things magically happen.

aguaviva, 2021-04-02:
Actually it is not that magical, the USB cable is just to set in the controller the bt address of your "console", once it has it, when you press "PS" the controller will initiate the connection and the "console" will accept it. That's it.

dlech, 2021-04-02:
Right. I was trying to explain that BlueZ works the same way.

aguaviva, 2021-04-09:
I foudn the issue. My controller turned out to be a GASIA clone.

While real Sony controllers need the following sequence start streaming data:

> 0x53, 0xF4, 0x42, 0x03, 0x00, 0x00 

Gasia ones need the following:

> 0xA2,  0x01,
> 0x00, 0x00, 0x00, 0x00, 0x00,   // rumble values [0x00, right-timeout, right-force, left-timeout, left-force]
> 0x00, 0x00, 0x00, 0x00, 0x02,   // 0x02=LED1 .. 0x10=LED4
> 0xff, 0x27, 0x10, 0x00, 0x32,   // LED 4
> 0xff, 0x27, 0x10, 0x00, 0x32,   // LED 3
> 0xff, 0x27, 0x10, 0x00, 0x32,   // LED 2
> 0xff, 0x27, 0x10, 0x00, 0x32,   // LED 1
> 0x00, 0x00, 0x00, 0x00, 0x00,
> 0x00, 0x00, 0x00, 0x00, 0x00,
> 0x00, 0x00, 0x00, 0x00, 0x00,
> 0x00, 0x00, 0x00
> 

That is why some people can get the control to connect but then they wont receive any data, for example jstest will jst print a few bogus values (16 or so) and will then it will stop updating.

For more info see:
https://github.com/falkTX/qtsixa/blob/c2f4ed9ee5646aefa68fb44fa14eda0460f02e66/sixad/shared.cpp#L272

closed 2025-08-28 by github-actions[bot]