bluez/bluez #121: missing bccmd since 5.56

jaccc, 2021-03-29:
can't set/get Default transmit power

bccmd psget -s 0x0000 0x0021
Default transmit power: 0x0014 (20)

Vudentz, 2021-03-31:
CSR specifics tools were remove by:

https://git.kernel.org/pub/scm/bluetooth/bluez.git/commit/?id=e2e198ced468122df7b431cb06158123ba60a2f2

This like transmission power my actually be something that needs to be done in the driver and might even consider some control over userspace, in fact there is some control over the TX Power for advertising:

https://git.kernel.org/pub/scm/bluetooth/bluez.git/tree/doc/advertising-api.txt#n157

closed 2021-03-31 by Vudentz

cobrafast, 2021-06-22:
To anyone like me, finding this issue wanting `bccmd` (in my case to be able to change transmit power), you have to manually compile and install bluez tag 5.55 with ell tag 0.35 in a parallel directory as described in readme.

```bash
./bootstrap
./configure --prefix=/usr --mandir=/usr/share/man --sysconfdir=/etc --localstatedir=/var --enable-btpclient --enable-midi --enable-sixaxis --enable-hid2hci --enable-mesh --enable-experimental --enable-library --enable-deprecated
make
sudo make install-binPROGRAMS install-dist_zshcompletionDATA install-man1
```

this should leave some other bluez stuff from your distro's package untouched.

ony, 2023-02-05:
You don't need to install whole thing. You can build it against recent bluez.
```sh
git clone https://github.com/bluez/bluez.git -b 5.55
cd bluez
gcc tools/{bccmd,csr*,ubcsp}.c -o tools/bccmd -I. $(pkg-config --cflags --libs bluez)
```
*(if you are on ArchLinux, you can use [`bluez-bccmd` from AUR](https://aur.archlinux.org/packages/bluez-bccmd))*

In my case `bccmd` needed to flip some flag in persisted storage of adapter to get Bluetooth working via `bccmd psset -r 0x028c 0x0001` as just power on doesn't work. See:
* https://answers.launchpad.net/ubuntu/+question/135094
* https://bugs.launchpad.net/ubuntu/+source/bluez/+bug/781556
* https://techienotes.blog/2014/06/26/how-to-fix-bluetooth-problem-on-hp-laptops-with-ubuntu-or-similar/