bluez/bluez #122: Write documentation on how to use btmon

aguaviva, 2021-03-31:
btmon is a tool that seems to help troubleshoot issues. I have seen a few maintainers ask casual developers to use it. However the documentation is pretty scarce, a thing that is a bit shocking and this is a barrier for many people that would like to participate and improve the repository.

Vudentz, 2021-03-31:
I guess you want something like a hcidump manpage:

https://linux.die.net/man/8/hcidump

Note that btmon --hep giver a lot of this information:

```
btmon - Bluetooth monitor
Usage:
	btmon [options]
options:
	-r, --read <file>      Read traces in btsnoop format
	-w, --write <file>     Save traces in btsnoop format
	-a, --analyze <file>   Analyze traces in btsnoop format
	-s, --server <socket>  Start monitor server socket
	-p, --priority <level> Show only priority or lower
	-i, --index <num>      Show only specified controller
	-d, --tty <tty>        Read data from TTY
	-B, --tty-speed <rate> Set TTY speed (default 115200)
	-V, --vendor <compid>  Set default company identifier
	-M, --mgmt             Open channel for mgmt events
	-t, --time             Show time instead of time offset
	-T, --date             Show time and date information
	-S, --sco              Dump SCO traffic
	-A, --a2dp             Dump A2DP stream traffic
	-E, --ellisys [ip]     Send Ellisys HCI Injection
	-P, --no-pager         Disable pager usage
	-J  --jlink <device>,[<serialno>],[<interface>],[<speed>]
	                       Read data from RTT
	-R  --rtt [<address>],[<area>],[<name>]
	                       RTT control block parameters
	-h, --help             Show help options
```

aguaviva, 2021-03-31:
I already knew that, and that is not documentation.  

There are a couple of bugs were you ask people to use btmon, it wouldn't hurt if you could add a bit of details explaining what to do exactly with it and what to look for, after all we are trying to help out.

tedd-an, 2021-04-15:
I created a wiki page for btmon to start: https://github.com/bluez/bluez/wiki/btmon
We can add more content as we move on.

Vudentz, 2021-05-05:
https://github.com/bluez/bluez/blob/master/monitor/btmon.rst

closed 2021-05-05 by Vudentz