bluez/bluez #123: Xbox Elite 2 controller stops sending data during inquiries from nearby devices

kakra, 2021-03-31:
Also posted to the mailing list but no replies received: https://lore.kernel.org/linux-bluetooth/CAC2ZOYuG_BBG0uisXkoTtroxwgvv+JO-yMFC2+bRQZjsvwK1qw@mail.gmail.com/T/#u

I'm the maintainer of the xpadneo driver for Xbox controllers in Bluetooth mode. I'm currently testing two different models of the
controllers:

1. Xbox One S (model 1708)
2. Xbox Elite Series 2 (model 1797)

With kernel commit 98d2c3e17310 ("Bluetooth: L2CAP: Try harder to accept device not knowing options") we can finally connect the controllers without disabling ERTM. Without the patch, all the controllers will fail to connect most of the time if not always.

Here's what works since ERTM can stay enabled:

With ERTM enabled, the controllers no longer crash in their firmware when sending rumble packets at specific varying rates. Previously, the controller tended to stop sending back input data, eventually rebooting after some time (but not always).

Some history / context about the problems (with ERTM disabled):

That looks like a crash or race condition in the firmware. The results looked pretty random but consistently between kernel or Bluez versions so it probably depends on some implementation detail: Sometimes the controller would stop sending any input data, sometimes at least one axis still worked, sometimes rumble would still work, or any combination of those. This problem does not seem to exist when using the Android bluetooth stack (fluoride?) - tho I didn't really test it there extensively. At least in Android, the controllers establish the Bluetooth connection instantly while in Linux, it may take up to 30 seconds to establish the connection / pairing. Often, it would not exchange the link key, resulting in pairing failing (but still able to connect manually with `bluetoothctl connect MAC`). But finally, this seems fixed when we enable ERTM after the patch referenced above is applied.

Now let's get to the core of this report (after applying the above referenced patch, and enabling ERTM):

From all what I could test, model 1708 works correctly now with some quirks: It may take some time / effort pairing the controller. Usually, it works fine using the KDE GUI client for pairing while it doesn't using `bluetoothctl` alone. I'm not sure why that is but once it's paired (with a link key in `/var/lib/bluetooth`, it reconnects fine in later attempts, tho it may take 20-30s to do that. After this, the gamepad runs just fine in games, rumble works flawlessly, no input lag, no loss of input data. Good work.

Model 1797 is a different story: This model now connects instantly, this is a success. Previously, it would show the same behavior as model 1708, often taking 20-30s to connect, or failing every once in a while, often even losing the connection again after being connected for a few seconds.

But now I'm seeing the following odd behavior: When running `btmon`, I see a constant flow of input data while moving the sticks in a circle (thus generating a constant stream of input events). Now, when a nearby device starts some inquiry, the gamepad stops sending any data until some time after the inquiry is done. Some of the last events then get sent back resembling the current state of the device as if the gamepad flushed some buffers. Often, that's exactly three input state buffers if you put the gamepad idle on your desk while the inquiry is running. This observation is repeatable. Also, during gaming, I may have seen issues where input events of the device may lag behind several seconds, however, I'm not sure if this was caused by a nearby device as I didn't log any data with `btmon` (neither I could associate that on a timeline with lagging events in the game).

Here's a (rather lengthy) btmon log of such an incident (packets starting with `a1 01` are input event packets from the controller, part of the constant stream of input data I generated using the sticks):
```
# kernel 5.10.15
# bluez 5.55
# dongle 00:1A:7D:DA:71:15 (Cambridge Silicon Radio), USB ID 0a12:0001
# also tested with kernel 5.10.26, bluez 5.56 (not this log)
... [multiple seconds of big amounts of input data before here]
> ACL Data RX: Handle 74 flags 0x02 dlen 44              #1018 [hci0] 14.917844
      Channel: 65 len 40 [PSM 0 mode Basic (0x00)] {chan 65535}
        a1 01 e0 81 ea 79 11 8e ff ff 00 00 00 00 00 00  .....y..........
        00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
        00 00 00 00 00 00 00 00                          ........
> ACL Data RX: Handle 74 flags 0x02 dlen 44              #1019 [hci0] 14.927824
      Channel: 65 len 40 [PSM 0 mode Basic (0x00)] {chan 65535}
        a1 01 e0 81 ea 79 6c 6b ff ff 00 00 00 00 00 00  .....ylk........
        00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
        00 00 00 00 00 00 00 00                          ........
> ACL Data RX: Handle 74 flags 0x02 dlen 44              #1020 [hci0] 14.937827
      Channel: 65 len 40 [PSM 0 mode Basic (0x00)] {chan 65535}
        a1 01 e0 81 ea 79 e5 67 ff ff 00 00 00 00 00 00  .....y.g........
        00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
        00 00 00 00 00 00 00 00                          ........
@ MGMT Command: Start Service Disc.. (0x003a) plen 4  {0x0001} [hci0] 14.946337
        Address type: 0x01
          BR/EDR
        RSSI: invalid (0x7f)
        UUIDs: 0
< HCI Command: Inquiry (0x01|0x0001) plen 5              #1021 [hci0] 14.946384
        Access code: 0x9e8b33 (General Inquiry)
        Length: 10.24s (0x08)
        Num responses: 0
> HCI Event: Command Status (0x0f) plen 4                #1022 [hci0] 14.947699
      Inquiry (0x01|0x0001) ncmd 1
        Status: Success (0x00)
@ MGMT Event: Command Complete (0x0001) plen 4        {0x0001} [hci0] 14.947758
      Start Service Discovery (0x003a) plen 1
        Status: Success (0x00)
        Address type: 0x01
          BR/EDR
@ MGMT Event: Discovering (0x0013) plen 2             {0x0001} [hci0] 14.947764
        Address type: 0x01
          BR/EDR
        Discovery: Enabled (0x01)
> ACL Data RX: Handle 74 flags 0x02 dlen 44              #1023 [hci0] 14.947929
[... 686 lines cut ...]
> ACL Data RX: Handle 74 flags 0x02 dlen 44              #1048 [hci0] 25.199137
      Channel: 65 len 40 [PSM 0 mode Basic (0x00)] {chan 65535}
        a1 01 e0 81 ea 79 6b 81 f6 7f 00 00 00 00 00 00  .....yk.........
        00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
        00 00 00 00 00 00 00 00                          ........
> ACL Data RX: Handle 74 flags 0x02 dlen 44              #1049 [hci0] 25.207888
      Channel: 65 len 40 [PSM 0 mode Basic (0x00)] {chan 65535}
        a1 01 e0 81 ea 79 6b 81 f6 7f 00 00 00 00 00 00  .....yk.........
        00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
        00 00 00 00 00 00 00 00                          ........
> ACL Data RX: Handle 74 flags 0x02 dlen 44              #1050 [hci0] 25.217897
      Channel: 65 len 40 [PSM 0 mode Basic (0x00)] {chan 65535}
        a1 01 e0 81 ea 79 6b 81 f6 7f 00 00 00 00 00 00  .....yk.........
        00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
        00 00 00 00 00 00 00 00                          ........
```

As you can see, input data is received roughly around every 0.01s (which seems to be some internal heartbeat of the firmware, around 100 Hz, which matches with the maximum rate of rumble programming we may send to the controller without crashing it). But when the C4:57:6E device starts to show up in the log, the controller pauses sending any more data for multiple seconds. This behavior cannot be observed when connected to an Android device so we probably cannot say that this is a bug of the gamepad firmware (tho, it still has a lot of quirks I'd consider a bug).

Windows users may report similar problems, tho there's no indication of it being the same problem as there are no logs provided: https://answers.microsoft.com/de-de/xbox/forum/xba_wowpc-xba_gowpc/xbox-elite-series-2-disconnecting-from-pc/7e5a964a-16cc-4e83-8dec-aa8c858b875c

I've linked some more reports in the xpadneo project:
https://github.com/atar-axis/xpadneo/issues/198

I'd happily provide more data / details to debug this.

Vudentz, 2021-03-31:
But is it intentional to continue scanning/inquiring while connected to the controller? That sounds very odd and I guess most controller will observe some disruption in the connection given that it normally priority the scanning/inquiry over the connection, Im afraid the only solution to this would be to set so sort of QoS for the connection but that would require kernel changes.

kakra, 2021-03-31:
No it's not intentional. I was debugging an issue why the gamepad stopped receiving inputs every once in a while, and looking at evdev the input events didn't even seem to get into the kernel driver, so I launched `btmon` to go a level deeper and found the log I posted above. And I wondered why some neighbor devices which seem to continuously scanner the Bluetooth perimeter prevent my device from working properly.

Is this a setup issue with my local bluez installation? Or something the KDE Plasma GUI does? I wouldn't expect neighbor device I have no control over preventing my own paired devices from working properly.

kakra, 2021-03-31:
There are seemingly 6 or 7 Samsung TVs around me that constantly seem to look for devices to connect to, I even see pairing requests from time to time and I don't think that was started intentionally by the owners of such devices. It looks more like some aggressive algorithm to more easily connect to nearby devices for the first time setup, only that nobody ever cares to pair their TV with any other of their devices. I usually turn Bluetooth off in my own TV, I don't know what's that useful for, it just puts more stress on the RF environment.

kakra, 2021-03-31:
BTW: I just tried again, it has even become worse. Now there are not only 6 or 7 TVs around but there also home theatre audio equipment around, and the gamepad has almost no chance to ever send input data. Pairing it to my Android device and it works just fine. My other Xbox controller (the older model) also works mostly fine, with some occasional connection losses and automatic reconnects. It seems that something about the Bluetooth protocol should be working differently for Xbox controllers, and that it works fine when paired with Android.

Vudentz, 2021-03-31:
@kakra Make sure you don't have any rogue application scanning in the meantime, if you see inquiry result that tells you the adapter is discovering, or perhaps you have the bluetooth settings open all time scanning, that for sure will cause some of the problems you are seeing.

kakra, 2021-03-31:
I don't know what's causing it. I disabled the KDE Bluetooth background service, unloaded the bluetooth modules from pulseaudio, stopped Steam (as this seems to support Bluetooth controllers and may scan for them), restarted bluez, it's still showing this constant stream of discoveries (tested at each step), the Elite 2 controller doesn't work properly at all, the Xbox One S model mostly works, it takes around 20-30s to connect those devices. Pairing the Xbox controllers with my phone and they will immediately connect, no flaws, input works.

I then paired my phone to my PC and streamed audio via Bluetooth. Interestingly this seems to suppress the inquiries mostly and the Xbox Elite 2 controller works somewhat better but still loses a lot of input events.

But TBH, whatever is causing this, it's probably doing it for a lot of users then because my setup is mostly default for Bluetooth. I just changed `ReconnectIntervals=1,1,2,3,5,8,13,21,34,55` in an effort to reconnect the gamepads faster after connection lost. But setting this back to defaults doesn't change anything about this issue.

I believe that the Bluetooth implementation of the Xbox controllers isn't without quirks and flaws but the kernel and bluez should still manage to handle that just fine as, e.g., Windows, Android, and iPhones handle the controllers just fine. Looking at the xpadneo issues, getting the controllers to work properly is a big mess (aka pita). And stability seems to change for no apparent reason, we cannot even pinpoint that down to specific kernel changes or bluez updates except the L2CAP patch helps a lot because it allows running with ERTM enabled which in turn seems to prevent the Xbox controller firmware from glitching when we send rumble commands. You cannot even pair the controllers with ERTM enabled (tho, that seems to be a bit driver and adapter specific) unless the kernel has the L2CAP patch.

Vudentz, 2021-04-02:
@kakra So the adapter is always discovering? Even after you had restarted the service, etc? Does bluetoothctl reporting Discovering: yes when you use show command?

kakra, 2021-04-02:
It's changing like every 10s:
```
bluetoothctl
Agent registered
[Xbox Wireless Controller]# show
Controller 00:1A:7D:DA:71:15 (public)
        Name: jupiter
        Alias: jupiter
        Class: 0x001c0104
        Powered: yes
        Discoverable: no
        DiscoverableTimeout: 0x00000000
        Pairable: no
        UUID: Message Notification Se.. (00001133-0000-1000-8000-00805f9b34fb)
        UUID: A/V Remote Control        (0000110e-0000-1000-8000-00805f9b34fb)
        UUID: OBEX Object Push          (00001105-0000-1000-8000-00805f9b34fb)
        UUID: Message Access Server     (00001132-0000-1000-8000-00805f9b34fb)
        UUID: PnP Information           (00001200-0000-1000-8000-00805f9b34fb)
        UUID: IrMC Sync                 (00001104-0000-1000-8000-00805f9b34fb)
        UUID: Vendor specific           (00005005-0000-1000-8000-0002ee000001)
        UUID: Headset AG                (00001112-0000-1000-8000-00805f9b34fb)
        UUID: A/V Remote Control Target (0000110c-0000-1000-8000-00805f9b34fb)
        UUID: Generic Attribute Profile (00001801-0000-1000-8000-00805f9b34fb)
        UUID: Phonebook Access Server   (0000112f-0000-1000-8000-00805f9b34fb)
        UUID: Audio Sink                (0000110b-0000-1000-8000-00805f9b34fb)
        UUID: Device Information        (0000180a-0000-1000-8000-00805f9b34fb)
        UUID: Generic Access Profile    (00001800-0000-1000-8000-00805f9b34fb)
        UUID: Audio Source              (0000110a-0000-1000-8000-00805f9b34fb)
        UUID: Vendor specific           (185f3df4-3268-4e3f-9fca-d4d5059915bd)
        UUID: OBEX File Transfer        (00001106-0000-1000-8000-00805f9b34fb)
        Modalias: usb:v1D6Bp0246d0538
        Discovering: yes
        Roles: central
        Roles: peripheral
Advertising Features:
        ActiveInstances: 0x00 (0)
        SupportedInstances: 0x05 (5)
        SupportedIncludes: tx-power
        SupportedIncludes: appearance
        SupportedIncludes: local-name
[CHG] Controller 00:1A:7D:DA:71:15 Discovering: no
[CHG] Controller 00:1A:7D:DA:71:15 Discovering: yes
[CHG] Controller 00:1A:7D:DA:71:15 Discovering: no
[CHG] Controller 00:1A:7D:DA:71:15 Discovering: yes
[CHG] Controller 00:1A:7D:DA:71:15 Discovering: no
[CHG] Controller 00:1A:7D:DA:71:15 Discovering: yes
```

I have no idea what's causing it.

kakra, 2021-04-06:
This may be related: https://github.com/ValveSoftware/Proton/issues/4571

kakra, 2021-04-06:
So, I cherry-picked some commits from bluetooth-next, and now the behavior is different:

Even during inquiries, I see input events now with the Xbox Elite 2 controller. But it always sends reports now at 50 Hz, no matter if I touch controls or don't. Sending rumble data to the controller no longer works. Input reports may still be lost because often the repeatedly sent report is stuck to the same values. It looks like the Xbox controller firmware tries to flush buffers but doesn't remove the packets from the buffer. Maybe some sort of ACK is missing?

Watching `btmon`, and if I then generate some input data by moving the stick, I'll see a few new report data values, then the controller stops sending reports until I stop generating input data.

Here's the commit list: https://github.com/torvalds/linux/compare/v5.10...kakra:bluetooth/cherry-picks?expand=1

Vudentz, 2021-04-06:
@kakra You probably need to check why there is a discovery ongoing, controller normally prioritize the inquiry over the connection, also it is not normal to have a scanning/inquiry active all the time, if there is something is malfunctioning or perhaps you just forgot to disable the scannig (bluetoothctl> scan off)

kakra, 2021-04-06:
I don't think inquiry itself is the problem because after cherry-picking the patches, the Xbox controller stops sending data while I generate input data, while previously, it stopped sending data while an inquiry was running. With the current patches, I constantly receive data from the Xbox controller no matter if an inquiry is running.

kakra, 2021-04-06:
Killing the complete graphical shell seems to stop the "Discovering: on" messages in bluetoothctl, at btmon remains mostly silent. The Xbox controller is still difficult to connect (but that's another issue) but the Xbox Elite 2 model now sends data reliably.

The other model (Xbox One S controller) seems unaffected by the discovering mode - it just works and sends data as it should.

I'm not sure where the responsibility is for how Bluetooth traffic should be prioritized. You clearly state that the kernel doesn't do it, and the Xbox One S model seems unaffected by this. So it should probably be solved in the device itself, and the Xbox Elite 2 controller just has a bad implementation. Let's ignore that for a while.

Next step is probably finding which process/client causes the discovering: While having bluetoothctl running, and having the graphical shell running (KDE Plasma 5.21), I can see in bluetoothctl that "Discovering" is constantly switching on and off. There are two patterns: Either its ~10s off, then ~10s on, repeating. Or it's ~10s on, then off and immediately back on.

The input event loss of the Xbox controller doesn't match with those time windows. Simply the fact that discovering switches on and off makes it go crazy. Maybe it's not even the discovering itself that causes the problems but something else that's hammering on the Bluetooth stack. I still think that discovering devices should not block the controller from working properly but as written above, let's ignore that for now.

To debug this any further, it would be essential that anything in the Bluetooth stack would log which client or process causes the state change. Is there anything like that?

kakra, 2021-04-06:
> it is not normal to have a scanning/inquiry active all the time, if there is something is malfunctioning or perhaps you just forgot to disable the scannig (bluetoothctl> scan off)

While it is discovering, I'm not able to run "scan off", it says "dbus error". I can run "scan on", tho, and then I could run "scan off". This seems independent from discovering.

I can use "btmgmt stop-find -b" to stop discovering, only to see it flipped back on immediately.

Looking at the source code of bluedevil, it looks like it may be related to kio. But why this is constantly triggered, I have no idea: https://github.com/KDE/bluedevil/blob/bb28472500c1292c9059b06af937ed51b4cb02f8/src/kio/bluetooth/kiobluetooth.cpp#L159

kakra, 2021-04-07:
Reported here: https://bugs.kde.org/show_bug.cgi?id=435444

kakra, 2021-04-12:
We can probably attribute that to kdeconnectd or one of its dependers. I'll leave this open until further investigations have been done with kdeconnectd.

closed 2025-08-28 by github-actions[bot]