bluez/bluez #125: bluez 5.57. build error

joakim-tjernlund, 2021-04-02:
```
./configure --prefix=/usr --build=x86_64-pc-linux-gnu --host=x86_64-pc-linux-gnu --mandir=/usr/share/man --infodir=/usr/share/info --datadir=/usr/share --sysconfdir=/etc --localstatedir=/var/lib --disable-dependency-tracking --disable-silent-rules --docdir=/usr/share/doc/bluez-9999 --htmldir=/usr/share/doc/bluez-9999/html --with-sysroot=/ --libdir=/usr/lib64 --localstatedir=/var --disable-android --enable-datafiles --enable-optimization --disable-debug --enable-pie --enable-threads --enable-library --enable-tools --enable-manpages --enable-monitor --with-systemdsystemunitdir=/lib/systemd/system --with-systemduserunitdir=/usr/lib/systemd/user --disable-btpclient --disable-external-ell --enable-cups --disable-deprecated --disable-experimental --enable-mesh --enable-external-ell --disable-midi --enable-obex --enable-client --disable-systemd --disable-test --enable-udev --enable-hid2hci --enable-sixaxis
configure: loading site script /usr/share/config.site
checking for a BSD-compatible install... /usr/lib/portage/python3.8/ebuild-helpers/xattr/install -c
checking whether build environment is sane... yes
checking for a thread-safe mkdir -p... /bin/mkdir -p
checking for gawk... gawk
checking whether make sets $(MAKE)... yes
checking whether make supports nested variables... yes
checking how to create a pax tar archive... gnutar
checking whether make supports nested variables... (cached) yes
checking whether to enable maintainer-specific portions of Makefiles... no
checking for x86_64-pc-linux-gnu-pkg-config... /usr/bin/x86_64-pc-linux-gnu-pkg-config
checking pkg-config is at least version 0.9.0... yes
checking whether make supports the include directive... yes (GNU style)
checking for x86_64-pc-linux-gnu-gcc... x86_64-pc-linux-gnu-gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables... 
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether we are using the GNU C compiler... yes
checking whether x86_64-pc-linux-gnu-gcc accepts -g... yes
checking for x86_64-pc-linux-gnu-gcc option to accept ISO C89... none needed
checking whether x86_64-pc-linux-gnu-gcc understands -c and -o together... yes
checking dependency style of x86_64-pc-linux-gnu-gcc... none
checking for C/C++ restrict keyword... __restrict
checking for x86_64-pc-linux-gnu-gcc... (cached) x86_64-pc-linux-gnu-gcc
checking whether we are using the GNU C compiler... (cached) yes
checking whether x86_64-pc-linux-gnu-gcc accepts -g... (cached) yes
checking for x86_64-pc-linux-gnu-gcc option to accept ISO C89... (cached) none needed
checking whether x86_64-pc-linux-gnu-gcc understands -c and -o together... (cached) yes
checking dependency style of x86_64-pc-linux-gnu-gcc... (cached) none
checking whether x86_64-pc-linux-gnu-gcc accepts -fPIE... yes
checking build system type... x86_64-pc-linux-gnu
checking host system type... x86_64-pc-linux-gnu
checking how to print strings... printf
checking for a sed that does not truncate output... /bin/sed
checking for grep that handles long lines and -e... /bin/grep
checking for egrep... /bin/grep -E
checking for fgrep... /bin/grep -F
checking for ld used by x86_64-pc-linux-gnu-gcc... /usr/x86_64-pc-linux-gnu/bin/ld
checking if the linker (/usr/x86_64-pc-linux-gnu/bin/ld) is GNU ld... yes
checking for BSD- or MS-compatible name lister (nm)... /usr/bin/x86_64-pc-linux-gnu-nm -B
checking the name lister (/usr/bin/x86_64-pc-linux-gnu-nm -B) interface... BSD nm
checking whether ln -s works... yes
checking the maximum length of command line arguments... 1572864
checking how to convert x86_64-pc-linux-gnu file names to x86_64-pc-linux-gnu format... func_convert_file_noop
checking how to convert x86_64-pc-linux-gnu file names to toolchain format... func_convert_file_noop
checking for /usr/x86_64-pc-linux-gnu/bin/ld option to reload object files... -r
checking for x86_64-pc-linux-gnu-objdump... x86_64-pc-linux-gnu-objdump
checking how to recognize dependent libraries... pass_all
checking for x86_64-pc-linux-gnu-dlltool... no
checking for dlltool... no
checking how to associate runtime and link libraries... printf %s\n
checking for x86_64-pc-linux-gnu-ar... x86_64-pc-linux-gnu-ar
checking for archiver @FILE support... @
checking for x86_64-pc-linux-gnu-strip... x86_64-pc-linux-gnu-strip
checking for x86_64-pc-linux-gnu-ranlib... x86_64-pc-linux-gnu-ranlib
checking command to parse /usr/bin/x86_64-pc-linux-gnu-nm -B output from x86_64-pc-linux-gnu-gcc object... ok
checking for sysroot... /
checking for a working dd... /bin/dd
checking how to truncate binary pipes... /bin/dd bs=4096 count=1
checking for x86_64-pc-linux-gnu-mt... no
checking for mt... no
checking if : is a manifest tool... no
checking how to run the C preprocessor... x86_64-pc-linux-gnu-gcc -E
checking for ANSI C header files... yes
checking for sys/types.h... yes
checking for sys/stat.h... yes
checking for stdlib.h... yes
checking for string.h... yes
checking for memory.h... yes
checking for strings.h... yes
checking for inttypes.h... yes
checking for stdint.h... yes
checking for unistd.h... yes
checking for dlfcn.h... yes
checking for objdir... .libs
checking if x86_64-pc-linux-gnu-gcc supports -fno-rtti -fno-exceptions... no
checking for x86_64-pc-linux-gnu-gcc option to produce PIC... -fPIC -DPIC
checking if x86_64-pc-linux-gnu-gcc PIC flag -fPIC -DPIC works... yes
checking if x86_64-pc-linux-gnu-gcc static flag -static works... yes
checking if x86_64-pc-linux-gnu-gcc supports -c -o file.o... yes
checking if x86_64-pc-linux-gnu-gcc supports -c -o file.o... (cached) yes
checking whether the x86_64-pc-linux-gnu-gcc linker (/usr/x86_64-pc-linux-gnu/bin/ld -m elf_x86_64) supports shared libraries... yes
checking whether -lc should be explicitly linked in... no
checking dynamic linker characteristics... GNU/Linux ld.so
checking how to hardcode library paths into programs... immediate
checking whether stripping libraries is possible... yes
checking if libtool supports shared libraries... yes
checking whether to build shared libraries... yes
checking whether to build static libraries... no
checking for explicit_bzero... yes
checking for rawmemchr... yes
checking for signalfd... yes
checking for clock_gettime in -lrt... yes
checking for pthread_create in -lpthread... yes
checking for dlopen in -ldl... yes
checking linux/types.h usability... yes
checking linux/types.h presence... yes
checking for linux/types.h... yes
checking linux/if_alg.h usability... yes
checking linux/if_alg.h presence... yes
checking for linux/if_alg.h... yes
checking for glib-2.0 >= 2.28... yes
checking for gthread-2.0 >= 2.16... yes
checking for dbus-1 >= 1.6... yes
checking D-Bus configuration directory... /etc
checking D-Bus system bus services dir... /usr/share/dbus-1/system-services
checking D-Bus session bus services dir... /usr/share/dbus-1/services
${prefix}/share/zsh/site-functions
checking for libudev >= 172... yes
checking for udev_hwdb_new in -ludev... yes
checking udev directory... /lib/udev
checking for json-c >= 0.13... yes
checking for libical... yes
checking for ell >= 0.39... yes
checking readline/readline.h usability... yes
checking readline/readline.h presence... yes
checking for readline/readline.h... yes
checking that generated files are newer than configure... done
configure: creating ./config.status
config.status: creating Makefile
config.status: creating src/bluetoothd.8
config.status: creating lib/bluez.pc
config.status: creating config.h
config.status: executing depfiles commands
config.status: executing libtool commands
>>> Source configured.
>>> Compiling source in /var/tmp/portage/net-wireless/bluez-9999/work/bluez-9999 ...
 * abi_x86_64.amd64: running multilib-minimal_abi_src_compile
make -s -j8 
/bin/sh: line 4: ell/shared: No such file or directory
make: *** [Makefile:11628: ell/shared] Error 1
make: *** Waiting for unfinished jobs....
```

Vudentz, 2021-04-02:
If you want to enable building with ell as a shared library you will need to use --enable-external-ell, there doesn't seem to be a --disable-external-ell so Im not sure where you got that.

Vudentz, 2021-04-02:
It does look like it works with/without --disable-externa-ell: https://gist.github.com/Vudentz/c5ac9a44eb2eba38033b6f06caa7c715

joakim-tjernlund, 2021-04-02:
I added an explicit --enable-external-ell
But still I get :
```
make -j8 -l6 -s 
/bin/sh: line 4: ell/shared: No such file or directory
make: *** [Makefile:11628: ell/shared] Error 1
make: *** Waiting for unfinished jobs....

joakim-tjernlund, 2021-04-02:
To be clear the same build receipt work a few days ago, started to fail today.

Vudentz, 2021-04-02:
Hmm, perhaps the problem is with the instalation of libell, did you install it from source or from tar.gz?

joakim-tjernlund, 2021-04-02:
It is installed in Gentoo as a lib:
```
 qlist dev-libs/ell
/usr/share/doc/ell-0.39/README.bz2
/usr/share/doc/ell-0.39/ChangeLog.bz2
/usr/include/ell/tester.h
/usr/include/ell/acd.h
/usr/include/ell/icmp6.h
/usr/include/ell/path.h
/usr/include/ell/gpio.h
/usr/include/ell/time.h
/usr/include/ell/ecdh.h
/usr/include/ell/ecc.h
/usr/include/ell/cert.h
/usr/include/ell/dhcp6.h
/usr/include/ell/dhcp.h
/usr/include/ell/net.h
/usr/include/ell/dir.h
/usr/include/ell/file.h
/usr/include/ell/key.h
/usr/include/ell/uuid.h
/usr/include/ell/tls.h
/usr/include/ell/pem.h
/usr/include/ell/base64.h
/usr/include/ell/uintset.h
/usr/include/ell/random.h
/usr/include/ell/cipher.h
/usr/include/ell/hwdb.h
/usr/include/ell/dbus-client.h
/usr/include/ell/dbus-service.h
/usr/include/ell/dbus.h
/usr/include/ell/rtnl.h
/usr/include/ell/genl.h
/usr/include/ell/netlink.h
/usr/include/ell/checksum.h
/usr/include/ell/log.h
/usr/include/ell/ringbuf.h
/usr/include/ell/io.h
/usr/include/ell/timeout.h
/usr/include/ell/signal.h
/usr/include/ell/idle.h
/usr/include/ell/main.h
/usr/include/ell/settings.h
/usr/include/ell/string.h
/usr/include/ell/hashmap.h
/usr/include/ell/queue.h
/usr/include/ell/utf8.h
/usr/include/ell/strv.h
/usr/include/ell/test.h
/usr/include/ell/util.h
/usr/include/ell/ell.h
/usr/lib64/libell.so
/usr/lib64/libell.so.0
/usr/lib64/libell.so.0.0.2
/usr/lib64/pkgconfig/ell.pc

joakim-tjernlund, 2021-04-03:
More details:
```
make 
for f in  ; do \
	if [ ! -f $f ] ; then \
		ln -s -t ell -f /var/tmp/portage/net-wireless/bluez-9999/work/bluez-9999-abi_x86_64.amd64/../ell/$f ; \
	fi \
done > ell/shared
/bin/sh: line 4: ell/shared: No such file or directory
make: *** [Makefile:11628: ell/shared] Error 1
```
The do ```mkdir ell``` and all is well. Make builds all.

joakim-tjernlund, 2021-04-05:
Should I just add ```mkdir ell``` for git builds or will you?

joakim-tjernlund, 2021-04-05:
I just noted that this github repo i 10 days behind https://git.kernel.org/pub/scm/bluetooth/bluez.git/log/

Vudentz, 2021-04-05:
> I just noted that this github repo i 10 days behind https://git.kernel.org/pub/scm/bluetooth/bluez.git/log/

@tedd-an Can you take a look why we stopped syncing with upstream?

BluezTestBot, 2021-04-05:
@Vudentz  The action for source sync had been disabled by Github.
It is enabled again and sync action will be back shortly.

Vudentz, 2021-04-05:
5.58 shall have addressed this problem.

closed 2021-04-05 by Vudentz

joakim-tjernlund, 2021-04-06:
> 5.58 shall have addressed this problem.

5.58 did NOT fix this problem.

joakim-tjernlund, 2021-04-06:
Looking at the 5.58 tar ball there is an ell dir. with files in it. I am trying to build a git master checkout and there is no ell dir. there so make files.

Vudentz, 2021-04-06:
@joakim-tjernlund Are you still not able to build the tarball?

joakim-tjernlund, 2021-04-07:
> @joakim-tjernlund Are you still not able to build the tarball?

I didn't try, I can see the ell dir in the tarball so I assume it good. I am building from git tree and that fails because git does not have the ell dir.

Vudentz, 2021-04-07:
> > @joakim-tjernlund Are you still not able to build the tarball?
> 
> I didn't try, I can see the ell dir in the tarball so I assume it good. I am building from git tree and that fails because git does not have the ell dir.

When it is build from source it should generate the ell dir, did you start with a clean tree?

joakim-tjernlund, 2021-04-07:
Just did the same, still there:
```
./configure --prefix=/usr --build=x86_64-pc-linux-gnu --host=x86_64-pc-linux-gnu --mandir=/usr/share/man --infodir=/usr/share/info --datadir=/usr/sha
re --sysconfdir=/etc --localstatedir=/var/lib --disable-dependency-tracking --disable-silent-rules --docdir=/usr/share/doc/bluez-9999 --htmldir=/usr/
share/doc/bluez-9999/html --with-sysroot=/ --libdir=/usr/lib64 --localstatedir=/var --disable-android --enable-datafiles --enable-optimization --disa
ble-debug --enable-pie --enable-threads --enable-library --enable-tools --enable-manpages --enable-monitor --with-systemdsystemunitdir=/lib/systemd/s
ystem --with-systemduserunitdir=/usr/lib/systemd/user --disable-btpclient --disable-external-ell --enable-cups --disable-deprecated --disable-experim
ental --enable-mesh --enable-external-ell --disable-midi --enable-obex --enable-client --disable-systemd --disable-test --enable-udev --enable-hid2hc
i --enable-external-ell --enable-sixaxis
checking for a BSD-compatible install... /usr/bin/install -c
checking whether build environment is sane... yes
checking for a thread-safe mkdir -p... /bin/mkdir -p
checking for gawk... gawk
checking whether make sets $(MAKE)... yes
checking whether make supports nested variables... yes
checking how to create a pax tar archive... gnutar
checking whether make supports nested variables... (cached) yes
checking whether to enable maintainer-specific portions of Makefiles... no
checking for x86_64-pc-linux-gnu-pkg-config... /usr/bin/x86_64-pc-linux-gnu-pkg-config
checking pkg-config is at least version 0.9.0... yes
checking whether make supports the include directive... yes (GNU style)
checking for x86_64-pc-linux-gnu-gcc... x86_64-pc-linux-gnu-gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables... 
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether we are using the GNU C compiler... yes
checking whether x86_64-pc-linux-gnu-gcc accepts -g... yes
checking for x86_64-pc-linux-gnu-gcc option to accept ISO C89... none needed
checking whether x86_64-pc-linux-gnu-gcc understands -c and -o together... yes
checking dependency style of x86_64-pc-linux-gnu-gcc... none
checking for C/C++ restrict keyword... __restrict
checking for x86_64-pc-linux-gnu-gcc... (cached) x86_64-pc-linux-gnu-gcc
checking whether we are using the GNU C compiler... (cached) yes
checking whether x86_64-pc-linux-gnu-gcc accepts -g... (cached) yes
checking for x86_64-pc-linux-gnu-gcc option to accept ISO C89... (cached) none needed
checking whether x86_64-pc-linux-gnu-gcc understands -c and -o together... (cached) yes
checking dependency style of x86_64-pc-linux-gnu-gcc... (cached) none
checking whether x86_64-pc-linux-gnu-gcc accepts -fPIE... yes
checking build system type... x86_64-pc-linux-gnu
checking host system type... x86_64-pc-linux-gnu
checking how to print strings... printf
checking for a sed that does not truncate output... /bin/sed
checking for grep that handles long lines and -e... /bin/grep
checking for egrep... /bin/grep -E
checking for fgrep... /bin/grep -F
checking for ld used by x86_64-pc-linux-gnu-gcc... /usr/x86_64-pc-linux-gnu/bin/ld
checking if the linker (/usr/x86_64-pc-linux-gnu/bin/ld) is GNU ld... yes
checking for BSD- or MS-compatible name lister (nm)... /usr/bin/x86_64-pc-linux-gnu-nm -B
checking the name lister (/usr/bin/x86_64-pc-linux-gnu-nm -B) interface... BSD nm
checking whether ln -s works... yes
checking the maximum length of command line arguments... 1572864
checking how to convert x86_64-pc-linux-gnu file names to x86_64-pc-linux-gnu format... func_convert_file_noop
checking how to convert x86_64-pc-linux-gnu file names to toolchain format... func_convert_file_noop
checking for /usr/x86_64-pc-linux-gnu/bin/ld option to reload object files... -r
checking for x86_64-pc-linux-gnu-objdump... x86_64-pc-linux-gnu-objdump
checking how to recognize dependent libraries... pass_all
checking for x86_64-pc-linux-gnu-dlltool... no
checking for dlltool... no
checking how to associate runtime and link libraries... printf %s\n
checking for x86_64-pc-linux-gnu-ar... x86_64-pc-linux-gnu-ar
checking for archiver @FILE support... @
checking for x86_64-pc-linux-gnu-strip... x86_64-pc-linux-gnu-strip
checking for x86_64-pc-linux-gnu-ranlib... x86_64-pc-linux-gnu-ranlib
checking command to parse /usr/bin/x86_64-pc-linux-gnu-nm -B output from x86_64-pc-linux-gnu-gcc object... ok
checking for sysroot... /
checking for a working dd... /bin/dd
checking how to truncate binary pipes... /bin/dd bs=4096 count=1
checking for x86_64-pc-linux-gnu-mt... no
[... 70 lines cut ...]
/bin/mkdir -p lib/bluetooth
ln -s -f /home/jocke/bluez/lib/hci_lib.h lib/bluetooth/hci_lib.h
/bin/mkdir -p lib/bluetooth
ln -s -f /home/jocke/bluez/lib/sco.h lib/bluetooth/sco.h
/bin/mkdir -p lib/bluetooth
ln -s -f /home/jocke/bluez/lib/l2cap.h lib/bluetooth/l2cap.h
/bin/mkdir -p lib/bluetooth
ln -s -f /home/jocke/bluez/lib/sdp.h lib/bluetooth/sdp.h
/bin/mkdir -p lib/bluetooth
ln -s -f /home/jocke/bluez/lib/sdp_lib.h lib/bluetooth/sdp_lib.h
/bin/mkdir -p lib/bluetooth
ln -s -f /home/jocke/bluez/lib/rfcomm.h lib/bluetooth/rfcomm.h
/bin/mkdir -p lib/bluetooth
ln -s -f /home/jocke/bluez/lib/bnep.h lib/bluetooth/bnep.h
/bin/mkdir -p lib/bluetooth
ln -s -f /home/jocke/bluez/lib/cmtp.h lib/bluetooth/cmtp.h
/bin/mkdir -p lib/bluetooth
ln -s -f /home/jocke/bluez/lib/hidp.h lib/bluetooth/hidp.h
for f in  ; do \
	if [ ! -f $f ] ; then \
		ln -s -t ell -f /home/jocke/bluez/../ell/$f ; \
	fi \
done > ell/shared
/bin/sh: line 4: ell/shared: No such file or directory
make: *** [Makefile:11628: ell/shared] Error 1

joakim-tjernlund, 2021-04-07:
make --version
GNU Make 4.3
Built for x86_64-pc-linux-gnu

Also --disable-systemd if that matters

joakim-tjernlund, 2021-04-07:
It is --enable-maintainer-mode that makes it work.
Without it fails

joakim-tjernlund, 2021-04-07:
Fix:
```
--- a/Makefile.am
+++ b/Makefile.am
@@ -620,6 +620,7 @@ lib/bluetooth/%.h: lib/%.h
        $(AM_V_GEN)$(LN_S) -f $(abspath $<) $@
 
 ell/shared: Makefile
+       $(AM_V_at)$(MKDIR_P) ell
        $(AM_V_GEN)for f in $(ell_shared) ; do \
                if [ ! -f $$f ] ; then \
                        $(LN_S) -t ell -f $(abs_srcdir)/../ell/$$f ; \
```

I noted one thing though:
```
if MESH
unit_tests += unit/test-mesh-crypto
unit_test_mesh_crypto_CPPFLAGS = $(ell_cflags)
unit_test_mesh_crypto_SOURCES = unit/test-mesh-crypto.c \
				mesh/crypto.h ell/internal ell/ell.h
unit_test_mesh_crypto_LDADD = $(ell_ldadd)
endif
```
Here you depend on internal ell regardless. Should be fixed I think.

joakim-tjernlund, 2021-04-08:
Unless you are about to fix the error,  you mind opening the issue again?

joakim-tjernlund, 2021-04-09:
ping ?

joakim-tjernlund, 2021-04-27:
Ping ? This is still a bug and need the fix quoted above:
+       $(AM_V_at)$(MKDIR_P) ell