bluez/bluez #128: A2DP Reconnect by default breaks switching between devices

wengxt, 2021-04-15:
I'm not sure if this only applies to airpods or not since this is the only one that I use.

https://github.com/bluez/bluez/commit/20275c5a9b22c321853199d85c2b14b06befd0f6
In this commit, A2DP is enabled with reconnect by default. But this change breaks the use case for me:
1. Connect Airpod to PC
2. Connect Airpod to Phone
3. A few seconds later, the PC automatically reconnect to Airpods again. 

Relevant debug log from bluetoothd
bluetoothd[28768]: plugins/policy.c:disconnect_cb() reason 1
bluetoothd[28768]: plugins/policy.c:disconnect_cb() Device /org/bluez/hci0/dev_28_F0_33_8E_F0_DC identified for auto-reconnection

The reason 1 is "TIMEOUT", so current implementation will try to reconnect. This makes airpod unable to switch it to another device automatically since re-connect will always happen. I have to disconnect on the PC manually and then connect on the phone, which is very inconvenient. Also, this only happens to bluez based system. Switching between Windows / Phone use above method just works.

Remove A2DP profile from reconnect uuid in main.conf is a working workaround for me.

Vudentz, 2021-04-21:
@wengxt Not sure how you suggest us to handle this since the reason 1 is Timeout that is exactly what reconnect was design to reconnect, it looks like the Airpod is not disconnecting cleanly.

Vudentz, 2021-04-21:
@wengxt Anyway you can disable the reconnect for A2DP sink by editing main.conf with:

`ReconnectUUIDs=00001112-0000-1000-8000-00805f9b34fb,0000111f-0000-1000-8000-00805f9b34fb,0000110a-0000-1000-8000-00805f9b34fb
`

closed 2021-05-05 by Vudentz