bluez/bluez #129: git master build broken

joakim-tjernlund, 2021-04-15:
In Addition to https://github.com/bluez/bluez/issues/125 which is closed but unfixed(go figure) there is a new breakage:
```
libtool: link: x86_64-pc-linux-gnu-gcc -fPIC -O2 -pipe -Wno-error -fno-stack-check -Wno-deprecated-declarations -fdebug-prefix-map=..=/var/tmp/portage/net-wireless/bluez-9999 -pie -Wl,-z -Wl,now -Wl,-O1 -o tools/mesh-cfgtest tools/mesh-cfgtest.o  -Wl,--as-needed lib/.libs/libbluetooth-internal.a src/.libs/libshared-ell.a -lell
libtool: link: x86_64-pc-linux-gnu-gcc -fPIC -O2 -pipe -Wno-error -fno-stack-check -Wno-deprecated-declarations -fdebug-prefix-map=..=/var/tmp/portage/net-wireless/bluez-9999 -pie -Wl,-z -Wl,now -Wl,-O1 -o tools/mesh-cfgclient tools/mesh-cfgclient.o tools/mesh/cfgcli.o tools/mesh/keys.o tools/mesh/util.o tools/mesh/remote.o tools/mesh/agent.o tools/mesh/mesh-db.o mesh/util.o mesh/crypto.o  -Wl,--as-needed lib/.libs/libbluetooth-internal.a src/.libs/libshared-ell.a -lell -ljson-c -lreadline
/usr/lib/gcc/x86_64-pc-linux-gnu/9.3.0/../../../../x86_64-pc-linux-gnu/bin/ld: tools/mesh-cfgtest.o: in function `join_complete':
mesh-cfgtest.c:(.text+0x626): undefined reference to `l_tester_get_stage'
/usr/lib/gcc/x86_64-pc-linux-gnu/9.3.0/../../../../x86_64-pc-linux-gnu/bin/ld: tools/mesh-cfgtest.o: in function `dev_msg_recv_call':
mesh-cfgtest.c:(.text+0x97a): undefined reference to `l_tester_get_stage'
/usr/lib/gcc/x86_64-pc-linux-gnu/9.3.0/../../../../x86_64-pc-linux-gnu/bin/ld: tools/mesh-cfgtest.o: in function `attach_node_reply':
mesh-cfgtest.c:(.text+0xd0c): undefined reference to `l_tester_get_stage'
/usr/lib/gcc/x86_64-pc-linux-gnu/9.3.0/../../../../x86_64-pc-linux-gnu/bin/ld: tools/mesh-cfgtest.o: in function `import_subnet_reply':
mesh-cfgtest.c:(.text+0x11a0): undefined reference to `l_tester_get_stage'
/usr/lib/gcc/x86_64-pc-linux-gnu/9.3.0/../../../../x86_64-pc-linux-gnu/bin/ld: tools/mesh-cfgtest.o: in function `create_appkey_reply':
mesh-cfgtest.c:(.text+0x1261): undefined reference to `l_tester_get_stage'
/usr/lib/gcc/x86_64-pc-linux-gnu/9.3.0/../../../../x86_64-pc-linux-gnu/bin/ld: tools/mesh-cfgtest.o:mesh-cfgtest.c:(.text+0x1339): more undefined references to `l_tester_get_stage' follow
/usr/lib/gcc/x86_64-pc-linux-gnu/9.3.0/../../../../x86_64-pc-linux-gnu/bin/ld: tools/mesh-cfgtest.o: in function `main':
mesh-cfgtest.c:(.text.startup+0x259): undefined reference to `l_tester_add_full'
/usr/lib/gcc/x86_64-pc-linux-gnu/9.3.0/../../../../x86_64-pc-linux-gnu/bin/ld: mesh-cfgtest.c:(.text.startup+0x298): undefined reference to `l_tester_add_full'
/usr/lib/gcc/x86_64-pc-linux-gnu/9.3.0/../../../../x86_64-pc-linux-gnu/bin/ld: mesh-cfgtest.c:(.text.startup+0x2d7): undefined reference to `l_tester_add_full'
/usr/lib/gcc/x86_64-pc-linux-gnu/9.3.0/../../../../x86_64-pc-linux-gnu/bin/ld: mesh-cfgtest.c:(.text.startup+0x316): undefined reference to `l_tester_add_full'
/usr/lib/gcc/x86_64-pc-linux-gnu/9.3.0/../../../../x86_64-pc-linux-gnu/bin/ld: mesh-cfgtest.c:(.text.startup+0x355): undefined reference to `l_tester_add_full'
collect2: error: ld returned 1 exit status
```
Configure cmd:

```
./configure --prefix=/usr --build=x86_64-pc-linux-gnu --host=x86_64-pc-linux-gnu --mandir=/usr/share/man --infodir=/usr/share/info --datadir=/usr/share --sysconfdir=/etc --localstatedir=/var/lib --disable-dependency-tracking --disable-silent-rules --docdir=/usr/share/doc/bluez-9999 --htmldir=/usr/share/doc/bluez-9999/html --with-sysroot=/ --libdir=/usr/lib64 --localstatedir=/var --disable-android --enable-datafiles --enable-optimization --disable-debug --enable-pie --enable-threads --enable-library --enable-tools --enable-manpages --enable-monitor --with-systemdsystemunitdir=/lib/systemd/system --with-systemduserunitdir=/usr/lib/systemd/user --disable-btpclient --disable-external-ell --enable-cups --disable-deprecated --disable-experimental --enable-mesh --enable-external-ell --disable-midi --enable-obex --enable-client --disable-systemd --disable-test --enable-udev --enable-hid2hci --enable-external-ell --enable-sixaxis

gearhead, 2021-04-15:
yep. Me too!

joakim-tjernlund, 2021-04-16:
I think you need libell git: https://git.kernel.org/pub/scm/libs/ell/ell.git/commit/?id=8862504deaf47ce12d2d05d76b94f0b87c72f2bd

gearhead, 2021-04-16:
I have always been able to 'just build' bluez-git and it now no longer builds. I'll see if Arch has updated the basic ell package to the point that it will build.

gearhead, 2021-04-19:
I built and installed ell-git and was able to build

bgix, 2021-04-20:
This has been resolved in the master tip, and will be fixed in next bluez release.  In the meantime use the tip, and and make sure you have the ell master tip as well.

bgix, 2021-04-20:
Issue closed

closed 2021-04-20 by bgix