bluez/bluez #130: Bluetooth speaker/headset disconnects and reconnects multiple times one day

c02y, 2021-04-17:
This problem has been about 2 months, it happens every day, multiple times a day. It drives me crazy.

After I connect my Bluetooth speaker or headset to my Manjaro-Linux laptop,  when I'm watching streaming video or local video, it suddenly disconnects, and then after around 10s, it reconnects automatically, this happens multiple times one day **and every single day**. 

I googled month ago, and found https://github.com/bluez/bluez/issues/102 , I thought it was the v5.56 version problem, so I wait for Manjaro repo to update the v5.58 version to get it fixed, but after updating the v5.58 version of bluez/bluez-cups/bluez-libs/bluez-utils packages days ago, and used it for one day(after restarting my OS of cause), this problem is still here.

I've used Linux kernel 5.4-LTS/5.9/5.10-LTS/5.11,  all of them have this issue.

Here is the service status log:
```
● bluetooth.service - Bluetooth service
     Loaded: loaded (/usr/lib/systemd/system/bluetooth.service; enabled; vendor preset: disabled)
     Active: active (running) since Sat 2021-04-17 20:33:23 CST; 50min ago
       Docs: man:bluetoothd(8)
   Main PID: 1365 (bluetoothd)
     Status: "Running"
      Tasks: 1 (limit: 18879)
     Memory: 2.5M
     CGroup: /system.slice/bluetooth.service
             └─1365 /usr/lib/bluetooth/bluetoothd

Apr 17 20:33:37 chz bluetoothd[1365]: src/profile.c:ext_start_servers() L2CAP server failed for Message Access: setsockopt(L2CAP_OPTIONS): Invalid argument (22)
Apr 17 20:33:37 chz bluetoothd[1365]: src/profile.c:ext_start_servers() L2CAP server failed for Phone Book Access: setsockopt(L2CAP_OPTIONS): Invalid argument (22)
Apr 17 20:33:37 chz bluetoothd[1365]: src/profile.c:ext_start_servers() L2CAP server failed for File Transfer: setsockopt(L2CAP_OPTIONS): Invalid argument (22)
Apr 17 20:33:37 chz bluetoothd[1365]: src/profile.c:ext_start_servers() L2CAP server failed for Object Push: setsockopt(L2CAP_OPTIONS): Invalid argument (22)
Apr 17 20:34:29 chz bluetoothd[1365]: src/service.c:btd_service_connect() avrcp-controller profile connect failed for 60:AB:D2:0F:72:EB: Input/output error
Apr 17 20:34:31 chz bluetoothd[1365]: /org/bluez/hci0/dev_60_AB_D2_0F_72_EB/fd0: fd(27) ready
Apr 17 21:18:15 chz bluetoothd[1365]: profiles/audio/avdtp.c:avdtp_connect_cb() connect to 60:AB:D2:0F:72:EB: Host is down (112)
Apr 17 21:18:17 chz bluetoothd[1365]: plugins/policy.c:reconnect_timeout() Reconnecting services failed: Operation already in progress (114)
Apr 17 21:18:18 chz bluetoothd[1365]: src/service.c:btd_service_connect() avrcp-controller profile connect failed for 60:AB:D2:0F:72:EB: Input/output error
Apr 17 21:18:20 chz bluetoothd[1365]: /org/bluez/hci0/dev_60_AB_D2_0F_72_EB/fd1: fd(27) ready
```

If you need anything, please let me know.

bjsowa, 2021-04-18:
I have the exact same problem. I use my bluetooth headset a lot so it can happen up to 10 times a day. Also waited for the 5.58 release thinking it will fix this issue but it didn't. I really hope it can get resolved soon, because it drives me mad sometimes.

OS: Manjaro KDE
Kernel (`uname -r`): 5.10.30-1-MANJARO
Headset: Sennheiser HD 450BT

Vudentz, 2021-04-20:
It seems to be working fine with fedora 33, perhaps it is something related to Manjaro, could one of you please include the HCI traces (can be collected using btmon).

```
Apr 18 09:22:18 chz bluetoothd[1365]: src/profile.c:ext_start_servers() L2CAP server failed for Phone Book Access: setsockopt(L2CAP_OPTIONS): Invalid argument (22)
Apr 18 09:22:18 chz bluetoothd[1365]: src/profile.c:ext_start_servers() L2CAP server failed for File Transfer: setsockopt(L2CAP_OPTIONS): Invalid argument (22)
Apr 18 09:22:18 chz bluetoothd[1365]: src/profile.c:ext_start_servers() L2CAP server failed for Object Push: setsockopt(L2CAP_OPTIONS): Invalid argument (22)
```

This looks like another problem, is obexd the same version as BlueZ? It shouldn't be related to the problem you guys are having with headsets though, but it would be good to check if there is something wrong with obexd as well.

c02y, 2021-04-20:
FYI: no standalone obexd package, `/usr/lib/bluetooth/obexd is owned by bluez 5.58-1`

As for HCI traces, I'm sure this is the right way to get the traces:
```
$ sudo btmon -w hci.log
```
[hci.log](https://github.com/bluez/bluez/files/6346963/hci.log)

Vudentz, 2021-04-20:
> FYI: no standalone obexd package, `/usr/lib/bluetooth/obexd is owned by bluez 5.58-1`
> 
> As for HCI traces, I'm sure this is the right way to get the traces:
> 
> ```
> $ sudo btmon -w hci.log
> ```
> 
> [hci.log](https://github.com/bluez/bluez/files/6346963/hci.log)

That doesn't seem to contain much, did you attempt to connect while btmon was active?

c02y, 2021-04-20:
OK, here is the steps:
1. excute `sudo btmon -w hci.log` in terminal
2. open the blueman-applet
3. disconnet Bose Revolve+ SoundLink
4. reconnet it
5. Ctrl-c to exit the btmon, get the log

Don't know if this step is OK, I cannot reproduce the auto-disconnect-reconnect-issue process though.
[hci.log](https://github.com/bluez/bluez/files/6346996/hci.log)

bjsowa, 2021-04-21:
I managed to reproduce the issue and record the traces. 
The file is big, because it was recorded for 20 minutes (and I don't know how to truncate btsnoop files) but the disconnection happens in the last 30 seconds.

https://drive.google.com/file/d/1EMxpZtoMbeoItXnudUd0_am6OWtzYke4/view?usp=sharing

pasikarkkainen, 2021-04-21:
Do you happen to have Intel Bluetooth adapter in the laptop? There seems to be a firmware bug/issue with some of the recent Intel BT firmware versions.. (it's discussed in pipewire/pulseaudio issue trackers).

bjsowa, 2021-04-21:
Yes, I have the Intel AX200 Bluetooth device.

bjsowa, 2021-04-21:
I found the [thread](https://gitlab.freedesktop.org/pulseaudio/pulseaudio/-/issues/1155) on the pulseaudio issue tracker. Thanks for pointing that out!

c02y, 2021-04-21:
@bjsowa Where can I get the Bluetooth adapter device model info, I tried multiple methods, only know that it is from Intel.

bjsowa, 2021-04-21:
`lsusb` worked for me

c02y, 2021-04-21:
@yes, thanks, already got it before you replied.

pasikarkkainen, 2021-04-21:
Yep, and this is the related pipewire issue: https://gitlab.freedesktop.org/pipewire/pipewire/-/issues/732

Vudentz, 2021-04-21:
> I managed to reproduce the issue and record the traces.
> The file is big, because it was recorded for 20 minutes (and I don't know how to truncate btsnoop files) but the disconnection happens in the last 30 seconds.
> 
> https://drive.google.com/file/d/1EMxpZtoMbeoItXnudUd0_am6OWtzYke4/view?usp=sharing

Welll this is down to the link layer it seems as the connection times out.

```
> HCI Event: Disconnect Complete (0x05) plen 4                                                                                                                                                                                      #115919 [hci0] 1198.041550
        Status: Success (0x00)
        Handle: 256
        Reason: Connection Timeout (0x08)
@ MGMT Event: Device Disconnected (0x000c) plen 8                                                                                                                                                                                  {0x0001} [hci0] 1198.041609
        BR/EDR Address: 00:1B:66:C0:E9:57 (Sennheiser electronic GmbH & Co. KG)
        Reason: Connection timeout (0x01)

```

c02y, 2021-04-26:
This problem seems fixed
1. The fix commit is linux-firmware repo 73144e0, Fri Apr 9
2. It is included in linux-firmware v20210421.r1894.cfa004c

I have been using this firmware for days, so far so good.

closed 2021-04-26 by c02y

rnayabed, 2021-10-16:
I am running on Kernel 5.14 but having the same issue again.

```
Oct 16 19:55:09 rnayabed-laptop dbus-daemon[795]: [session uid=1000 pid=795] Successfully activated service 'org.bluez.obex'
Oct 16 19:55:09 rnayabed-laptop bluetoothd[710]: src/profile.c:ext_start_servers() L2CAP server failed for Message Notification: setsockopt(L2CAP_OPTIONS): Invalid argument (22)
Oct 16 19:55:09 rnayabed-laptop bluetoothd[710]: src/profile.c:ext_start_servers() L2CAP server failed for Message Access: setsockopt(L2CAP_OPTIONS): Invalid argument (22)
Oct 16 19:55:09 rnayabed-laptop bluetoothd[710]: src/profile.c:ext_start_servers() L2CAP server failed for Phone Book Access: setsockopt(L2CAP_OPTIONS): Invalid argument (22)
Oct 16 19:55:09 rnayabed-laptop bluetoothd[710]: src/profile.c:ext_start_servers() L2CAP server failed for File Transfer: setsockopt(L2CAP_OPTIONS): Invalid argument (22)
Oct 16 19:55:09 rnayabed-laptop bluetoothd[710]: src/profile.c:ext_start_servers() L2CAP server failed for Object Push: setsockopt(L2CAP_OPTIONS): Invalid argument (22)
Oct 16 19:55:09 rnayabed-laptop kded5[841]: bluedevil: ObexManager operational changed true
Oct 16 19:55:09 rnayabed-laptop kded5[841]: bluedevil: ObexAgent registered
Oct 16 19:55:34 rnayabed-laptop bluetoothd[710]: src/service.c:btd_service_connect() avrcp-controller profile connect failed for E8:D0:3C:36:51:8D: Input/output error
Oct 16 19:55:34 rnayabed-laptop bluetoothd[710]: /org/bluez/hci0/dev_E8_D0_3C_36_51_8D/sep1/fd0: fd(27) ready
Oct 16 19:55:34 rnayabed-laptop pulseaudio[930]: org.freedesktop.DBus.Properties.Get /org/bluez/hci0/dev_E8_D0_3C_36_51_8D/sep1/fd0 Volume failed: org.freedesktop.DBus.Error.InvalidArgs: No such property 'Volume'
Oct 16 19:57:08 rnayabed-laptop bluetoothd[710]: /org/bluez/hci0/dev_E8_D0_3C_36_51_8D/sep1/fd1: fd(27) ready
Oct 16 19:57:08 rnayabed-laptop pulseaudio[930]: org.freedesktop.DBus.Properties.Get /org/bluez/hci0/dev_E8_D0_3C_36_51_8D/sep1/fd1 Volume failed: org.freedesktop.DBus.Error.InvalidArgs: No such property 'Volume'
Oct 16 19:57:59 rnayabed-laptop bluetoothd[710]: /org/bluez/hci0/dev_E8_D0_3C_36_51_8D/sep1/fd2: fd(27) ready
Oct 16 19:57:59 rnayabed-laptop pulseaudio[930]: org.freedesktop.DBus.Properties.Get /org/bluez/hci0/dev_E8_D0_3C_36_51_8D/sep1/fd2 Volume failed: org.freedesktop.DBus.Error.InvalidArgs: No such property 'Volume'
Oct 16 20:04:19 rnayabed-laptop bluetoothd[710]: /org/bluez/hci0/dev_E8_D0_3C_36_51_8D/sep1/fd3: fd(27) ready
Oct 16 20:04:19 rnayabed-laptop pulseaudio[930]: org.freedesktop.DBus.Properties.Get /org/bluez/hci0/dev_E8_D0_3C_36_51_8D/sep1/fd3 Volume failed: org.freedesktop.DBus.Error.InvalidArgs: No such property 'Volume'
```

c02y, 2021-10-16:
It was fixed in `linux-firmware v20210421.r1894.cfa004c`, I don't have this `disconnects-reconnects` issue anymore, but later, another bug (unable to auto-connect after reboot and wakeup from Hiberate) occurs, and after several times of googling, I don't care about this anymore :(

Now, I use to shortcut bound to a simple script to manually reconnect it.
```
#!/bin/bash

bluetoothctl trust your_device_address
bluetoothctl disconnect your_device_address
bluetoothctl connect your_device_address
```
I failed to autostart it using systemd or from startup part in i3 config, anyone knows it, please let me know.

rnayabed, 2021-10-17:
I kind of gave up, cant live like this. The bluetooth on my PC works fine though so for now ill use speakers and stuff there only.

Whimfoome, 2022-10-30:
I still have this issue on:
- Fedora 36 with Gnome
- Linux 5.19.16-200.fc36.x86_64
- Device: realme buds air 2
- I think I am running with pipewire
- bluez 5.65

Logs from `journalctl | grep -i “blue*” `:
```
окт 30 14:10:41 fedora bluetoothd[782]: profiles/audio/avdtp.c:cancel_request() Suspend: Connection timed out (110)
окт 30 14:10:42 fedora bluetoothd[782]: profiles/audio/avdtp.c:handle_unanswered_req() No reply to Suspend request
окт 30 14:10:42 fedora bluetoothd[782]: src/profile.c:ext_io_disconnected() Unable to get io data for Hands-Free Voice gateway: getpeername: Transport endpoint is not connected (107)
окт 30 14:10:44 fedora bluetoothd[782]: plugins/policy.c:reconnect_timeout() Reconnecting services failed: Operation already in progress (114)
```