bluez/bluez #131: Detailed error reasons for method calls.

mcchou-cr, 2021-04-23:
Chromium OS has been working closely with Linux Bluetooth community to improve BlueZ stack, and there are increasing needs from applications building their features around Bluetooth. One of the major feedback from these application is the lack of the detailed failure reasons as return for D-Bus method call, and these failure reasons can be used in metrics (), optimizing retry mechanism, hinting the reproduce scenario to improve BlueZ stack. The current org.bluez.Error.* are serving the generic errors well. However,g given org.bluez.Error.* errors are used across different interface context which does not serve the the detailed failure reasons well.

My initial proposal is to leverage the string attached to org.bluez.Error.* method return. Take org.bluez.Device1.Connect() for instance, a set of detailed errors can be defined as follows, and this error code can be attach to that string to provide more meaning failure reason under the context of Connect() method call.

Looking forward to your feedback and comments. Thanks.

```
enum error_device_connect {
	/* BR/EDR connection failure reasons */

	/* Either the profile is already connected or ACL connection is in
	 * place.
	 * strerror: EALREADY, EISCONN
	 */
	CONNECT_FAIL_BREDR_ALREADY_CONNECTED = 1,
	/* Failed due to page timeout.
	 * strerror: EHOSTDOWN
	 */
	CONNECT_FAIL_BREDR_PAGE_TIMEOUT,
	/* Failed to find connectable services or the target service.
	 * strerror: ENOPROTOOPT
	 */
	CONNECT_FAIL_BREDR_PROFILE_UNAVAILABLE,
	/* Failed to complete the SDP search.
	 * strerror: none
	 */
	CONNECT_FAIL_BREDR_SDP_SEARCH,
	/* Failed to create or connect to BT IO socket.
	 * strerror: EIO
	 */
	CONNECT_FAIL_BREDR_CREATE_SOCKET,
	/* Failed due to invalid arguments.
	 * strerror: EINVAL
	 */
	CONNECT_FAIL_BREDR_INVALID_ARGUMENTS,
	/* Failed due to adapter not powered.
	 * strerror: EHOSTUNREACH
	 */
	CONNECT_FAIL_BREDR_ADAPTER_NOT_POWERED,
	/* Failed due to unsupported state transition of L2CAP channel or other
	 * features either by the local host or the remote.
	 * strerror: EOPNOTSUPP, EPROTONOSUPPORT
	 */
	CONNECT_FAIL_BREDR_NOT_SUPPORTED,
	/* Failed due to the socket is in bad state.
	 * strerror: EBADFD
	 */
	CONNECT_FAIL_BREDR_BAD_SOCKET,
	/* Failed to allocate memory.
	 * strerror: ENOMEM
	 */
	CONNECT_FAIL_BREDR_MEMORY_ALLOC,
	/* Failed due to other ongoing operations, such as pairing, busy L2CAP
	 * channel or the operation disallowed by the controller.
	 * strerror: EBUSY
	 */
	CONNECT_FAIL_BREDR_BUSY,
	/* Failed due to reaching the synchronous connection limit to a device.
	 * strerror: EMLINK
	 */
	CONNECT_FAIL_BREDR_SYNC_CONNECT_LIMIT,
	/* Failed due to authentication failure.
	 * strerror: EACCES
	 */
	CONNECT_FAIL_BREDR_AUTHENTICATION,
	/* Failed due to connection timeout
	 * strerror: ETIMEDOUT
	 */
	CONNECT_FAIL_BREDR_TIMEOUT,
	/* Refused by the remote device due to limited resource, security reason
	 * or unacceptable address type.
	 * strerror: ECONNREFUSED
	 */
	CONNECT_FAIL_BREDR_REFUSED,
	/* Terminated by the remote device due to limited resource or power
	 * off.
	 * strerror: ECONNRESET
	 */
	CONNECT_FAIL_BREDR_TERM_BY_REMOTE,
	/* Terminated by the local host.
	 * strerror: ECONNABORTED
	 */
	CONNECT_FAIL_BREDR_TERM_BY_LOCAL,
	/* Failed due to repeated attempts.
	 * strerror: ELOOP
	 */
	CONNECT_FAIL_BREDR_REPEATED_ATTEMPTS,
	/* Failed due to protocol error.
	 * strerror: EPROTO
	 */
	CONNECT_FAIL_BREDR_PROTO_ERROR,
	/* Unknown reason. */
	CONNECT_FAIL_BREDR_OTHER,
};
```

Vudentz, 2021-04-24:
Use something like BTD_ERR_BREDR_CONN_<NAME>, btw are you planning to have a different domain for each procedure? That would be awful big and I assume the D-Bus interface and method already give enough information about the procedure so we don't have to create an error domain for each of them.

mcchou-cr, 2021-04-30:
It was my initial thought, but then I realized that the group of errors should be reused among all procedure. For instance, the BR/EDR connection errors are completely applicable for method Connect() and ConnectProfile(), so it doesn't make sense to duplicate the error codes cased on domain.

closed 2021-10-15 by mcchou-cr