bluez/bluez #132: Build-Option "--enable-experimental" should add a "--experimental" to "ExecStart" in Service-Unit

schmidicom, 2021-04-27:
If "bluez" is built and installed with "--enable-experimental" the option "ExecStart=/usr/libexec/bluetooth/bluetoothd" in the Service-Unit "bluetooth.service" should also contain an "--experimental".

Example:
ExecStart=/usr/libexec/bluetooth/bluetoothd --experimental

Vudentz, 2021-04-27:
@schmidicom there are not necessarily connected, --enable-experimental is to build experimental plugins/tools, --experimental is for enabling experimental D-Bus Interfaces.

schmidicom, 2021-04-28:
OK, but then it might be a good idea to control the activation of these experimental D-Bus interfaces via the configuration files?
This would be the nicer way than modifying the Service-Unit.

Vudentz, 2021-05-05:
> OK, but then it might be a good idea to control the activation of these experimental D-Bus interfaces via the configuration files?
> This would be the nicer way than modifying the Service-Unit.

You mean have an entry inside main.conf so one don't have to pass over the command line?

schmidicom, 2021-05-05:
Yes, but it's just a suggestion.

I would like to explain why I wanted to enable this in the first place:
My PipeWire complained that my bluez was missing some features and I should install bluez with "--enable_experimental". Which I was able to do quickly and easily thanks to my distribution "Gentoo". But after that it became clear that this was not enough, it was also necessary to modify the systemd service unit.

And modifying systemd service units to enable/disable features (even experimental ones) is not a nice way to go. If this could be done in a configuration file (like "main.conf"), it would be much nicer.

closed 2021-05-17 by BluezTestBot

schmidicom, 2021-05-19:
Thanks for the work.
I tried the patch on version 5.58 and it seems to work. The "Battery Provider Manager" which is required by PipeWire is available according to the log.

> Mai 19 07:50:20 shs-pulse systemd[1]: Starting Bluetooth service...
> Mai 19 07:50:20 shs-pulse bluetoothd[54971]: Bluetooth daemon 5.58
> Mai 19 07:50:20 shs-pulse bluetoothd[54971]: src/main.c:check_options() Unknown key Experimental for group General in /etc/bluetooth/main.co>
> Mai 19 07:50:20 shs-pulse systemd[1]: Started Bluetooth service.
> Mai 19 07:50:20 shs-pulse bluetoothd[54971]: Starting SDP server
> Mai 19 07:50:20 shs-pulse bluetoothd[54971]: Bluetooth management interface 1.19 initialized
> Mai 19 07:50:20 shs-pulse bluetoothd[54971]: Battery Provider Manager created
> Mai 19 07:50:20 shs-pulse bluetoothd[54971]: Adv Monitor Manager created with supported features:0x00000000, enabled features:0x00000000, ma>

Only the check_options() function might need to be modified.