bluez/bluez #134: maintaining mesh-cfgclient net/app keys

FBNH, 2021-05-04:
When working with the meshctl, it is very easy to control & align the keys between multiple nodes.
Is there a way to do the same with mesh-cfgclient?

inga-s, 2022-01-19:
meshctl is an old tool that hasn't been mainatained (kept inline with the latest mesh spec) in quite some time. I expect it will be deprecated.
 Not clear what is meant by "easy to control and align"? Is some specific functionality missing from mesh-cfgclient?

sentineta-prime, 2022-11-19:
> meshctl is an old tool that hasn't been mainatained (kept inline with the latest mesh spec) in quite some time. I expect it will be deprecated. Not clear what is meant by "easy to control and align"? Is some specific functionality missing from mesh-cfgclient?

I have a question, there is a cmd which can control gerneral model 1000 by sending on/off in meshctl tool, in 5.64 version bluez, I didn't find similar functionality in mesh-cfgclient. Will you add this cmd in the future version?

inga-s, 2023-04-06:
The intent of the mesh-cfgclient tool is to provide only configuration functionaity, i.e., it implements strictly Configuration Client Model. There are no plans to extend the tool with anything that is not relevant to configuring a mesh network.

The sample Python implementation of "on/off" model can be found in test/test-mesh