bluez/bluez #135: using acquire-write / acquire-notify

1am, 2021-05-07:
Hello,
Does maybe anyone here know how to use the acquire-notify / acquire-write commands from bluetoothctl on Ubuntu 20 kernel 5.8.0-50-generic / 5.12.1-051201-generic or actually any other.

It worked once for me and I've found it amazing but now repeatedly I can't get it to work again. Looking as far as tu Bluez 5.55/5.56/5.58 source code, btmon and dbus-monitor everything seems fine, the write/notify gets acquired and a proper FD shows up in `/proc/$(pgrep bluetoothctl)/fd/` folder. The thing is that whatever I do, neither `echo 30313233 > /proc/$(pgrep bluetoothctl)/fd/7` or `cat /proc/$(pgrep bluetoothctl)/fd/8` (7 is my FD for acquire-write, 8 for acquire-notify) bash throws `No such device or address`.

I can't find any info on what could be the problem here, If I'm missing something and why did it work once before and later stopped. Obviously I've tried resetting everything, changing HCI dongles and all sorts of the classical things. What I'm missing is some sort of indication on what could I debug further and learn more about the Bluez internals. The code in https://github.com/bluez/bluez/blob/5.58/client/gatt.c#L954 seems straightforward and in my case executes without any problems, just that the FD is later finding the proper socket to write to / read from. Any help would be really appreciated as all my leads on the internet have dried up.

I've tried debugging the FDs with `strace` but there's not really much more info except non existence of the device or address.

```
$ strace -e open,openat -f cat /proc/$(pgrep bluetoothctl)/fd/8
openat(AT_FDCWD, "/etc/ld.so.cache", O_RDONLY|O_CLOEXEC) = 3
openat(AT_FDCWD, "/lib/x86_64-linux-gnu/libc.so.6", O_RDONLY|O_CLOEXEC) = 3
openat(AT_FDCWD, "/usr/lib/locale/locale-archive", O_RDONLY|O_CLOEXEC) = 3
openat(AT_FDCWD, "/proc/11647/fd/8", O_RDONLY) = -1 ENXIO (No such device or address)
cat: /proc/11647/fd/8openat(AT_FDCWD, "/usr/share/locale/locale.alias", O_RDONLY|O_CLOEXEC) = 3
openat(AT_FDCWD, "/usr/share/locale/en_US.UTF-8/LC_MESSAGES/libc.mo", O_RDONLY) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/usr/share/locale/en_US.utf8/LC_MESSAGES/libc.mo", O_RDONLY) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/usr/share/locale/en_US/LC_MESSAGES/libc.mo", O_RDONLY) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/usr/share/locale/en.UTF-8/LC_MESSAGES/libc.mo", O_RDONLY) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/usr/share/locale/en.utf8/LC_MESSAGES/libc.mo", O_RDONLY) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/usr/share/locale/en/LC_MESSAGES/libc.mo", O_RDONLY) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/usr/share/locale-langpack/en_US.UTF-8/LC_MESSAGES/libc.mo", O_RDONLY) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/usr/share/locale-langpack/en_US.utf8/LC_MESSAGES/libc.mo", O_RDONLY) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/usr/share/locale-langpack/en_US/LC_MESSAGES/libc.mo", O_RDONLY) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/usr/share/locale-langpack/en.UTF-8/LC_MESSAGES/libc.mo", O_RDONLY) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/usr/share/locale-langpack/en.utf8/LC_MESSAGES/libc.mo", O_RDONLY) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/usr/share/locale-langpack/en/LC_MESSAGES/libc.mo", O_RDONLY) = -1 ENOENT (No such file or directory)
: No such device or address
+++ exited with 1 +++
```

What makes me wonder is that using `notify on` on bluetoothctl works perfectly fine and I'm receiving the data from BLE peripheral as expected. Just can't get the FDs to work

Vudentz, 2021-05-15:
Wait, who said writing to someones else via filedescriptor would work? Afaik the fd namespace is private to the process and allowing some other process to write/read the filedescriptor most likely would lead to other problems, or are you actually trying to use bluetoothctl to send/receive?

1am, 2021-05-15:
Initially I was trying to open a FD with bluetoothctl and write it from bash just to test. Initially I've found out about that in https://unix.stackexchange.com/a/392106 and during my manual tests it actually worked. Later on it stopped and couldn't replicate that.
The idea for me was to test it out by hand before writing an application which uses AcquireRead/AcquireWrite calls by itself and manages FD it on it's own. This part luckily worked very well.

closed 2021-05-15 by 1am