bluez/bluez #138: profiles/deviceinfo/deviceinfo.c - Invalid pdu length recived

TheNerdyHamster, 2021-05-10:
Hi,

I recently got a PELTOR WS ALERT XPI Headset, but when I try to connect it to my laptop, but works fine on android and windows.
Tried both with pulseaudio and pipewire, no difference.

Journalctl:
```
May 10 22:21:16 tnh-comp systemd[1]: Starting Bluetooth service...
May 10 22:21:16 tnh-comp bluetoothd[1437230]: Bluetooth daemon 5.58
May 10 22:21:16 tnh-comp systemd[1]: Started Bluetooth service.
May 10 22:21:16 tnh-comp bluetoothd[1437230]: Starting SDP server
May 10 22:21:16 tnh-comp bluetoothd[1437230]: Bluetooth management interface 1.19 initialized
May 10 22:21:16 tnh-comp bluetoothd[1437230]: Endpoint registered: sender=:1.296 path=/MediaEndpoint/A2DPSource/ldac
May 10 22:21:16 tnh-comp bluetoothd[1437230]: Endpoint registered: sender=:1.296 path=/MediaEndpoint/A2DPSink/aptx_hd
May 10 22:21:16 tnh-comp bluetoothd[1437230]: Endpoint registered: sender=:1.296 path=/MediaEndpoint/A2DPSource/aptx_hd
May 10 22:21:16 tnh-comp bluetoothd[1437230]: Endpoint registered: sender=:1.296 path=/MediaEndpoint/A2DPSink/aptx
May 10 22:21:16 tnh-comp bluetoothd[1437230]: Endpoint registered: sender=:1.296 path=/MediaEndpoint/A2DPSource/aptx
May 10 22:21:16 tnh-comp bluetoothd[1437230]: Endpoint registered: sender=:1.296 path=/MediaEndpoint/A2DPSource/aac
May 10 22:21:16 tnh-comp bluetoothd[1437230]: Endpoint registered: sender=:1.296 path=/MediaEndpoint/A2DPSink/sbc
May 10 22:21:16 tnh-comp bluetoothd[1437230]: Endpoint registered: sender=:1.296 path=/MediaEndpoint/A2DPSource/sbc
May 10 22:23:32 tnh-comp bluetoothd[1437230]: profiles/deviceinfo/deviceinfo.c:read_pnpid_cb() Error reading PNP_ID: Invalid pdu length received
```

Bluez version: 5.58
Kernel version: 5.12
Distro: Archlinux

Please tell me if you need more info.

Vudentz, 2021-05-15:
Please add the btmon logs, I don't think this is related to the pnpid it is more likely some other problem.

TheNerdyHamster, 2021-05-15:
```
                                                                                    #141 [hci0] 120.734595
      LE Set Extended Scan Parameters (0x08|0x0041) ncmd 1
        Status: Success (0x00)
< HCI Command: LE Set Extended Scan Enable (0x08|0x0042) plen 6                                                                                                                               #142 [hci0] 120.734617
        Extended scan: Enabled (0x01)
        Filter duplicates: Enabled (0x01)
        Duration: 0 msec (0x0000)
        Period: 0.00 sec (0x0000)
> HCI Event: Command Complete (0x0e) plen 4                                                                                                                                                   #143 [hci0] 120.736638
      LE Set Extended Scan Enable (0x08|0x0042) ncmd 2
        Status: Success (0x00)
> HCI Event: LE Meta Event (0x3e) plen 57                                                                                                                                                     #144 [hci0] 120.750622
      LE Extended Advertising Report (0x0d)
        Num reports: 1
        Entry 0
          Event type: 0x0013
            Props: 0x0013
              Connectable
              Scannable
              Use legacy advertising PDUs
            Data status: Complete
          Legacy PDU Type: ADV_IND (0x0013)
          Address type: Public (0x00)
          Address: 00:15:F3:24:B4:D3 (PELTOR AB)
          Primary PHY: LE 1M
          Secondary PHY: No packets
          SID: no ADI field (0xff)
          TX power: 127 dBm
          RSSI: -51 dBm (0xcd)
          Periodic advertising invteral: 0.00 msec (0x0000)
          Direct address type: Public (0x00)
          Direct address: 00:00:00:00:00:00 (OUI 00-00-00)
          Data length: 0x1f
        02 01 19 03 03 53 fe 09 ff d0 02 00 00 00 00 00  .....S..........
        00 0d 09 33 4d 20 43 48 31 2d 34 44 33 20 20     ...3M CH1-4D3
< HCI Command: LE Set Extended Scan Enable (0x08|0x0042) plen 6                                                                                                                               #145 [hci0] 120.750757
        Extended scan: Disabled (0x00)
        Filter duplicates: Disabled (0x00)
        Duration: 0 msec (0x0000)
        Period: 0.00 sec (0x0000)
> HCI Event: Command Complete (0x0e) plen 4                                                                                                                                                   #146 [hci0] 120.753598
      LE Set Extended Scan Enable (0x08|0x0042) ncmd 2
        Status: Success (0x00)
< HCI Command: LE Extended Create Connection (0x08|0x0043) plen 26                                                                                                                            #147 [hci0] 120.753661
        Filter policy: White list is not used (0x00)
        Own address type: Public (0x00)
        Peer address type: Public (0x00)
        Peer address: 00:15:F3:24:B4:D3 (PELTOR AB)
        Initiating PHYs: 0x01
        Entry 0: LE 1M
          Scan interval: 60.000 msec (0x0060)
          Scan window: 60.000 msec (0x0060)
          Min connection interval: 90.00 msec (0x0048)
          Max connection interval: 110.00 msec (0x0058)
          Connection latency: 4 (0x0004)
          Supervision timeout: 8000 msec (0x0320)
          Min connection length: 0.000 msec (0x0000)
          Max connection length: 0.000 msec (0x0000)
> HCI Event: Command Status (0x0f) plen 4                                                                                                                                                     #148 [hci0] 120.755603
      LE Extended Create Connection (0x08|0x0043) ncmd 2
        Status: Success (0x00)
> HCI Event: LE Meta Event (0x3e) plen 31                                                                                                                                                     #149 [hci0] 120.783608
      LE Enhanced Connection Complete (0x0a)
        Status: Success (0x00)
        Handle: 3585
        Role: Master (0x00)
        Peer address type: Public (0x00)
        Peer address: 00:15:F3:24:B4:D3 (PELTOR AB)
        Local resolvable private address: 00:00:00:00:00:00 (Non-Resolvable)
[... 81 lines cut ...]
        Status: Success (0x00)
        Handle: 3585
        Connection interval: 105.00 msec (0x0054)
        Connection latency: 4 (0x0004)
        Supervision timeout: 8000 msec (0x0320)
> HCI Event: Encryption Change (0x08) plen 4                                                                                                                                                  #163 [hci0] 124.104609
        Status: PIN or Key Missing (0x06)
        Handle: 3585
        Encryption: Disabled (0x00)
< HCI Command: Disconnect (0x01|0x0006) plen 3                                                                                                                                                #164 [hci0] 124.104662
        Handle: 3585
        Reason: Authentication Failure (0x05)
> HCI Event: Command Status (0x0f) plen 4                                                                                                                                                     #165 [hci0] 124.106603
      Disconnect (0x01|0x0006) ncmd 1
        Status: Success (0x00)
[... cut ...]
        Status: Success (0x00)
        Handle: 3585
        Reason: Connection Terminated By Local Host (0x16)
@ MGMT Event: Device Disconnected (0x000c) plen 8                                                                                                                                         {0x0001} [hci0] 124.209668
        LE Address: 00:15:F3:24:B4:D3 (PELTOR AB)
        Reason: Connection terminated due to authentication failure (0x04)
```

Seems like an authentication problem to the device?

Vudentz, 2021-05-18:
Looks like the remote has removed the link key, you will need to repair.

```
> HCI Event: Encryption Change (0x08) plen 4                                                                                                                                                  #163 [hci0] 124.104609
        Status: PIN or Key Missing (0x06)
        Handle: 3585
        Encryption: Disabled (0x00)
```

TheNerdyHamster, 2021-06-02:
Ohh, that kinda explains the problem!

Is their away to disable the Pin/Key requierment from my side on my host machine? Or is it any otherway that this can get solved?

Already tried factory reset etc.
Need to study how bluetooth work this summer :D

Thx before hand!

closed 2025-08-27 by github-actions[bot]