bluez/bluez #145: bluetoothd segfault at 20 ip

KingsWorldProject, 2021-05-20:
5月 18 19:25:23 uos-PC bluetoothd[14578]: src/device.c:btd_device_set_temporary() temporary 1
5月 18 19:25:23 uos-PC bluetoothd[14578]: src/device.c:device_remove() Removing device /org/bluez/hci0/dev_6C_5D_63_2B_69_9C
5月 18 19:25:23 uos-PC bluetoothd[14578]: src/service.c:change_state() 0xc667c0: device 6C:5D:63:2B:69:9C profile input-hid state changed: disconnected -> unavailable (0)
5月 18 19:25:23 uos-PC bluetoothd[14578]: src/service.c:btd_service_unref() 0xc667c0: ref=2
5月 18 19:25:23 uos-PC bluetoothd[14578]: profiles/input/device.c:input_device_unregister() /org/bluez/hci0/dev_6C_5D_63_2B_69_9C
5月 18 19:25:23 uos-PC bluetoothd[14578]: src/service.c:btd_service_unref() 0xc667c0: ref=1
5月 18 19:25:23 uos-PC bluetoothd[14578]: src/service.c:btd_service_unref() 0xc667c0: ref=0
5月 18 19:25:23 uos-PC bluetoothd[14578]: src/device.c:btd_device_unref() Freeing device /org/bluez/hci0/dev_6C_5D_63_2B_69_9C
5月 18 19:25:23 uos-PC bluetoothd[14578]: src/device.c:device_free() 0xc54c80
5月 18 19:25:23 uos-PC bluetoothd[14578]: src/adapter.c:remove_whitelist_complete() 6C:5D:63:2B:69:9C removed from kernel whitelist
5月 18 19:25:23 uos-PC bluetoothd[14578]: src/adapter.c:connect_failed_callback() hci0 6C:5D:63:2B:69:9C status 2
5月 18 19:25:23 uos-PC bluetoothd[14578]: src/adapter.c:bonding_attempt_complete() hci0 bdaddr 6C:5D:63:2B:69:9C type 0 status 0x2
5月 18 19:25:23 uos-PC bluetoothd[14578]: src/adapter.c:resume_discovery()
5月 18 19:25:23 uos-PC bluetoothd[14578]: src/adapter.c:trigger_start_discovery()
5月 18 19:25:23 uos-PC bluetoothd[14578]: src/adapter.c:cancel_passive_scanning()
5月 18 19:25:23 uos-PC bluetoothd[14578]: connect error: Transport endpoint is not connected (107)
5月 18 19:25:23 uos-PC kernel: bluetoothd[14578]: segfault at 20 ip 0000000000473c20 sp 00007fffe13d6918 error 4 in bluetoothd[40b000+a0000]
5月 18 19:25:23 uos-PC kernel: Code: e8 b5 74 f9 ff 48 89 df 48 89 05 db 61 09 00 e8 46 74 f9 ff b8 01 00 00 00 5b c3 31 c0 c3 66 66 2e 0f 1f 84 00 00 00 00 00 90 <8b> 47 20 83 e8 01 83 f8 01 77 25 85 f6 75 11 31 d2 be 03 00 00 00

Vudentz, 2021-05-20:
What version?

KingsWorldProject, 2021-05-20:
5.50

> What version?

Vudentz, 2021-05-25:
> 5.50
> 
> > What version?

Can you try with a more recent version, 5.50 is really old.

closed 2021-09-01 by Vudentz