bluez/bluez #146: Bluez 5.58 scan device removal issue

Vishal8523, 2021-05-20:
Bluez Version: 5.58

When the scan is performed each time for the same agent, the devices are seen as NEW devices.
Tried to disable by setting the `duplicate-data on/off` but the devices were removed after `scan off` every time.

Is there a way to not remove the devices after every scan ?

```
root@iot2000:~# bluetoothctl
Agent registered
[CHG] Controller BC:54:2F:4D:A5:D9 Pairable: yes
[bluetooth]# version
Version 5.58
[bluetooth]# power off
Changing power off succeeded
[bluetooth]# power on
[CHG] Controller BC:54:2F:4D:A5:D9 Class: 0x00000104
Changing power on succeeded
[CHG] Controller BC:54:2F:4D:A5:D9 Powered: yes
[bluetooth]# scan on
Discovery started
[CHG] Controller BC:54:2F:4D:A5:D9 Discovering: yes
**[NEW] Device AA:BB:CC:DD:EE:01 Samsung
[NEW] Device FF:EE:DD:CC:BB:00 Samsung
[NEW] Device C8:28:32:E1:8E:13 TV**
[bluetooth]# scan off
Discovery stopped
[CHG] Device C8:28:32:E1:8E:13 RSSI is nil
[CHG] Device FF:EE:DD:CC:BB:00 TxPower is nil
[CHG] Device FF:EE:DD:CC:BB:00 RSSI is nil
[CHG] Device AA:BB:CC:DD:EE:01 TxPower is nil
[CHG] Device AA:BB:CC:DD:EE:01 RSSI is nil
[CHG] Controller BC:54:2F:4D:A5:D9 Discovering: no
**[DEL] Device AA:BB:CC:DD:EE:01 Samsung
[DEL] Device FF:EE:DD:CC:BB:00 Samsung
[DEL] Device C8:28:32:E1:8E:13 TV**
[bluetooth]# scan on
Discovery started
[CHG] Controller BC:54:2F:4D:A5:D9 Discovering: yes
**[NEW] Device AA:BB:CC:DD:EE:01 Samsung
[NEW] Device FF:EE:DD:CC:BB:00 Samsung
[NEW] Device C8:28:32:E1:8E:13 TV
[NEW] Device 5D:60:69:03:9D:F9 5D-60-69-03-9D-F9**
[bluetooth]# scan off
Discovery stopped
[CHG] Device 5D:60:69:03:9D:F9 TxPower is nil
[CHG] Device 5D:60:69:03:9D:F9 RSSI is nil
[CHG] Device C8:28:32:E1:8E:13 RSSI is nil
[CHG] Device FF:EE:DD:CC:BB:00 TxPower is nil
[CHG] Device FF:EE:DD:CC:BB:00 RSSI is nil
[CHG] Device AA:BB:CC:DD:EE:01 TxPower is nil
[CHG] Device AA:BB:CC:DD:EE:01 RSSI is nil
[CHG] Controller BC:54:2F:4D:A5:D9 Discovering: no
**[DEL] Device AA:BB:CC:DD:EE:01 Samsung
[DEL] Device FF:EE:DD:CC:BB:00 Samsung
[DEL] Device C8:28:32:E1:8E:13 TV
[DEL] Device 5D:60:69:03:9D:F9 5D-60-69-03-9D-F9**
[bluetooth]#
```

Appreciate your help in this regards.

Vishal8523, 2021-05-20:
**Additional Update:**
Updated the main.conf with the flag TemporaryTimeout
The scenario that I am interested is to not to remove any Temprorary Devices 

Appreciate any inputs here.

Vudentz, 2021-05-20:
> **Additional Update:**
> Updated the main.conf with the flag _TemporaryTimeout_
> The scenario that I am interested is to **not to remove any Temporary Devices**
> 
> Appreciate any inputs here.

Are you setting it to 0? It appears that is not doing what the documentation seems to imply which is not never remove them, perhaps use a big value, not though that if you restart the daemon the temporary would still be gone.

closed 2021-06-18 by BluezTestBot

Vishal8523, 2021-06-30:
@Vudentz : Is a fix expected for this issue in the upcoming releases ?

thorleifjacobsen, 2023-03-29:
Have this been fixed? I'm seeing the exact same issue. 

```
[CHG] Device C7:97:F8:XX:XX:XX RSSI is nil
[DEL] Device C7:97:F8:XX:XX:XX P mesh
```
Everytime I do "scan off". If i connet to it while scanning it stays until I disconnect. But I expected this to stay in the "devices" list until the timeout atleast.. But this instantly goes on "scan off"