bluez/bluez #147: BleMesh : Create an unprovisionned device

neicureuil, 2021-05-20:
Is it possible to use BlueZ Mesh API to create an unprovisionned node that will be provisionned by an other BluezMesh application on an other linux device ?

I have successfully made a node provisionner using the Mesh API but not an unprovisionned node.
I have tried to make a node application with no provisionner and provisionning interface, but the other devices don't see it.

(I have made that with the dbus api, is it possible to use an other thing that the dbus api ? like a dynamic library ?)

isakwestin, 2021-05-20:
Take a look at the `Join` dbus method.

https://github.com/bluez/bluez/blob/f7a568d72263cead9a69ab18ca27897df9943633/doc/mesh-api.txt#L11

When calling `Join` your node starts to broadcast the Unprovisioned Device Beacon which other provisioners (e.g. another linux device or phone app) can detect to initiate the provisioning.

neicureuil, 2021-05-20:
Hi,
Thanks for your reply.
I have just tried to call the `Join` method and it seems to do a thing, but I can't see my devices on other devices (I have tried on android tablet with the applications : nRF Mesh and ST BLE Mesh). 

Do you think it's a app problem (and which app do you advise to me ?) or a problem with my code ?

neicureuil, 2021-05-20:
Ok, I have tried with 2 linux device (one provisioner and one node) and it seems to work (thanks). 
However, when I tried to provisionning my node with my provisionner I got a `timeout` error on the node and `cannot-assign-addresses` error on the provisioner .... 
What cause this errors ?

Futhermore, once the node is provisioned, how to send a message and how to define the models ?

neicureuil, 2021-05-20:
An other question, what is the difference betwen `Send` and `Publish` on the node interface ?
If I want to send a message from my provisioner to the provisioned node, which one can I use ?

neicureuil, 2021-05-20:
Some time, when miraculously the unprovisioned node succesfully provisionned, I tried to send with the `Send` method a message from the provisionner to the node, but I get the error `[org.bluez.mesh.Error.InvalidArgs] Key not found`.

How can I get this key ?

closed 2021-05-21 by neicureuil