bluez/bluez #154: Immediate disconnect with Soundcore Life P2 (5.59)

sents, 2021-06-15:
When connecting to my Soundcore Life P2 blutooth earphones the connection lasts for a couple seconds and then disconnects again. The most immediate error message I could find was:
```
= bluetoothd: profiles/audio/avdtp.c:avdtp_parse_rej() SET_CONFIGURATION request rejected: Configuration not supported (41)                                                              11.490967
= bluetoothd: src/service.c:btd_service_connect() a2dp-source profile connect failed for AC:12:2F:72:B6:1D: Device or resource busy                                                      11.491006
```
I am using an up to date Arch Linux with bluez-5.59.

<details><summary>bluetoothd log</summary>
<pre><code>
bluetoothd[4778]: src/device.c:connect_profiles() /org/bluez/hci0/dev_AC_12_2F_72_B6_1D (all), client :1.140
bluetoothd[4778]: profiles/audio/a2dp.c:a2dp_sink_connect() path /org/bluez/hci0/dev_AC_12_2F_72_B6_1D
bluetoothd[4778]: profiles/audio/avdtp.c:avdtp_ref() 0x559a532c36a0: ref=1
bluetoothd[4778]: profiles/audio/avdtp.c:avdtp_ref() 0x559a532c36a0: ref=2
bluetoothd[4778]: profiles/audio/a2dp.c:setup_ref() 0x559a532a0580: ref=1
bluetoothd[4778]: profiles/audio/source.c:source_set_state() State changed /org/bluez/hci0/dev_AC_12_2F_72_B6_1D: SOURCE_STATE_DISCONNECTED -> SOURCE_STATE_CONNECTING
bluetoothd[4778]: profiles/audio/sink.c:sink_set_state() State changed /org/bluez/hci0/dev_AC_12_2F_72_B6_1D: SINK_STATE_DISCONNECTED -> SINK_STATE_CONNECTING
bluetoothd[4778]: profiles/audio/sink.c:sink_connect() stream creation in progress
bluetoothd[4778]: src/service.c:change_state() 0x559a532b33c0: device AC:12:2F:72:B6:1D profile a2dp-sink state changed: disconnected -> connecting (0)
bluetoothd[4778]: src/adapter.c:connected_callback() hci0 device AC:12:2F:72:B6:1D connected eir_len 19
bluetoothd[4778]: profiles/audio/avdtp.c:avdtp_connect_cb() AVDTP: connected signaling channel to AC:12:2F:72:B6:1D
bluetoothd[4778]: profiles/audio/avdtp.c:avdtp_connect_cb() AVDTP imtu=672, omtu=895
bluetoothd[4778]: profiles/audio/avdtp.c:avdtp_register_remote_sep() seid 4 type 1 media 0 delay_reporting false
bluetoothd[4778]: profiles/audio/a2dp.c:register_remote_sep() Found remote SEP: /org/bluez/hci0/dev_AC_12_2F_72_B6_1D/sep4
bluetoothd[4778]: profiles/audio/avdtp.c:avdtp_register_remote_sep() seid 3 type 1 media 0 delay_reporting false
bluetoothd[4778]: profiles/audio/a2dp.c:register_remote_sep() Found remote SEP: /org/bluez/hci0/dev_AC_12_2F_72_B6_1D/sep3
bluetoothd[4778]: profiles/audio/avdtp.c:avdtp_register_remote_sep() seid 2 type 1 media 0 delay_reporting false
bluetoothd[4778]: profiles/audio/a2dp.c:register_remote_sep() Found remote SEP: /org/bluez/hci0/dev_AC_12_2F_72_B6_1D/sep2
bluetoothd[4778]: profiles/audio/avdtp.c:avdtp_register_remote_sep() seid 12 type 1 media 0 delay_reporting false
bluetoothd[4778]: profiles/audio/a2dp.c:register_remote_sep() Found remote SEP: /org/bluez/hci0/dev_AC_12_2F_72_B6_1D/sep12
bluetoothd[4778]: profiles/audio/avdtp.c:avdtp_register_remote_sep() seid 11 type 1 media 0 delay_reporting false
bluetoothd[4778]: profiles/audio/a2dp.c:register_remote_sep() Found remote SEP: /org/bluez/hci0/dev_AC_12_2F_72_B6_1D/sep11
bluetoothd[4778]: profiles/audio/avdtp.c:avdtp_register_remote_sep() seid 10 type 1 media 0 delay_reporting false
bluetoothd[4778]: profiles/audio/a2dp.c:register_remote_sep() Found remote SEP: /org/bluez/hci0/dev_AC_12_2F_72_B6_1D/sep10
bluetoothd[4778]: profiles/audio/avdtp.c:avdtp_register_remote_sep() seid 8 type 0 media 0 delay_reporting false
bluetoothd[4778]: profiles/audio/a2dp.c:register_remote_sep() Found remote SEP: /org/bluez/hci0/dev_AC_12_2F_72_B6_1D/sep8
bluetoothd[4778]: profiles/audio/avdtp.c:avdtp_register_remote_sep() seid 7 type 0 media 0 delay_reporting false
bluetoothd[4778]: profiles/audio/a2dp.c:register_remote_sep() Found remote SEP: /org/bluez/hci0/dev_AC_12_2F_72_B6_1D/sep7
bluetoothd[4778]: profiles/audio/avdtp.c:avdtp_register_remote_sep() seid 6 type 0 media 0 delay_reporting false
bluetoothd[4778]: profiles/audio/a2dp.c:register_remote_sep() Found remote SEP: /org/bluez/hci0/dev_AC_12_2F_72_B6_1D/sep6
bluetoothd[4778]: profiles/audio/avdtp.c:avdtp_register_remote_sep() seid 1 type 0 media 0 delay_reporting false
bluetoothd[4778]: profiles/audio/a2dp.c:register_remote_sep() Found remote SEP: /org/bluez/hci0/dev_AC_12_2F_72_B6_1D/sep1
bluetoothd[4778]: profiles/audio/avdtp.c:session_cb()
bluetoothd[4778]: profiles/audio/avdtp.c:avdtp_parse_resp() DISCOVER request succeeded
bluetoothd[4778]: profiles/audio/avdtp.c:avdtp_discover_resp() seid 4 type 1 media 0 in use 0
bluetoothd[4778]: profiles/audio/avdtp.c:avdtp_discover_resp() seid 3 type 1 media 0 in use 0
bluetoothd[4778]: profiles/audio/avdtp.c:avdtp_discover_resp() seid 2 type 1 media 0 in use 0
bluetoothd[4778]: profiles/audio/avdtp.c:avdtp_discover_resp() seid 12 type 1 media 0 in use 0
bluetoothd[4778]: profiles/audio/avdtp.c:avdtp_discover_resp() seid 11 type 1 media 0 in use 0
bluetoothd[4778]: profiles/audio/avdtp.c:avdtp_discover_resp() seid 10 type 1 media 0 in use 0
bluetoothd[4778]: profiles/audio/avdtp.c:avdtp_discover_resp() seid 8 type 0 media 0 in use 0
bluetoothd[4778]: profiles/audio/avdtp.c:avdtp_discover_resp() seid 7 type 0 media 0 in use 0
bluetoothd[4778]: profiles/audio/avdtp.c:avdtp_discover_resp() seid 6 type 0 media 0 in use 0
bluetoothd[4778]: profiles/audio/avdtp.c:avdtp_discover_resp() seid 1 type 0 media 0 in use 0
bluetoothd[4778]: profiles/audio/a2dp.c:discover_cb() version 0x0103 err (nil)
bluetoothd[4778]: profiles/audio/sink.c:discovery_complete() Discovery complete
bluetoothd[4778]: profiles/audio/a2dp.c:setup_ref() 0x559a532a0580: ref=2
bluetoothd[4778]: profiles/audio/a2dp.c:setup_ref() 0x559a532a0580: ref=3
bluetoothd[4778]: profiles/audio/media.c:media_endpoint_async_call() Calling SelectConfiguration: name = :1.40 path = /MediaEndpoint/A2DPSource/sbc
bluetoothd[4778]: profiles/audio/a2dp.c:setup_unref() 0x559a532a0580: ref=2
bluetoothd[4778]: profiles/audio/a2dp.c:a2dp_config() a2dp_config: selected SEP 0x559a532be7f0
bluetoothd[4778]: profiles/audio/a2dp.c:setup_ref() 0x559a532a0580: ref=3
bluetoothd[4778]: profiles/audio/avdtp.c:avdtp_set_configuration() 0x559a532c36a0: int_seid=2, acp_seid=4
bluetoothd[4778]: profiles/audio/a2dp.c:setup_unref() 0x559a532a0580: ref=2
bluetoothd[4778]: profiles/audio/a2dp.c:setup_unref() 0x559a532a0580: ref=1
bluetoothd[4778]: profiles/audio/avdtp.c:session_cb()
bluetoothd[4778]: profiles/audio/avdtp.c:avdtp_parse_rej() SET_CONFIGURATION request rejected: Configuration not supported (41)
bluetoothd[4778]: profiles/audio/a2dp.c:setconf_cfm() Source 0x559a532be7f0: Set_Configuration_Cfm
bluetoothd[4778]: profiles/audio/a2dp.c:setup_ref() 0x559a532a0580: ref=2
bluetoothd[4778]: profiles/audio/avdtp.c:avdtp_unref() 0x559a532c36a0: ref=1
bluetoothd[4778]: src/service.c:change_state() 0x559a532b33c0: device AC:12:2F:72:B6:1D profile a2dp-sink state changed: connecting -> disconnected (-5)
[... 857 lines cut ...]
        Destination CID: 65
        Source CID: 1156
> HCI Event: Mode Change (0x14) plen 6                                                                                                                                        #97 [hci0] 16.275492
        Status: Success (0x00)
        Handle: 256
        Mode: Sniff (0x02)
        Interval: 500.000 msec (0x0320)
> HCI Event: Number of Completed Packets (0x13) plen 5                                                                                                                        #98 [hci0] 16.276498
        Num handles: 1
        Handle: 256
        Count: 1
> HCI Event: Mode Change (0x14) plen 6                                                                                                                                        #99 [hci0] 17.252488
        Status: Success (0x00)
        Handle: 256
        Mode: Active (0x00)
        Interval: 0.000 msec (0x0000)
> HCI Event: Disconnect Complete (0x05) plen 4                                                                                                                               #100 [hci0] 17.410586
        Status: Success (0x00)
        Handle: 256
        Reason: Remote User Terminated Connection (0x13)
@ MGMT Event: Device Disconnected (0x000c) plen 8                                                                                                                        {0x0001} [hci0] 17.410662
        BR/EDR Address: AC:12:2F:72:B6:1D (Fantasia Trading LLC)
        Reason: Connection terminated by remote host (0x03)
</code></pre>
</details>

Vudentz, 2021-06-15:
```
< ACL Data TX: Handle 256 flags 0x00 dlen 18                                                                                                                                  #61 [hci0] 11.401548
      Channel: 1025 len 14 [PSM 25 mode Basic (0x00)] {chan 0}
      AVDTP: Set Configuration (0x03) Command (0x00) type 0x00 label 3 nosp 0
        ACP SEID: 4
        INT SEID: 2
        Service Category: Media Transport (0x01)
        Service Category: Media Codec (0x07)
          Media Type: Audio (0x00)
          Media Codec: SBC (0x00)
            Frequency: 44100 (0x20)
            Channel Mode: Mono (0x08)
            Block Length: 16 (0x10)
            Subbands: 8 (0x04)
            Allocation Method: Loudness (0x01)
            Minimum Bitpool: 2
            Maximum Bitpool: 31
```

Looks like your device doesn't support the configuration above, the configuration is usually the last used configuration or in case it is a brand new device the sound server would be the entity configuring based on its own endpoints. Did this headset ever worked? Have you tried to repair?

sents, 2021-06-16:
The headset works for my android device but I've never successfully paired it using bluez on my computer. It's a new device and I have tried "resetting" the device in hope that it resets any faulty configuration. This didn't help.

I use pulseaudio. So you say pulseaudio might not be serving a working configuration? I'll try to find out more and try pipewire.

sents, 2021-06-16:
Ok, after trying a bunch of times I finally got it to connect with pipewire. It's still weird that when using pipewire the first few times it gave me the same unsupported configuration error as before.

sents, 2021-06-16:
I've now found out that a2dp also doesn't work with pipwire, I never have the option to choose an a2dp profile. Pulseaudio on the other hand seems to want to use a2dp. I've also tested it on a very similar machine (same laptop, same os, but with different bluetooth adapter) and there I can connect via pulseaudio with a2dp. Maybe that is what is going wrong. My bluetooth adapter is Intel 7260 (8087:07dc). This [thread](https://bugzilla.kernel.org/show_bug.cgi?id=203535) seems to indicate that the adapter does in principle support.

sents, 2021-06-16:
I've tested other bluetooth headphones and they work with a2dp without problems. The

joakim-tjernlund, 2021-06-20:
you may have to update to bluez git, https://github.com/bluez/bluez/issues/157

sents, 2021-06-21:
The patch related to #157 fixed my problem. Thank you!

closed 2021-06-21 by sents