bluez/bluez #155: [5.59] Xbox Series X/S Controller won't connect, enters connect/disconnect loop in pairing mode

Termuellinator, 2021-06-16:
Hi there,

i'm encountering issues with the newest family of xbox controllers when trying to use them wirelessly (via xpadneo driver).
The first pairing seems to go normal, the controller vibrates - but doesn't leave pairing mode and doesn't detect inputs.
When switiching the controller of and on after that, it won't connect. If you then enter pairing mode again, it will connect and disconnect rapidly until you leave pairing mode again.
I've tried it with two different bluetooth-devices (x570 Taichi Intel Corp. Wireless-AC 3168 and Realtek rtl8761b dongle, both BT-LE capable), `privacy` is set to `device` in `/etc/bluetooth/main.conf`,  i tried disabling ERTM (both on 5.10 and 5.12) - all with the same result. Only thing i haven't tried is updating firmware, will do that once i get my hand on a windows pc (but while https://github.com/bluez/bluez/issues/66 suggests it's a firmware issue, the linked reddit thread only mentiones that it "works" with the wrong firmware, but has massive lags then, so i'm not sure if thats really the case?)
I'm on Manjaro testing, currently on 5.13rc6, Bluez 5.59

This issue is already discussed in https://github.com/atar-axis/xpadneo/issues/295 , where the dev believes its maybe something in bluez

here's a log pairing the controller and then entering pairing mode (with the connect/disconnect loop) again. Let me know if any other info might be helpful.

https://github.com/atar-axis/xpadneo/files/6602450/btmon_from_start.log

RuiGuilherme, 2021-06-16:
I'm probably the same issue. 

**Model Number**: 1914
**Model**: Xbox Wireless Controller X/S Carbon Black  + USB-C Cable
[https://www.xbox.com/en-us/accessories/controllers/xbox-wireless-controller-usb-c](https://www.xbox.com/en-us/accessories/controllers/xbox-wireless-controller-usb-c)

<br />

>Only thing i haven't tried is updating firmware

I updated the firmware using Windows, and even so it did not solve the issue. 

Logs/Infos:
=======

#### $ uname -a
`Linux ArchLinux 5.12.10-arch1-1 #1 SMP PREEMPT Thu, 10 Jun 2021 16:34:50 +0000 x86_64 GNU/Linux`

#### $ journalctl -f
```
jun 14 00:27:05 ArchLinux bluetoothd[498]: profiles/deviceinfo/deviceinfo.c:read_pnpid_cb() Error reading PNP_ID value: Request attribute has encountered an unlikely error
jun 14 00:27:05 ArchLinux bluetoothd[498]: profiles/deviceinfo/dis.c:read_pnpid_cb() Error reading PNP_ID value: Request attribute has encountered an unlikely error
jun 14 00:27:05 ArchLinux bluetoothd[498]: profiles/input/hog-lib.c:info_read_cb() HID Information read failed: Request attribute has encountered an unlikely error
jun 14 00:27:05 ArchLinux bluetoothd[498]: profiles/input/hog-lib.c:report_map_read_cb() Report Map read failed: Request attribute has encountered an unlikely error
jun 14 00:27:05 ArchLinux bluetoothd[498]: profiles/input/hog-lib.c:report_reference_cb() Read Report Reference descriptor failed: Request attribute has encountered an unlikely error
jun 14 00:27:05 ArchLinux bluetoothd[498]: profiles/input/hog-lib.c:report_reference_cb() Read Report Reference descriptor failed: Request attribute has encountered an unlikely error
```

#### # bluetooth
```
# pair 44:16:22:C5:68:4A
    Attempting to pair with 44:16:22:C5:68:4A
    [CHG] Device 44:16:22:C5:68:4A Connected: yes
    Failed to pair: org.bluez.Error.AuthenticationFailed
    [CHG] Device 44:16:22:C5:68:4A Connected: no
```

#### # btmgmt

```
[] Accept pairing with 44:16:22:C5:68:4A (yes/no) >>  yes
    User Confirm Reply successful
    hci0 44:16:22:C5:68:4A type LE Public connected eir_len 14
    hci0 44:16:22:C5:68:4A User Confirm 339245 hint 1
    hci0 44:16:22:C5:68:4A auth failed with status 0x05 (Authentication Failed)
    hci0 44:16:22:C5:68:4A type LE Public disconnected with reason 2
    hci0 44:16:22:C5:68:4A type BR/EDR connect failed (status 0x04, Connect Failed)
    hci0 44:16:22:C5:68:4A type BR/EDR connect failed (status 0x04, Connect Failed)
```

kakra, 2021-06-17:
The errors went away for me after I compiled the `algif_hash` and `algif_skcipher` modules into the kernel. All the logged errors from above (`journalctl -f`) went away for me then.

Previously, loading the bluetooth.service would log:
```
Jun 16 21:10:18 jupiter bluetoothd[724]: Bluetooth management interface 1.18 initialized
Jun 16 21:10:18 jupiter bluetoothd[724]: No IRK stored
Jun 16 21:10:18 jupiter bluetoothd[724]: src/adapter.c:generate_and_write_irk() Failed to open crypto
```

while it should look like this:
```
Jun 17 05:52:47 jupiter bluetoothd[101932]: Bluetooth management interface 1.18 initialized
Jun 17 05:52:47 jupiter bluetoothd[101932]: No IRK stored
Jun 17 05:52:47 jupiter bluetoothd[101932]: Generated IRK successfully
```

So in the end, it may be possible that the Arch kernel doesn't enable all the necessary bits of the crypto user API in the kernel.

kakra, 2021-06-17:
These however

> ```
> jun 14 00:27:05 ArchLinux bluetoothd[498]: profiles/deviceinfo/deviceinfo.c:read_pnpid_cb() Error reading PNP_ID value: Request attribute has encountered an unlikely error
> jun 14 00:27:05 [...]

can still be observed when the device disconnects due to idle time. So it's probably not the issue itself but just the fact that the gamepad entered the reconnect loop.

However, I believe these errors should be properly caught when the device disconnects from Bluetooth, and not generate/log such errors. But I don't think these contribute to the issue, the missing bit is just "Generated IRK successfully".

Termuellinator, 2021-06-17:
Due to @kakra 's findings i verified that CONFIG_CRYPTO_USER_API_HASH was available and loaded and because of that i just tried again and the controller connected as expected.
I'll have to investigate if that is something that has changed in 5.13 rc6 (i think rc5 was the last i tested) or something that changed in yesterdays manjaro testing update.
I'll report my findings here.

RuiGuilherme, 2021-06-17:
@kakra 
I will create an new issue. 

#### I'll leave just some more information about this my issue: 
1. I tried **linux**, **linux-zen** and **linux-lts**.
2. Disable and Enable ertm.
3. Change Privacy to "device" and "on"
4. [Problems with all BLE devices on kernel 5.9+](https://wiki.archlinux.org/title/Bluetooth#Problems_with_all_BLE_devices_on_kernel_5.9+)
5. Disable **BR/EDR**
6. Enable **le advertising**
7. Check modules algif_**hash**/**skcipher**

I made this video showing the attempt to connect:
[https://youtu.be/juTWDAu_rGw](https://youtu.be/juTWDAu_rGw)
After this failure it enter connect/disconnect loop if I activate the pairing mode.

Termuellinator, 2021-06-17:
So it really seems that 5.13rc6 changed something - in 5.12 i can't connect. Although things are different - the connect/disconnect loop in pairing mode is gone.
When i now try to pair the controller, i get:
```
[bluetooth]# pair 44:16:22:D1:40:A2
Attempting to pair with 44:16:22:D1:40:A2
[CHG] Device 44:16:22:D1:40:A2 Connected: yes
Failed to pair: org.bluez.Error.AuthenticationFailed
[CHG] Device 44:16:22:D1:40:A2 Connected: no
```

on 5.13 i can pair it and it connects properly, if i turn off the controller it automatically connects after turning it on again - as long as i don't reboot. After reboot, i have to cancel the pairing and re-pair to make it work again. Plus, i get about i'd say 100-200ms lag in input, kind of as described in the mentioned reddit post, too.

kakra, 2021-06-17:
The delay is described in #156

Vudentz, 2021-06-17:
The following is the problem:

```
> ACL Data RX: Handle 3585 flags 0x02 dlen 6             #251 [hci0] 108.841520
      SMP: Pairing Failed (0x05) len 1
        Reason: Confirm value failed (0x04)
@ MGMT Event: Authentication Failed (0x0011) plen 8  {0x0001} [hci0] 108.841543
        LE Address: 44:16:22:D1:40:A2 (Microsoft Corporation)
        Status: Authentication Failed (0x05)
```

I don't recall having any changes on the pairing procedure in a while to cause something like this, besides it seems device specific, but perhaps this is due to some changes in the kernel crypto modules.

kakra, 2021-06-18:
FTR, it paired for me with kernel 5.10.43 LTS but only with the aforementioned kernel modules included in the kernel config. As a first step, it could make sense to improve the errors messages during service load time: When IRK fails, it could (a) also log the errno and path name of the system call, and (b) maybe hint for the missing modules if the errno indicates that a missing crypto module may be the issue.

Termuellinator, 2021-06-18:
I finally managed to get my controllers firmware updated on a windows pc, after that i connected and was pleasantly surprised that i couldn't see any lag anymore, the reconnect worked fine and everything.
Unfortunately, that only lasted until reboot, now i can't even pair after `cancel-pairing` but i'm getting the same error as on 5.12:
```
bluetooth]# pair 44:16:22:D1:40:A2
Attempting to pair with 44:16:22:D1:40:A2
[CHG] Device 44:16:22:D1:40:A2 Connected: yes
Failed to pair: org.bluez.Error.AuthenticationFailed
[CHG] Device 44:16:22:D1:40:A2 Connected: no
```
Haven't tried another reboot yet, but thought i'd add this here...

kakra, 2021-06-18:
You may need to `remove $MAC` and pair it again, or pair it to another system via Bluetooth, then try again on the original system.

YinXunzhe, 2021-07-26:
Try to enable LE Privacy:

sudo btmgmt power off
sudo btmgmt privacy on
sudo btmgmt power on

kakra, 2021-08-12:
> Try to enable LE Privacy:

Do not do this, stick with the defaults. Privacy does not seem to be needed. In fact, the Xbox Controller connection tends to become unstable with this feature enabled. I believe this is just sidestepping a problem in another place.

Current evidence shows that this is more likely a problem in the cache of bluez. Removing the cache from `/var/lib/bluetooth/$MAC` and restarting the service usually is enough to make the gamepad finally pair just fine, and even auto-connect just fine when turning it off and on in later attempts.

So the connection may finally be fixed in latest bluez versions. But there seems to be an artifact in the cache from older versions which breaks compatibility with the gamepad.

DataBeaver, 2022-01-06:
So far the only way I've been able to get this type of gamepad work is the following:
```
service bluetooth start
btmgmt power off
btmgmt privacy on
btmgmt power on
bluetoothctl
> scan on
(turn on the gamepad and enter pairing mode, wait for it to be detected)
> scan off
> pair <address>
> connect <address>
> trust <address>
(power-cycle the gamepad)
[agent] Accept pairing (yes/no): yes
```

This produces the following messages in syslog:
```
Jan  6 14:01:17 marmot bluetoothd[4969]: Bluetooth daemon 5.62
Jan  6 14:01:17 marmot bluetoothd[4969]: Starting SDP server
Jan  6 14:01:17 marmot bluetoothd[4969]: Bluetooth management interface 1.21 initialized
Jan  6 14:01:17 marmot bluetoothd[4969]: profiles/sap/server.c:sap_server_register() Sap driver initialization failed.
Jan  6 14:01:17 marmot bluetoothd[4969]: sap-server: Operation not permitted (1)
Jan  6 14:02:01 marmot bluetoothd[4969]: src/device.c:load_gatt_db() No cache for F4:6A:D7:1A:50:7C
Jan  6 14:02:02 marmot kernel: Bluetooth: hci0: unexpected SMP command 0x08 from f4:6a:d7:1a:50:7c
Jan  6 14:02:02 marmot kernel: Bluetooth: hci0: unexpected SMP command 0x09 from f4:6a:d7:1a:50:7c
Jan  6 14:02:03 marmot bluetoothd[4969]: src/service.c:service_accept() input-hog profile accept failed for F4:6A:D7:1A:50:7C
Jan  6 14:02:44 marmot kernel: input: Xbox Wireless Controller as /devices/virtual/misc/uhid/0005:045E:0B13.0066/input/input124
Jan  6 14:02:44 marmot kernel: hid-generic 0005:045E:0B13.0066: input,hidraw4: BLUETOOTH HID v5.09 Gamepad [Xbox Wireless Controller] on 5c:f3:70:83:7c:ee
```

It doesn't always go exactly the same way; sometimes the gamepad successfully connects on the first attempt and the "unexpected SMP command" and "input-hog profile accept failed" messages don't appear in the log.  Other times it requires more than one attempt to connect.  Once the connection is successful, it will reconnect as long as bluetoothd is not restarted.

If I do not enable the privacy feature with btmgmt, bluez thinks the device is paired successfully, device nodes are created, but the gamepad never exits pairing mode and there's no input.  Power-cycling the gamepad causes it to enter a connect/disconnect loop, with the errors from https://github.com/bluez/bluez/issues/155#issuecomment-863011065 spammed into syslog.

This is on Linux 5.15.4 (custom configuration) and bluez 5.62, with an ASUS BT400 bluetooth dongle (0b05:17cb).

Other things I've tried:
The uhid module was necessary to get the gamepad to work
The algif modules mentioned by kakra did not seem to have any effect
Setting Privacy="device" in bluez configuration does not help
Clearing the cache from /var/lib/bluetooth didn't help

So if I'm not supposed to enable the privacy feature, and removing the cache does not help, what else could I try?

kakra, 2022-01-06:
> So if I'm not supposed to enable the privacy feature, and removing the cache does not help, what else could I try?

How does the dongle show up in `lsusb`?

At a first glance it looks like it might be a problem at the BT driver level (unexpected SMP command). So if possible, maybe give a CSR dongle a try.

DataBeaver, 2022-01-06:
Bus 008 Device 002: ID 0b05:17cb ASUSTek Computer, Inc. Broadcom BCM20702A0 Bluetooth

In another computer I have a built-in bluetooth chip which looks like this:

Bus 001 Device 004: ID 8087:0a2a Intel Corp. Bluetooth wireless interface

However the last I tried, that one had massive trouble communicating with Xbox controllers - even the Xbox One models which work flawlessly on the Asus dongle.  It was Debian's stock kernel though, so I don't know if my custom one might fare better.

Do you have reliable product links for a CSR dongle in the EU?  I found some products with "CSR 4.0" stamped on them, but Chinese manufacturers are prone to labeling their products with whatever sells...

kakra, 2022-01-06:
This one works for me: [TP-LINK BT400](https://www.amazon.de/TP-Link-UB400-Bluetooth-Computer-unterst%C3%BCtzt-Black-Gold/dp/B07NQ5YGDW/ref=sr_1_1?__mk_de_DE=%C3%85M%C3%85%C5%BD%C3%95%C3%91&crid=3JOC70BRBTSX7&keywords=tp+link+bt400&qid=1641504109&sprefix=tp+link+bt400%2Caps%2C85&sr=8-1)

The AX200 chips (Intel) have problems even on Windows as far as I could test - not only wiht Xbox controllers. The problems are very similar if not exactly the same.

Early in the project, BCM chipset were also problematic. You may have found a bug which may be worth reporting to the bluez project. But I'd first try if a CSR-based dongle shows the same problems.

DataBeaver, 2022-01-06:
Looks like that one is available in my local stores as well.  I'll obtain one and report back on how it works, hopefully next week.

DataBeaver, 2022-01-15:
Got the new device.  It looks like this:

Bus 008 Device 002: ID 0a12:0001 Cambridge Silicon Radio, Ltd Bluetooth Dongle (HCI mode)

It seems to pair with the gamepad reliably on the first try, without SMP errors, though my sample size here isn't exactly large (three tries).  However it does require privacy mode to be enabled and won't reconnect after restarting bluetoothd, just like the Asus dongle.

kakra, 2022-01-15:
It doesn't require privacy mode here... It may be an attribute caching issue. Did you reset the bluetooth stack (i.e., clear its state and directories to purge any old/damaged info)? Did you try changing the GATT cache option in bluez? Also, it may depend on the bluez version, mine was 5.62 as of testing, and now 5.63 (didn't test it yet).

Maybe there's a different issue: Maybe pairing needs privacy mode, but later you can just revert the privacy setting and normal re-connect does still work? I'm not sure if I tried that, maybe you can.

DataBeaver, 2022-01-15:
The new adapter didn't have a cache directory to begin with.  I tried to delete the cache directory for the old adapter from /var/lib/bluetooth at one point, but it did not help.  The remove command from bluetoothctl removes the device-specific cache directory, and is required for re-pairing to work.

Disabling GATT cache does not affect the initial pairing, nor reconnections while bluetoothd stays up.  After restarting bluetoothd it doesn't react at all to powering on the gamepad if GATT cache is disabled.

After restarting bluetoothd, reconnecting does not work whether or not privacy mode is enabled.

I also have bluez version 5.62 on the machine I'm testing this on.

I noticed that the log messages for creating the devices for the older model gamepad look slightly different:
```
Jan 14 21:28:04 marmot kernel: input: Xbox Wireless Controller as /devices/pci0000:00/0000:00:1d.2/usb8/8-1/8-1:1.0/bluetooth/hci0/hci0:11/0005:045E:02FD.0005/input/input27
Jan 14 21:28:04 marmot kernel: input: Xbox Wireless Controller Consumer Control as /devices/pci0000:00/0000:00:1d.2/usb8/8-1/8-1:1.0/bluetooth/hci0/hci0:11/0005:045E:02FD.0005/input/input28
Jan 14 21:28:04 marmot kernel: hid-generic 0005:045E:02FD.0005: input,hidraw4: BLUETOOTH HID v9.03 Gamepad [Xbox Wireless Controller] on 5c:f3:70:83:7c:ee
```
The device ID is different as can be expected, but what's up with the HID version?  Why would the newer device have an older version?  Or is that a device-specific version which is just placed confusingly in the message?

There's an extra consumer control device as well, which the newer pad doesn't have, but as far as I know that could just be completely unnecessary.

kakra, 2022-01-15:
> The device ID is different as can be expected, but what's up with the HID version?

This seems to be created from the firmware version of the controller, it doesn't really mean anything for HID. If the controller works in Android emulation mode, the pretended version is usually fixed to 9.03. We are using that fact to work around some SDL2 fixups by faking different models or firmware versions.

The consumer device (or compound device mode for that matter) is something of the Elite 2 controller which can be a keyboard and gamepad in the same device. It thus moved some of its non-gamepad inputs (Guide, Share) to a consumer control sub-device. `hid-xpadneo` is following that and creates synthetic sub-devices for those models that don't have them - and moves the inputs over to those sub-devices.

OTOH, `hid-generic` doesn't do these magic modifications - it just passed the HID reports as-is to user-space, including the sparse button map, differing mappings between firmware versions, and compound device mode. `hid-microsoft` just adds rumble support to the controller (main motors only, Linux has no protocol support for trigger rumble) and uses the rest of the functionality from `hid-generic`. `hid-xpadneo` is a completely self-contained implementation not depending on either of the upstream kernel drivers (and implementing trigger rumble by deriving their values from the pressure and main motors).

TapeWerm, 2022-12-24:
Xbox Series Controller firmware 5.15.3168.0 fixed it for me, but it was annoying. I typically use that controller with my Windows 10 22H2 desktop anyways, but I ran into this bug when I tried split screen Steam Link on my Ubuntu 22.04 NUC with Intel Wi-Fi AX201. I was too lazy to bother updating the controller until I ran into this.

N0th1ngM4tt3rs, 2023-04-08:
I ran into this issue today. I have a TP-Link UB500 and an Intel AX210 BT adapter.  I don't know what firmware is on the controller. Neither adapter worked using gui interfaces. However, DataBeaver's manual pairing process worked for me
https://github.com/bluez/bluez/issues/155#issuecomment-1006563750

frioux, 2023-10-15:
Wanted to come here and confirm that upgrading the firmware of the controller fixed the issues for me as well, even with a cheapo Chinese BT dongle.

belegdol, 2023-10-29:
I believe I am seeing this issue with Fedora 38 and my controller updated to the latest firmware as of today:
[xbox.log](https://github.com/bluez/bluez/files/13197713/xbox.log)

kakra, 2023-10-29:
For Fedora: I was reading somewhere that downgrading the linux-firmware package can fix this for some chipsets. Also, some needed to downgrade bluez to the previous version because of a known bug. I'm not sure if this is fixed in the latest package.

belegdol, 2023-10-29:
Manual pairing does not work either:
```
$ sudo bluetoothctl
Agent registered
[CHG] Device A8:8C:3E:24:4E:C3 TxPower: 0x0014 (20)
[CHG] Device A8:8C:3E:24:4E:C3 ManufacturerData Key: 0x0006 (6)
[CHG] Device A8:8C:3E:24:4E:C3 ManufacturerData Value:
  00                                               .               
[bluetooth]# pair A8:8C:3E:24:4E:C3
Attempting to pair with A8:8C:3E:24:4E:C3
hci0 type 7 discovering off
hci0 A8:8C:3E:24:4E:C3 type LE Public connected eir_len 14
[CHG] Device A8:8C:3E:24:4E:C3 Connected: yes
hci0 A8:8C:3E:24:4E:C3 User Confirm 916653 hint 1
[] Accept pairing with A8:8C:3E:24:4E:C3 (yes/no) >>  [CHG] Device A8:8C:3E:24:4E:C3 ServicesResolved: yes
[] Accept pairing with A8:8C:3E:24:4E:C3 (yes/no) >>  yes
User Confirm Reply successful
hci0 A8:8C:3E:24:4E:C3 auth failed with status 0x05 (Authentication Failed)
Failed to pair: org.bluez.Error.AuthenticationFailed
hci0 A8:8C:3E:24:4E:C3 type LE Public disconnected with reason 2
[CHG] Device A8:8C:3E:24:4E:C3 ServicesResolved: no
[CHG] Device A8:8C:3E:24:4E:C3 Connected: no
[DEL] Device 7E:4B:16:86:02:5E 7E-4B-16-86-02-5E
hci0 type 7 discovering on
[CHG] Device A8:8C:3E:24:4E:C3 ManufacturerData Key: 0x0006 (6)
[CHG] Device A8:8C:3E:24:4E:C3 ManufacturerData Value:
```

belegdol, 2023-10-29:
> For Fedora: I was reading somewhere that downgrading the linux-firmware package can fix this for some chipsets. Also, some needed to downgrade bluez to the previous version because of a known bug. I'm not sure if this is fixed in the latest package.

I have the following versions installed:
```
$ rpm -q linux-firmware bluez
linux-firmware-20230919-1.fc38.noarch
bluez-5.70-1.fc38.x86_64
```
How far does one need to downgrade?

belegdol, 2023-10-29:
This worked:
```
$ rpm -q linux-firmware bluez
linux-firmware-20230310-148.fc38.noarch
bluez-5.66-5.fc38.x86_64
```

belegdol, 2023-10-29:
With AX200 adapter the firmware did not change between the two linux-firmware packages, so I upgraded it back:
```
kernel: Bluetooth: hci0: Found device firmware: intel/ibt-20-1-3.sfi
```
With only bluez downgraded I could still pair my controller. Thanks a lot!

Do you have the link to the "known bug" you referred to earlier? Maybe a fix for it could be backported to the Fedora package.

belegdol, 2023-10-29:
bluez-5.70 with https://github.com/bluez/bluez/commit/3a9c637010f8dc1ba3e8382abe01065761d4f5bb applied on top works as well.

AwkwardGecko, 2024-12-01:
I've got the same issue, on NixOS,

though I've booted into my EndeavourOS partition and I'm having the same issues as you guys regardless of OS.

Plugging the controller in with a cable is fine, and I have no trouble.

Through my testing I'll occasionally get notifications regarding the battery level, stating it's either at 100% or at 50%, no inbetween.

my NixOS config file:
```
{
   hardware = { 
      enableAllFirmware = true;
      bluetooth = {
         enable = true;
         powerOnBoot = true;
         settings.General.Experimental = true;
   };

   services.blueman.enable = true;
   environment.systemPackages = with pkgs; [
      bluez
      bluez-alsa
      bluez-tools
   ];
}
```
I've tried changing many of my config file options, but I can't seem to get anything to work.

Starting with bluetoothctl:
```
[bluetooth]# hci0 new_settings: powered connectable discoverable bondable ssp br/edr le secure-conn wide-band-speech 
[bluetooth]# [CHG] Controller 50:2F:9B:B1:CD:B1 Pairable: yes
[bluetooth]# AdvertisementMonitor path registered
[bluetooth]# power on
[bluetooth]# Changing power on succeeded
[bluetooth]# agent on
Agent is already registered
[bluetooth]# default-agent
[bluetooth]# Default agent request successful
[bluetooth]# scan on
[bluetooth]# SetDiscoveryFilter success
[bluetooth]# Discovery started
[bluetooth]# [CHG] Controller 50:2F:9B:B1:CD:B1 Discovering: yes
[bluetooth]# [NEW] Device AC:8E:BD:31:ED:3D Xbox Wireless Controller
[bluetooth]# pair AC:8E:BD:31:ED:3D 
Attempting to pair with AC:8E:BD:31:ED:3D
[bluetooth]# [CHG] Device AC:8E:BD:31:ED:3D Connected: yes
[Xbox Wireless Controller]# [CHG] Device AC:8E:BD:31:ED:3D Bonded: yes
[Xbox Wireless Controller]# [CHG] Device AC:8E:BD:31:ED:3D UUIDs: 00000001-5f60-4c4f-9c83-a7953298d40d
[Xbox Wireless Controller]# [CHG] Device AC:8E:BD:31:ED:3D UUIDs: 00001800-0000-1000-8000-00805f9b34fb
[Xbox Wireless Controller]# [CHG] Device AC:8E:BD:31:ED:3D UUIDs: 00001801-0000-1000-8000-00805f9b34fb
[Xbox Wireless Controller]# [CHG] Device AC:8E:BD:31:ED:3D UUIDs: 0000180a-0000-1000-8000-00805f9b34fb
[Xbox Wireless Controller]# [CHG] Device AC:8E:BD:31:ED:3D UUIDs: 0000180f-0000-1000-8000-00805f9b34fb
[Xbox Wireless Controller]# [CHG] Device AC:8E:BD:31:ED:3D UUIDs: 00001812-0000-1000-8000-00805f9b34fb
[Xbox Wireless Controller]# [CHG] Device AC:8E:BD:31:ED:3D ServicesResolved: yes
[Xbox Wireless Controller]# [CHG] Device AC:8E:BD:31:ED:3D Paired: yes
[Xbox Wireless Controller]# [NEW] Primary Service (Handle 0x0008)
	/org/bluez/hci0/dev_AC_8E_BD_31_ED_3D/service0008
	00001801-0000-1000-8000-00805f9b34fb
	Generic Attribute Profile
[Xbox Wireless Controller]# [NEW] Primary Service (Handle 0x0009)
	/org/bluez/hci0/dev_AC_8E_BD_31_ED_3D/service0009
	0000180a-0000-1000-8000-00805f9b34fb
	Device Information
[Xbox Wireless Controller]# [NEW] Characteristic (Handle 0x000a)
	/org/bluez/hci0/dev_AC_8E_BD_31_ED_3D/service0009/char000a
	00002a29-0000-1000-8000-00805f9b34fb
	Manufacturer Name String
[Xbox Wireless Controller]# [NEW] Characteristic (Handle 0x000c)
[... 175 lines cut ...]
	name nixos
	short name 
hci0:	Configuration options
	supported options: public-address 
	missing options: 
```
**hciconfig -a hci0:**
```
hci0:	Type: Primary  Bus: USB
	BD Address: 50:2F:9B:B1:CD:B1  ACL MTU: 1021:4  SCO MTU: 96:6
	UP RUNNING PSCAN ISCAN 
	RX bytes:25973 acl:801 sco:0 events:495 errors:0
	TX bytes:10759 acl:154 sco:0 commands:272 errors:0
	Features: 0xbf 0xfe 0x0f 0xfe 0xdb 0xff 0x7b 0x87
	Packet type: DM1 DM3 DM5 DH1 DH3 DH5 HV1 HV2 HV3 
	Link policy: RSWITCH SNIFF 
	Link mode: PERIPHERAL ACCEPT 
	Name: 'nixos'
	Class: 0x7c0104
	Service Classes: Rendering, Capturing, Object Transfer, Audio, Telephony
	Device Class: Computer, Desktop workstation
	HCI Version: 5.2 (0xb)  Revision: 0x2184
	LMP Version: 5.2 (0xb)  Subversion: 0x2184
	Manufacturer: Intel Corp. (2)
```

htqx, 2025-01-23:
I am an arch linux system.
I also have the problem of being unable to use the Xbox S S controller. The Bluetooth display can be linked and can even vibrate, but the buttons are invalid. The second time the link will keep flickering.
Then I updated the firmware of the controller in windows and it worked fine.
The packages I installed are:

1.xone-dongle-firmware
2. xone-dkms-git

There are many other packages online, but something doesn't feel right, this one can be identified as an xbox seriex x controller (although I have an s controller), not a 360 controller.

>uname -a
Linux arch-pc-lv3 6.12.10-arch1-1 #1 SMP PREEMPT_DYNAMIC Sat, 18 Jan 2025 02:26:57 +0000 x86_64 GNU/Linux

kakra, 2025-01-23:
xone is not for Bluetooth connections, it only supports the dongle and probably direct USB connection.

ech0r, 2025-02-09:
@AwkwardGecko Hello fellow Nix user, I was able to get my controller to reliably pair my Xbox Elite Wireless Controller Series 2 with the following

        # ==== SYSTEM PACKAGES ====
          environment.systemPackages = with pkgs; [
            linuxKernel.packages.linux_zen.xpadneo
          ];

          hardware.bluetooth = {
            enable = true; # enables support for Bluetooth
            powerOnBoot = true; # powers up the default Bluetooth controller on boot
            settings = {
              General = {
                Privacy = "device";
                JustWorksRepairing = "always";
                Class = "0x000100";
                FastConnectable = "true";
              };
            };
          };

badnetmask, 2025-03-09:
>             Privacy = "device";
>             JustWorksRepairing = "always";
>             Class = "0x000100";
>             FastConnectable = "true";

In case anyone ever stumbles upon this bug in Fedora, like me, I'll leave this to help:
* [Installed xpadneo from copr](https://copr.fedorainfracloud.org/coprs/atim/xpadneo/)
* Edited /etc/bluetooth/main.conf
* Modified the exact same parameters as indicated above
* Rebooted
* Controller paired and working well
I'm using Fedora 41 as of this day.

closed 2025-08-27 by github-actions[bot]

fluxoz, 2025-12-15:
Hi All, if anyone still needs help on NixOS, I've refactored my config and this works beautifully on 25.11

```
  hardware.xpadneo.enable = true;
```

[xpadneo](https://github.com/atar-axis/xpadneo) is a functional driver for the Xbox Series controllers, 10/10 works great.

Amaderis, 2025-12-17:
> Please help with this issue. The controller is new, and the firmware has been updated to the latest version. Here is what I found in the logs. The reconnects happen randomly, but roughly every one to two minutes.
> 
> ```
> [SIGNAL] LE.Disconnected - org.bluez.Reason.Unknown, Unspecified
> [SIGNAL] Disconnected - org.bluez.Reason.Unknown, Unspecified
> ```
> 
> ```
> > HCI Event: Disconnect Complete (0x05) plen 4                      #106 [hci0] 216.523611
>         Status: Success (0x00)
>         Handle: 16
>         Reason: LMP Response Timeout / LL Response Timeout (0x22)
> @ MGMT Event: Device Disconnected (0x000c) plen 8               {0x0001} [hci0] 216.523620
>         LE Address: AC:8E:BD:57:DB:68 (Microsoft Corporation)
>         Reason: Unspecified (0x00)
> ```

I solved the problem by installing rtl8761b-firmware.