bluez/bluez #157: [5.59] A2DP sink profile is unavailable on newly added/paired devices

knovoselic, 2021-06-17:
Seems like we've discovered a new bug related to ServiceRecords/Endpoints discovery on 5.59. After pairing my headphones for the first time on bluez 5.59, I couldn't get A2DP profile working (it would be marked as unavailable). 

After some experimentation I've found out that downgrading to 5.58 works, but only if I remove `/var/lib/bluetooth` folder before pairing the headphones again.
After a little bit of digging, I've narrowed the issue down to the cache file for the headphones. The cache file created by 5.59 seems to have some of the services/endpoints not listed properly. Here's the cache file created by 5.58, where A2DP works fine:

```
───────┬───────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────
       │ File: bluetooth_working/00:1A:7D:DA:71:13/cache/00:16:94:20:DF:B3
───────┼───────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────
   1   │ [General]
   2   │ Name=PXC 550
   3   │ 
   4   │ [ServiceRecords]
   5   │ 0x00010001=35300900000A0001000109000135031918010900043513350619010009001F35091900070900060900060900053503191002
   6   │ 0x00010002=35300900000A00010002090001350319180F0900043513350619010009001F350919000709000709000A0900053503191002
   7   │ 0x00010003=353E0900000A0001000309000135111C6333135823C111E5B696FEFF819CDC9F0900043513350619010009001F350919000709000B09FFFF0900053503191002
   8   │ 0x00010004=35590900000A00010004090001350619111E191203090004350C35031901003505190003080A090006350909656E09006A0901000900093508350619111E090106090100250F48616E64732D4672656
       │ 520756E697409031109003B
   9   │ 0x00010005=354A0900000A0001000509000135111C1DDCE62AECB1445581530743C87AEC9F090004350C350319010035051900030801090006350909656E09006A09010009010025084E617574696C7573
  10   │ 0x00010007=35380900000A00010007090001350319110B0900043510350619010009001935061900190901030900093508350619110D090103090311090001
  11   │ 0x00010008=353B0900000A00010008090001350619110E19110F0900043510350619010009001735061900170901040900093508350619110E090106090311090001
  12   │ 0x00010009=35380900000A00010009090001350319110C0900043510350619010009001735061900170901040900093508350619110E090106090311090002
  13   │ 0x000100A4=35530900000A000100A40900013509191108191131191203090004350C35031901003505190003080D090006350909656E09006A0901000900093508350619110809010209010025074865616473657
       │ 40903022801
  14   │ 0x00010000=35410900000A000100000900013503191200090006350909656E09006A0901000902000901030902010900820902020900460902030901060902042801090205090001
  15   │ 0x000100A6=35530900000A000100A60900013509191108191131191203090004350C35031901003505190003080D090006350909656E09006A0901000900093508350619110809010209010025074865616473657
       │ 40903022801
  16   │ 0x000100A7=35530900000A000100A70900013509191108191131191203090004350C35031901003505190003080B090006350909656E09006A0901000900093508350619110809010209010025074865616473657
       │ 40903022801
  17   │ 
  18   │ [Endpoints]
  19   │ 03=01:ff:01:4f0000000100ff
  20   │ 01=01:00:01:ffff0235
  21   │ LastUsed=02:01
───────┴───────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────

```

and here's the same file created on 5.59, where A2DP is reported as unavailable:
```
───────┬───────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────
       │ File: bluetooth_broken/00:1A:7D:DA:71:13/cache/00:16:94:20:DF:B3
───────┼───────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────
   1   │ [General]
   2   │ Name=PXC 550
   3   │ 
   4   │ [ServiceRecords]
   5   │ 0x00010001=35300900000A0001000109000135031918010900043513350619010009001F35091900070900060900060900053503191002
   6   │ 0x00010002=35300900000A00010002090001350319180F0900043513350619010009001F350919000709000709000A0900053503191002
   7   │ 0x00010003=353E0900000A0001000309000135111C6333135823C111E5B696FEFF819CDC9F0900043513350619010009001F350919000709000B09FFFF0900053503191002
   8   │ 0x00010005=354A0900000A0001000509000135111C1DDCE62AECB1445581530743C87AEC9F090004350C350319010035051900030801090006350909656E09006A09010009010025084E617574696C7573
   9   │ 0x00010007=35380900000A00010007090001350319110B0900043510350619010009001935061900190901030900093508350619110D090103090311090001
  10   │ 0x00010008=353B0900000A00010008090001350619110E19110F0900043510350619010009001735061900170901040900093508350619110E090106090311090001
  11   │ 0x00010009=35380900000A00010009090001350319110C0900043510350619010009001735061900170901040900093508350619110E090106090311090002
  12   │ 0x000100A9=35590900000A000100A9090001350619111E191203090004350C35031901003505190003080C090006350909656E09006A0901000900093508350619111E090106090100250F48616E64732D4672656
       │ 520756E697409031109003B
  13   │ 0x000100AE=35530900000A000100AE0900013509191108191131191203090004350C35031901003505190003080B090006350909656E09006A0901000900093508350619110809010209010025074865616473657
       │ 40903022801
  14   │ 0x00010000=35410900000A000100000900013503191200090006350909656E09006A0901000902000901030902010900820902020900460902030901060902042801090205090001
  15   │ 
  16   │ [Endpoints]
  17   │ 03=01:00:00:ffff0235
  18   │ 01=01:00:00:ffff0235
───────┴───────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────
```

There are differences in both `ServiceRecords` and `Endpoints` sections. 

Please not that this bug is visible only on 5.59 if you're pairing the device for the first time, or if you manually remove `/var/lib/bluetooth` folder. If the folder contains the cache file for your device created by one of the older `bluez` (5.58 or older), A2DP profile will work fine, even if you remove and re-pair the device. It only doesn't work if the cache file doesn't exist yet (if 5.59 version creates the file).

You can read more info about how we've discovered the bug here: https://bbs.archlinux.org/viewtopic.php?pid=1978509

Vudentz, 2021-06-18:
Try with the following patch:

https://git.kernel.org/pub/scm/bluetooth/bluez.git/commit/?id=0388794dc5fdb73a4ea88bcf148de0a12b4364d4

abhijeetviswa, 2021-06-18:
I have the same issue as OP. 
@Vudentz I installed `bluez-git` on my Arch Linux machine and this patch fixed the issue.

Vudentz, 2021-06-18:
@abhijeetviswa thanks for confirming it, we might have to release 5.60 sooner.

knovoselic, 2021-06-19:
@Vudentz thanks for the quick fix, the patch works. Tested with it and without it (just a commit before) to be sure,

marcelofern, 2021-06-19:
I have the same issue as OP. Just posting it here to register that this might not be something rare.

joakim-tjernlund, 2021-06-20:
You may have to update to bluez git, see https://github.com/bluez/bluez/issues/157

mtekman, 2021-06-20:
Exact same problem, and fix from bluez-git worked

tim-hilt, 2021-06-21:
I hoped this would solve the issues I'm having with my AirPods Pro, however it got even worse. Here's how my connection-procedure looked like with bluez from arch/extra repo:

1. Set `ControllerMode = bredr` in /etc/bluetooth/main.conf
2. `systemctl restart bluetooth`
3. Try to connect with the AirPods (which would always fail, but after that, the device shows up in the settings
4. Comment out `ControllerMode = bredr`
5. `systemctl restart bluetooth`
6. Connect successfully to the AirPods (even works automatically whenever I open the case).

After that the connection would be established automatically, however there's only HSP/HFP would be available. As soon as I switch the profile to A2DP (within KDEs bluetooth-settings), the output would get deactivated.

Now, that I have installed bluez-git, I can still connect to the AirPods Pro with the same procedure as above, however, the headphones aren't registered as audio-sink anymore.

Is this to be expected with Apple hardware or am I hitting another bug here?

joakim-tjernlund, 2021-06-21:
@tim-hilt , you may need to rm /var/lib/bluetooth folder and repair the your AirPods, according to above.

tim-hilt, 2021-06-21:
@joakim-tjernlund thanks for the , but unfortunately this doesn't change the behavior for me.

joakim-tjernlund, 2021-06-21:
Just another bug I would guess, perhaps not bluez but pipewire/pulsaudio or kernel

tim-hilt, 2021-06-21:
Thanks for the hint! Will investigate this further.

artemkovalyov, 2021-06-21:
I'm also affected by this one. Downgrading the bluez to 5.58 now.

artemkovalyov, 2021-06-21:
Downgrading bluez to 5.58 and removing `/var/lib/bluetooth` worked. Thanks for the workaround but would be really nice to fix this bug.

joakim-tjernlund, 2021-06-21:
@tim-hilt , perhaps check out https://gitlab.freedesktop.org/pipewire/pipewire/-/issues/1328

kmoad, 2021-06-22:
Encountered this on Fedora 34 with bluez 5.59 and Jabra 75T earbuds. Steps below to fix it.

```
dnf downgrade bluez
```
This will install bluez 5.56

Either go into bluetooth settings and remove the device, or `sudo rm -rf /var/lib/bluetooth`. Doing it in device settings is preferred, since the cmdline option will remove all your bluetooth connections.

Reboot

Re-pair with the device

Lock your version of bluez while we wait for a fix. This can be done with

```
dnf install dnf-plugin-versionlock
dnf versionlock bluez
```

Mythbusters123, 2021-06-23:
> Exact same problem, and fix from bluez-git worked
> 
> Edit: Temporarily worked... I confusingly cannot connect again, despite pairing/trusting fine.
> Edit2: Same result (can pair, can't connect) with 5.58 downgrade:
> 
> ```
> sudo pacman -R bluez-git  
> sudo pacman -U https://archive.archlinux.org/packages/b/bluez-utils/bluez-utils-5.58-1-x86_64.pkg.tar.zst  
> sudo pacman -U https://archive.archlinux.org/packages/b/bluez-libs/bluez-libs-5.58-1-x86_64.pkg.tar.zst  
> sudo pacman -U https://archive.archlinux.org/packages/b/bluez/bluez-5.58-1-x86_64.pkg.tar.zst
> ```
> 
> Edit3: Fixed, with
> 
> 1. Installed pipewire and pipewire-pulse (this replaces pulseaudio and **pulseaudio-bluetooth**),
> 2. `systemctl --user start pipewire-pulse`
> 3. `sudo systemctl stop bluetooth`
> 4. `rm -rf /var/lib/bluetooth` (for the nth time)
> 5. `sudo systemctl start bluetooth`
> 6. `bluetoothctl` (power on, scan on, pair, connect)
> 
> In my case I believe the problem was simply pulseaudio-bluetooth

I am also using arch linux, but I did not have to change from pulseaudio to pipewire. you only needed to 
1. ```sudo pacman -R bluez-git  
sudo pacman -U https://archive.archlinux.org/packages/b/bluez-utils/bluez-utils-5.58-1-x86_64.pkg.tar.zst  
sudo pacman -U https://archive.archlinux.org/packages/b/bluez-libs/bluez-libs-5.58-1-x86_64.pkg.tar.zst  
sudo pacman -U https://archive.archlinux.org/packages/b/bluez/bluez-5.58-1-x86_64.pkg.tar.zst```
2. `sudo systemctl stop bluetooth`
3. `rm -rf /var/lib/bluetooth`
4. `sudo systemctl start bluetooth`
5. `bluetoothctl`
These worked for 1st gen AirPods, too.

marcelofern, 2021-06-23:
Sorry to be a pain but is there an estimated date for release `5.6`?

infinytum, 2021-06-25:
I hit the same issue with AirPods, downgrading to 5.58 and removing the bluetooth folder fixed it. I hope this will be fixed in the next update?

tim-hilt, 2021-06-25:
To give an update from my side: The issue I saw before was fixed by swapping out `pulseaudio` in favor of `pipewire-pulse`. @infinytum maybe this will also help in your case.

I also had problems with the firmware of my bluetooth-card (Intel AX200). I could listen to music, but experienced a very small sound-droput every approx. 6 seconds. I actually timed that and the 6 second delta was surprisingly uniform. Apparently the issues got fixed in `linux-firmware 20210518`.

In my case, I just installed `linux-firmware-git` via the AUR. Powered the laptop off (Important! Don't just reboot - power off, since the bluetooth card resets itself by being without voltage for a bit of time).

Now I don't have any disconnects, deactivations or dropout, but I'm experiencing low volume levels on the AirPods and none of the [suggested fixes from the Arch Wiki](https://wiki.archlinux.org/title/Bluetooth_headset#Apple_AirPods_have_low_volume) (running bluetoothd with `--nolplugin=avrcp`, disabling spatial audio, turning AirPods to Mono-Audio) worked in my case. So if anyone has another idea about that: Thanks in advance!

@joakim-tjernlund actually pointed me in the right directions to solve the audio-artifacts [over at pipewires GitLab](https://gitlab.freedesktop.org/pipewire/pipewire/-/issues/1328). I had them on all of my bluetooth-headphones, although with different characteristics. Thanks again Joakim!

infinytum, 2021-06-25:
@tim-hilt I used a smartphone to increase my AirPods audio level, then reconnect them to the PC and they somehow were louder. Maybe that works? It seems that the AirPods have their own "volume" that is synced on iOS but not Linux?

If I try with AirPods Max, I can adjust the volume on the headphones + the system volume separately. I had to turn them to maximum volume to be able to use the Linux controls in normal levels.

tim-hilt, 2021-06-26:
@infinytum oh wow, that's some interesting insight! However, I couldn't reproduce this behavior unfortunately. I set the volume to different extremes on the iPhone, but didn't notice any change in volume level on the laptop. This holds true for my ThinkPad T14S, as well as my ThinkPad X1 Carbon 5th gen.

Other than that, the volume-dropouts started reappearing on the T14S, whereas I didn't notice any audio-artifacts on the X1 Carbon. Strange...

SebiderSushi, 2021-06-28:
> Now I don't have any disconnects, deactivations or dropout, but I'm experiencing low volume levels on the AirPods and none of the [suggested fixes from the Arch [...]

This is how i deal with the AirPods volume: https://github.com/bluez/bluez/issues/17#issuecomment-683365208
basically the issue is that PulseAudio did not support absolute volume for bluetooth devices for a very long time ([the patch for it](https://gitlab.freedesktop.org/pulseaudio/pulseaudio/-/merge_requests/239) was merged about one month ago).

I use a script that iterates through a few of the possible d-bus paths and tries setting the remote volume to the maximum value of '127' for the MAC address of my AirPods. I mapped the script to a keyboard shortcut and that's good enough as a workaround for me.

[This is the script i use](https://gist.github.com/SebiderSushi/f471e864617e49c743051ca996ffc47c)

joakim-tjernlund, 2021-06-29:
@tim-hilt , Intel AX2xx just got new FW at https://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git/log/

tim-hilt, 2021-06-29:
@joakim-tjernlund man, I really should watch those logs more regularly! Updated, rebooted, works again.

tim-hilt, 2021-06-29:
@SebiderSushi thanks for the tip! Again, that are some "exotic" recommendations that I've hadn't heard of before. I just blindly copied and executed your script and it then printed

```
Error org.freedesktop.DBus.Error.UnknownObject: Method "Set" with signature "ssv" on interface "org.freedesktop.DBus.Properties" doesn't exist

Error org.freedesktop.DBus.Error.UnknownProperty: No such property 'Volume'
Error org.freedesktop.DBus.Error.UnknownObject: Method "Set" with signature "ssv" on interface "org.freedesktop.DBus.Properties" doesn't exist

[ ... repeat last line 15 times ]
```

it might stem from the fact, that I don't even use `pulseaudio` directly, but `pipewire-pulse`! When taking a look at `d-feet`, I don't even see `org.bluez.MediaTransport1`, never mind a `Volume` property.

Let's not spam this thread any further, as it gravitates towards my AirPods-issues rather than being concerned with the original bluez-issue. Is there a different way to reach you? You can also find some contact-information about myself in my GitHub-profile, if you want to reach out to me. Thanks again for the tip so far!

CathalOConnorRH, 2021-07-02:
also seeing this with Sony XM4 on Fedora 34, These hadn't been paired before. 
Followed the steps in [1] and working perfectly. 

[1] https://github.com/bluez/bluez/issues/157#issuecomment-866271080

DanielKotik, 2021-07-03:
Can confirm this with my new Poly Voyager 8200 UC on Fedora 34. Workaround https://github.com/bluez/bluez/issues/157#issuecomment-866271080 has helped me.

jonkoops, 2021-07-04:
I am experiencing this issue as well on my Sony WH-1000XM3, running Fedora Silverblue (34). @Vudentz would it be possible to get a hot fix release for this issue, it seems very widespread.

garrett, 2021-07-05:
Firstly: This is the first time I can remember having Bluetooth broken like this in Linux. Bluetooth is generally excellent on Linux, thanks to this project _Huge thanks_ to everyone at working on Bluez. :clap: :+1: 

However, Bluetooth audio has been broken for over two weeks on all Linux distributions that have upgraded to Bluez 5.59, including (but probably not limited to) Fedora and Arch.

Having broken Bluetooth audio is bad in most times, but it's compounded by the need to use headphones for many people to be able to do video calling for remote work, remote schooling, calling friends and family, and so on, in the age of the pandemic.

It would be excellent to have a new release with the fix, either as a full release or as a hotfix.

---

@jonkoops: I'm also using my Sony WH-1000XM3 on Fedora Silverblue 34. At first, I did a workaround by adding a wacky USB audio device that exports as Bluetooth (needed for a Nintendo Switch) and paired to that until I found this issue and did an override for Silverblue. (I couldn't do a rollback, as I had to do a fresh reinstall two weeks ago.)

I had to [hunt down builds of the previous version of Bluez](https://koji.fedoraproject.org/koji/buildinfo?buildID=1748733), download RPMs (x86_64), and do an override for the packages that are installed by default: `rpm-ostree override replace bluez bluez-libs bluez-obexd bluez-cups`

FWIW: There's a Fedora bug at https://bugzilla.redhat.com/show_bug.cgi?id=1976795.

abhijeetviswa, 2021-07-05:
@garrett I agree. The regression is serious enough that  warrants an immediate hot fix release. Hopefully we see one soon 🤞

DenisValcke, 2021-07-06:
The downgrade also solved the issue for me.

ErikReider, 2021-07-06:
For me using bluez-git from the AUR and removing `/var/lib/bluetooth` worked for my Airpods Pro

Toadfield, 2021-07-06:
For me this bug happens even though I am using pipewire and not pulseaudio.

garrett, 2021-07-06:
@Toadfield: Yep, everyone on Fedora 34 is on PipeWire by default. There's a big enough mix of people above in this issue using PipeWire and PulseAudio that I think we can safely say this issue is really about Bluez, irrespective of PipeWire or PulseAudio.

Hopefully there will be an official fix soon, as most people definitely won't find this bug and know to downgrade Bluez... they'll just have broken Bluetooth audio like we all have had for the past few weeks.

Toadfield, 2021-07-06:
When will the fix be released?
Cause I don't want to downgrade or install bluez-git.

SocketByte, 2021-07-07:
My god, it's such a blessing that I found this issue. I've been pulling my hair for the past few days because my bluetooth would just not cooperate at all. I honestly didn't think it could've been an issue with bluez, as I'm always taking the blame on myself when something doesn't work haha. 

Using `bluez-git` and removing `/var/lib/bluetooth` did it for me, finally. Hopefully we'll see a patched version soon!

Vudentz, 2021-07-07:
5.60 has been released:

https://github.com/bluez/bluez/commit/e1432e7b8d09f557115d2537dc36f2fce76be209

closed 2021-07-07 by Vudentz

jonkoops, 2021-07-12:
5.60 has indeed resolved the problem and has been rolled out to the latest Silverblue. I did however still need to remove the `/var/lib/bluetooth` directory and restart the service before things were working properly.

TheJuli, 2021-07-16:
> 5.60 has indeed resolved the problem and has been rolled out to the latest Silverblue. I did however still need to remove the `/var/lib/bluetooth` directory and restart the service before things were working properly.

This is critical. I was having these problems and only when I removed the `/var/lib/bluetooth` folder, I got it all working again.

Luc45, 2021-12-11:
Same issue here, started happening a couple months ago, probably after an update

MarcelRobitaille, 2022-02-01:
I was not seeing the A2DP profile even after following all these steps. My issue was that there was a `--noplugin a2dp` in the `ExecStart` of `/etc/systemd/system/bluetooth.service`. Not sure how that got there

Sorry if this isn't the place for it, but maybe this will help somebody else out.

pourmand1376, 2022-02-03:
I also had this problem and it showed `Failed to connect: org.bluez.Error.Failed` and loading `module-bluetooth-discover` gave an error.
```bash
pactl load-module module-bluetooth-discover
Failure: Module initialization failed
```

All I have to do as mentioned was to:
```bash
    sudo systemctl stop bluetooth
    rm -rf /var/lib/bluetooth
    sudo systemctl start bluetooth
    bluetoothctl
```

alvarlagerlof, 2022-05-31:
I have tried a bunch of things here but I keep on getting this. I am on the latest Fedora Silverblue.

```
maj 31 23:55:21 rehoboam systemd[1]: Started bluetooth.service - Bluetooth service.
maj 31 23:55:21 rehoboam bluetoothd[931]: Bluetooth management interface 1.21 initialized
maj 31 23:56:59 rehoboam bluetoothd[931]: src/service.c:btd_service_connect() a2dp-sink profile connect failed for 94:DB:56:EA:BA:2A: Protocol not available
maj 31 23:57:00 rehoboam bluetoothd[931]: src/service.c:btd_service_connect() a2dp-sink profile connect failed for 94:DB:56:EA:BA:2A: Protocol not available
maj 31 23:57:00 rehoboam bluetoothd[931]: src/service.c:btd_service_connect() a2dp-sink profile connect failed for 94:DB:56:EA:BA:2A: Protocol not available
maj 31 23:57:01 rehoboam bluetoothd[931]: src/service.c:btd_service_connect() a2dp-sink profile connect failed for 94:DB:56:EA:BA:2A: Protocol not available
maj 31 23:57:01 rehoboam bluetoothd[931]: src/service.c:btd_service_connect() a2dp-sink profile connect failed for 94:DB:56:EA:BA:2A: Protocol not available
maj 31 23:57:02 rehoboam bluetoothd[931]: src/service.c:btd_service_connect() a2dp-sink profile connect failed for 94:DB:56:EA:BA:2A: Protocol not available
maj 31 23:57:02 rehoboam bluetoothd[931]: src/service.c:btd_service_connect() a2dp-sink profile connect failed for 94:DB:56:EA:BA:2A: Protocol not available
maj 31 23:57:13 rehoboam bluetoothd[931]: src/service.c:btd_service_connect() a2dp-sink profile connect failed for 94:DB:56:EA:BA:2A: Protocol not available
```

barraIhsan, 2025-01-22:
Sorry for bumping this issue, but is there any fix planned? Using `bluez 5.79` and I need to do https://github.com/bluez/bluez/issues/157#issuecomment-866271080, (remove the `/var/lib/bluetooth` and restart bluetooth) everytime I connect to a bluetooth audio.