bluez/bluez #159: Unable to select SEP

nd-duy, 2021-06-21:
I can connect bluetooth with HSP/HFP profile but cannot with A2DP profile
When I check the status of bluetooth.service, it shows this:
`Jun 21 10:34:45 duy bluetoothd[22248]: src/profile.c:record_cb() Unable to get Hands-Free Voice gateway SDP record: Host is down
Jun 21 10:35:10 duy bluetoothd[22248]: profiles/audio/a2dp.c:a2dp_select_capabilities() Unable to select SEP
Jun 21 10:35:12 duy bluetoothd[22248]: profiles/audio/a2dp.c:a2dp_select_capabilities() Unable to select SEP
Jun 21 10:36:00 duy bluetoothd[22248]: src/profile.c:ext_io_disconnected() Unable to get io data for Hands-Free Voice gateway: getpeername: Transport endpoint is not connected (107)
Jun 21 10:39:49 duy bluetoothd[22248]: src/profile.c:ext_connect() Hands-Free Voice gateway failed connect to 28:52:E0:FF:64:16: Connection refused (111)
Jun 21 10:39:49 duy bluetoothd[22248]: profiles/audio/a2dp.c:a2dp_select_capabilities() Unable to select SEP
Jun 21 10:39:49 duy bluetoothd[22248]: profiles/audio/avdtp.c:caps_to_list() Invalid capability data in getcap resp
Jun 21 10:39:50 duy bluetoothd[22248]: /org/bluez/hci0/dev_28_52_E0_FF_64_16/fd0: fd(43) ready
Jun 21 10:41:11 duy bluetoothd[22248]: profiles/audio/a2dp.c:a2dp_select_capabilities() Unable to select SEP
Jun 21 10:41:14 duy bluetoothd[22248]: profiles/audio/a2dp.c:a2dp_select_capabilities() Unable to select SEP`
Some how I manage to connect with A2DP but then it disconnects immediately
My bluetooth device profile:
`Profiles:
	off: Off (sinks: 0, sources: 0, priority: 0, available: yes)
	headset-head-unit: Headset Head Unit (HSP/HFP) (sinks: 1, sources: 1, priority: 0, available: yes)
        headset-head-unit-cvsd: Headset Head Unit (HSP/HFP, codec CVSD) (sinks: 1, sources: 1, priority: 0, available: yes)
	Active Profile: headset-head-unit-cvsd`

nd-duy, 2021-06-21:
I found the solution
I follow #157  
I remove `/var/lib/bluetooth` folder and then reconnect. Some how all the A2DP profile is there

closed 2021-06-21 by nd-duy

ocean2706, 2022-10-17:
Actually, the issue can be related to systemd bluetooth.service configuration. 
Play by changing from  "true" to false each "security" settings.