bluez/bluez #160: [5.56 - 5-59] Bluetooth force reconnects disconnected device

validatedev, 2021-06-23:
I have an AirPods Pro. While the device is connected to a PC, if I try to switch the device to an iPhone, after 10-20 sec it reconnects to the PC again. Does not matter whether I use Pulseaudio or Pipewire. It has been happening on Ubuntu 21.04 (5.56) and Fedora 34 (5.59), but not on Ubuntu 20.04 LTS (5.55). On Ubuntu 20.04 LTS, the behavior is what it should be, in my opinion. I can switch between iPhone and PC without any reconnecting between any device. Thanks in advance.

bluetoothctl output, you can see ` Connected: no` and ` Connected: yes`:
```
[validate@validatepc ~]$ bluetoothctl
Agent registered
[NEW] Device 64:A0:AC:49:3F:61 64-A0-AC-49-3F-61
[CHG] Device 64:A0:AC:49:3F:61 RSSI: -60
[CHG] Device 64:A0:AC:49:3F:61 ManufacturerData Key: 0x004c
[CHG] Device 64:A0:AC:49:3F:61 ManufacturerData Value:
  07 19 01 0e 20 2b 99 8f 01 00 04 5a 32 e5 e5 72  .... +.....Z2..r
  e3 0e 12 e5 37 31 66 ea 2b ce dc                 ....71f.+..     
[NEW] Device 51:EE:40:1B:50:0F 51-EE-40-1B-50-0F
[CHG] Device 51:EE:40:1B:50:0F RSSI: -42
[CHG] Device 51:EE:40:1B:50:0F ManufacturerData Key: 0x004c
[CHG] Device 51:EE:40:1B:50:0F ManufacturerData Value:
  10 06 58 1d 4d ba 9f 80                          ..X.M...        
[CHG] Device 14:C8:8B:D6:E2:8A ServicesResolved: no
[CHG] Device 14:C8:8B:D6:E2:8A Connected: no
[CHG] Device 51:EE:40:1B:50:0F RSSI: -43
[CHG] Device 64:A0:AC:49:3F:61 RSSI: -65
[CHG] Device 64:A0:AC:49:3F:61 ManufacturerData Key: 0x004c
[CHG] Device 64:A0:AC:49:3F:61 ManufacturerData Value:
  07 19 01 0e 20 2b 99 8f 01 00 05 05 fa 22 50 3f  .... +......."P?
  da 6a af 33 cf 8b 21 46 ec 8b d0                 .j.3..!F...     
[CHG] Device 14:C8:8B:D6:E2:8A Connected: yes
[CHG] Device 64:A0:AC:49:3F:61 RSSI: -65
[CHG] Device 51:EE:40:1B:50:0F RSSI: -40
[CHG] Device 64:A0:AC:49:3F:61 RSSI: -67
[CHG] Device 64:A0:AC:49:3F:61 ManufacturerData Key: 0x004c
[CHG] Device 64:A0:AC:49:3F:61 ManufacturerData Value:
  07 19 01 0e 20 2b 99 8f 01 00 04 eb 84 60 1e c5  .... +.......`..
  9b 63 2e 24 8f fd 1a 95 b5 a7 39                 .c.$......9     
[CHG] Device 51:EE:40:1B:50:0F RSSI: -40
[CHG] Device 64:A0:AC:49:3F:61 RSSI: -78
[NEW] Device 55:B4:33:AF:5E:C4 55-B4-33-AF-5E-C4
[CHG] Device 51:EE:40:1B:50:0F RSSI: -41
[CHG] Device 64:A0:AC:49:3F:61 RSSI: -52
[CHG] Device 64:A0:AC:49:3F:61 ManufacturerData Key: 0x004c
[CHG] Device 64:A0:AC:49:3F:61 ManufacturerData Value:
  07 19 01 0e 20 2b 99 8f 01 00 04 96 7d 2f 8b 8e  .... +......}/..
  d5 40 a3 94 22 de 7b 11 90 8b 78                 .@..".{...x     
[CHG] Device 51:EE:40:1B:50:0F RSSI: -42
[CHG] Device 64:A0:AC:49:3F:61 RSSI: -53
[CHG] Device 51:EE:40:1B:50:0F RSSI: -42
[CHG] Device 64:A0:AC:49:3F:61 RSSI: -59
[CHG] Device 51:EE:40:1B:50:0F RSSI: -41
[Mert Can’s AirPods Pro]# 
[Mert Can’s AirPods Pro]# 
[CHG] Device 64:A0:AC:49:3F:61 RSSI: -63
[CHG] Device 51:EE:40:1B:50:0F RSSI: -40
[Mert Can’s AirPods Pro]# exit
```

Vudentz, 2021-06-23:
Probably duplicated of: https://github.com/bluez/bluez/issues/157

validatedev, 2021-06-23:
> Probably duplicated of: #157

I don't think it refers to the exact issue, but I'll downgrade to 5.56 and try the same thing.

validatedev, 2021-06-23:
With 5.56 on Fedora 34, the same issue has been happening.

validatedev, 2021-06-23:
Unfortunately, it is still happening with the latest git.

apandit, 2021-06-28:
The original Airpods had a bug where they would emit a `Connection Timeout` when disconnecting (even when doing so legitimately to connect to a different host). This would result in the automatic reconnect logic in bluez kicking in.

You can test this by modifying your main.conf to remove the a2dp-sink uuid from ReconnectUUIDs: https://git.kernel.org/pub/scm/bluetooth/bluez.git/tree/src/main.conf#n227

See https://patchwork.kernel.org/project/bluetooth/patch/20201207154903.bluez.1.I3e043a481273442748bcff0728b2f0e208017cd2@changeid/ for context

validatedev, 2021-06-29:
Oh thanks for the info! I'll try the workaround as soon as possible.

closed 2021-07-09 by Vudentz

validatedev, 2021-07-31:
Worked flawlessly. Thanks for the info @apandit