bluez/bluez #161: [5.59] No device after putting AirPods Pro on case then reputting on the ears again

validatedev, 2021-06-23:
I have an Airpods Pro. While using Pulseaudio, when I put the earphones in their box and plug them back in after a while, although there is Bluetooth connectivity, no device is visible, I have to disconnect and reconnect the device. When using Pipewire, the device shows up, but I can't get any sound. It doesn't matter even if I try to change the sink manually, again I have to disconnect and reconnect. I'm using Fedora 34 with pulseaudio 14.2 and bluez 5.59. With bluez 5.55 and pulseaudio 13.99 on Ubuntu 20.04 LTS, the experience is flawless. I can't speak for sure, but I remember it was not a problem on Ubuntu 21.04 either (with bluez 5.56 and pulseaudio 14.2). Thanks in advance.

Vudentz, 2021-06-23:
Probably duplicated of: https://github.com/bluez/bluez/issues/157

validatedev, 2021-06-23:
> Probably duplicated of: #157

I don't think it refers to the exact issue, but I'll downgrade to 5.56 and try the same thing.

validatedev, 2021-06-23:
Yeah also the same issue with 5.56, it looks connected on `gnome-control-center` Bluetooth section, I also heard a connection sound on Airpods, but no device on Sound section. `pactl list sinks` only represents the default speakers.

Vudentz, 2021-06-23:
> Yeah also the same issue with 5.56, it looks connected on `gnome-control-center` Bluetooth section, I also heard a connection sound on Airpods, but no device on Sound section. `pactl list sinks` only represents the default speakers.

Try updating to the latest git.

validatedev, 2021-06-23:
> > Yeah also the same issue with 5.56, it looks connected on `gnome-control-center` Bluetooth section, I also heard a connection sound on Airpods, but no device on Sound section. `pactl list sinks` only represents the default speakers.
> 
> Try updating to the latest git.

I’ll try ASAP and get back to you. Thanks!

validatedev, 2021-06-23:
Unfortunately, it is still happening with the latest git.

Vudentz, 2021-06-23:
@validatedev you might need to repair since the bug was with the cache, that said it would be good to have the btmon and btluetoothd traces in order to check if there is something else going on.

validatedev, 2021-06-23:
Since I do not know exactly the structure of the package, I want to ask, where are the caches stored and which files-directories should I delete? I would be very grateful if you could guide me on this.

Vudentz, 2021-06-23:
See https://github.com/bluez/bluez/issues/157, that is discussed at length there.

validatedev, 2021-06-23:
> See #157, that is discussed at length there.

Thanks, will try asap.

validatedev, 2021-06-23:
Hi. Got the logs with a clean installation of the latest git. Unfortunately, it is still happening.
What I tried:
First, tried https://github.com/bluez/bluez/issues/161
Second, tried https://github.com/bluez/bluez/issues/160
After that, tried https://github.com/bluez/bluez/issues/161 many times. Sometimes bluetoothd crashed, maybe you can see it on logs.
The other logs are from the last boot: `-- Boot cb5a70558c3148148cfd282e6cfa7a99 --`, you can see the line on `bluetoothdlog.txt`.

 
[bluetoothctllog.txt](https://github.com/bluez/bluez/files/6705319/bluetoothctllog.txt)
[bluetoothdlog.txt](https://github.com/bluez/bluez/files/6705320/bluetoothdlog.txt)
[hcitrace.txt](https://github.com/bluez/bluez/files/6705321/hcitrace.txt)

validatedev, 2021-07-31:
Still happening on 5.60.

Vudentz, 2021-08-02:
@validatedev Have you tried clearing the endpoint cache?

validatedev, 2021-08-03:
> @validatedev Have you tried clearing the endpoint cache?

How can I clear? In addition, that was on a clean system (reinstalled), paired with 5.60 only.

Vudentz, 2021-09-10:
```
> ACL Data RX: Handle 256 flags 0x02 dlen 22            #2081 [hci0] 348.178527
      Channel: 65 len 18 [PSM 25 mode Basic (0x00)] {chan 1}
      AVDTP: Set Configuration (0x03) Command (0x00) type 0x00 label 7 nosp 0
        ACP SEID: 4
        INT SEID: 2
        Service Category: Media Transport (0x01)
        Service Category: Media Codec (0x07)
          Media Type: Audio (0x00)
          Media Codec: MPEG-2,4 AAC (0x02)
            Object Type: MPEG-2 AAC LC (0xc0)
            Frequency: 44100 (0x100)
            Channels: 2 (0x04)
            Bitrate: 320000bps
            VBR: Yes
        Service Category: Delay Reporting (0x08)
= bluetoothd: profiles/audio/media.c:endpoint_reply() Endpoint r..   348.179837
< ACL Data TX: Handle 256 flags 0x00 dlen 8             #2082 [hci0] 348.179901
      Channel: 514 len 4 [PSM 25 mode Basic (0x00)] {chan 1}
      AVDTP: Set Configuration (0x03) Response Reject (0x03) type 0x00 label 7 nosp 0
        Service Category: Reserved (0x29)
        Error code: UNSUPPORTED_CONFIGURATION (0x29)
```

```
> ACL Data RX: Handle 256 flags 0x02 dlen 22            #2158 [hci0] 378.971380
      Channel: 64 len 18 [PSM 25 mode Basic (0x00)] {chan 0}
      AVDTP: Set Configuration (0x03) Command (0x00) type 0x00 label 14 nosp 0
        ACP SEID: 4
        INT SEID: 2
        Service Category: Media Transport (0x01)
        Service Category: Media Codec (0x07)
          Media Type: Audio (0x00)
          Media Codec: MPEG-2,4 AAC (0x02)
            Object Type: MPEG-2 AAC LC (0xc0)
            Frequency: 44100 (0x100)
            Channels: 2 (0x04)
            Bitrate: 320000bps
            VBR: Yes
        Service Category: Delay Reporting (0x08)
= bluetoothd: profiles/audio/media.c:endpoint_reply() Endpoint r..   378.974357
< ACL Data TX: Handle 256 flags 0x00 dlen 8             #2159 [hci0] 378.974263
      Channel: 1025 len 4 [PSM 25 mode Basic (0x00)] {chan 0}
      AVDTP: Set Configuration (0x03) Response Reject (0x03) type 0x00 label 14 nosp 0
        Service Category: Reserved (0x29)
        Error code: UNSUPPORTED_CONFIGURATION (0x29)
```

It really beats me why would anyone change the SEID on every session but it lloks like that is the case here, anyway with the following patch perhaps that should be a retry after failing:

https://git.kernel.org/pub/scm/bluetooth/bluez.git/commit/profiles/audio/a2dp.c?id=07ae6eb9cdfe73fce94b9af0016c3e896ff5dce9

validatedev, 2021-09-10:
Waiting for the new version of `bluez` then. Thanks for all of your effort!

andirsun, 2023-02-23:
I dont have any problems with my airpods pro on fedora pipewire, maybe this was fixed ?

stephane-klein, 2023-06-07:
@andirsun same issue for me on Fedora 38 with Airpods Gen1.