bluez/bluez #162: gnome-bluetooth will not work properly after turn off in UI at first time.

andch-nn, 2021-06-23:
Hi, 
When I turned off Bluetooth in gnome UI then turn it on again, devices are not listed.
[test environment]
OS: ubuntu 20.04
kernel: 5.13-rc6

Below is the bug in gnome-bluetooth and Benjamin send a patch to linux-bluetooth.
Bug from gnome-bluetooth:
https://gitlab.gnome.org/GNOME/gnome-bluetooth/-/issues/38
Patch discussion in linux-bluetooth:
https://marc.info/?t=162004758900003&r=1&w=2

Can we merge this patch back to bluez?
Thanks.

validatedev, 2021-06-23:
The experience was flawless (no issue like this one) with Ubuntu 20.04, but on Ubuntu 21.04 the problem is happening.

Vudentz, 2021-06-23:
Fix merged: https://git.kernel.org/pub/scm/bluetooth/bluez.git/commit/?id=c939747f543a76ffd556312f753cf9d36c047c94

closed 2021-06-23 by Vudentz