bluez/bluez #166: Can't connect Amazon Echo Dot

RuiGuilherme, 2021-06-29:
$ `uname -a`
```bash
Linux ArchLinuxRui 5.12.13-arch1-2 #1 SMP PREEMPT Fri, 25 Jun 2021 22:56:51 +0000 x86_64 GNU/Linux
```

$ `pacman -Qn | grep pulse`
```bash
lib32-libpulse 14.2-2
libpulse 14.2-3
pulseaudio 14.2-3
pulseaudio-alsa 1:1.2.5-2
pulseaudio-bluetooth 14.2-3
pulseaudio-equalizer 14.2-3
pulseaudio-equalizer-ladspa 3.0.2-4
```

$ `pacman -Qn | grep blue`
```bash
bluedevil 1:5.22.2.1-1
bluez 5.59-2
bluez-libs 5.59-2
bluez-qt 5.83.0-1
bluez-tools 0.2.0-5
bluez-utils 5.59-2
pulseaudio-bluetooth 14.2-3
```

$ `lspci | grep Network`
```bash
00:14.3 Network controller: Intel Corporation Wireless-AC 9560 [Jefferson Peak] (rev 10)
```

$ `lsusb | grep Bluetooth`
```bash
Bus 001 Device 004: ID 8087:0aaa Intel Corp. Bluetooth 9460/9560 Jefferson Peak (JfP)
```
Vendor: https://ark.intel.com/content/www/us/en/ark/products/99446/intel-wireless-ac-9560.html

Logs:
========

$ `journalctl -f`
```bash
jun 29 15:16:00 ArchLinuxRui bluetoothd[19339]: profiles/audio/avdtp.c:caps_to_list() Invalid capability data in getcap resp
jun 29 15:16:00 ArchLinuxRui bluetoothd[19339]: profiles/audio/avdtp.c:caps_to_list() Invalid capability data in getcap resp
jun 29 15:16:00 ArchLinuxRui bluetoothd[19339]: profiles/audio/a2dp.c:a2dp_select_capabilities() Unable to select SEP
jun 29 15:16:00 ArchLinuxRui audit[19339]: ANOM_ABEND auid=4294967295 uid=0 gid=0 ses=4294967295 pid=19339 comm="bluetoothd" exe="/usr/lib/bluetooth/bluetoothd" sig=11 res=1                                   
jun 29 15:16:00 ArchLinuxRui bluetoothd[19339]: src/service.c:btd_service_connect() a2dp-source profile connect failed for 90:A8:22:74:05:EB: Device or resource busy

jun 29 15:16:00 ArchLinuxRui kernel: bluetoothd[19339]: segfault at 3 ip 00007f3833d4d248 sp 00007fff31665eb0 error 4 in libdbus-1.so.3.19.13[7f3833d28000+2f000]
jun 29 15:16:00 ArchLinuxRui kernel: Code: ec 08 4c 89 e9 44 89 e2 53 41 b9 6c 00 00 00 41 89 c0 48 89 ee bf 01 00 00 00 e8 c3 f9 ff ff 5a 59 e9 8e fe ff ff 0f 1f 40 00 <0f> b6 11 44 89 e6 ff 15 2c 0d 02 00 85 c0 0f 84 76 fe ff ff b8 01

jun 29 15:16:00 ArchLinuxRui systemd-coredump[19623]: [🡕] Process 19339 (bluetoothd) of user 0 dumped core.                                              
        Stack trace of thread 19339:
        #0  0x00007f3833d4d248 n/a (libdbus-1.so.3 + 0x33248)
        #1  0x00007f3833d37e7c _dbus_type_writer_write_basic (libdbus-1.so.3 + 0x1de7c)
        #2  0x00007f3833d3c5c9 dbus_message_iter_append_basic (libdbus-1.so.3 + 0x225c9)
        #3  0x000055b12297b340 n/a (bluetoothd + 0x26340)
        #4  0x000055b122a0ae69 n/a (bluetoothd + 0xb5e69)
        #5  0x000055b122a0b54a n/a (bluetoothd + 0xb654a)
        #6  0x000055b122a0b5e1 n/a (bluetoothd + 0xb65e1)
        #7  0x00007f3833ddeb08 g_slist_foreach (libglib-2.0.so.0 + 0x6fb08)
        #8  0x000055b122a0b6ef n/a (bluetoothd + 0xb66ef)
        #9  0x000055b122a0cf27 g_dbus_send_message (bluetoothd + 0xb7f27)
        #10 0x000055b1229ef6dc n/a (bluetoothd + 0x9a6dc)
        #11 0x000055b1229e2952 n/a (bluetoothd + 0x8d952)
        #12 0x000055b12297a5e0 n/a (bluetoothd + 0x255e0)
        #13 0x000055b12297f1ca n/a (bluetoothd + 0x2a1ca)
        #14 0x000055b122981517 n/a (bluetoothd + 0x2c517)
        #15 0x000055b122986dc9 n/a (bluetoothd + 0x31dc9)
[... 253 lines cut ...]
  [CHG] Controller 7C:B2:7D:E1:C1:01 UUIDs: 00001800-0000-1000-8000-00805f9b34fb
  [CHG] Controller 7C:B2:7D:E1:C1:01 UUIDs: 00001108-0000-1000-8000-00805f9b34fb
  [CHG] Controller 7C:B2:7D:E1:C1:01 UUIDs: 0000110a-0000-1000-8000-00805f9b34fb
  [CHG] Controller 7C:B2:7D:E1:C1:01 UUIDs: 00001106-0000-1000-8000-00805f9b34fb
  [CHG] Controller 7C:B2:7D:E1:C1:01 Class: 0x003c010c
  [CHG] Controller 7C:B2:7D:E1:C1:01 UUIDs: 00001133-0000-1000-8000-00805f9b34fb
  [CHG] Controller 7C:B2:7D:E1:C1:01 UUIDs: 0000110e-0000-1000-8000-00805f9b34fb
  [CHG] Controller 7C:B2:7D:E1:C1:01 UUIDs: 00001105-0000-1000-8000-00805f9b34fb
  [CHG] Controller 7C:B2:7D:E1:C1:01 UUIDs: 00001132-0000-1000-8000-00805f9b34fb
  [CHG] Controller 7C:B2:7D:E1:C1:01 UUIDs: 00001200-0000-1000-8000-00805f9b34fb
  [CHG] Controller 7C:B2:7D:E1:C1:01 UUIDs: 00001104-0000-1000-8000-00805f9b34fb
  [CHG] Controller 7C:B2:7D:E1:C1:01 UUIDs: 00005005-0000-1000-8000-0002ee000001
  [CHG] Controller 7C:B2:7D:E1:C1:01 UUIDs: 00001112-0000-1000-8000-00805f9b34fb
  [CHG] Controller 7C:B2:7D:E1:C1:01 UUIDs: 0000110c-0000-1000-8000-00805f9b34fb
  [CHG] Controller 7C:B2:7D:E1:C1:01 UUIDs: 00001801-0000-1000-8000-00805f9b34fb
  [CHG] Controller 7C:B2:7D:E1:C1:01 UUIDs: 0000112f-0000-1000-8000-00805f9b34fb
  [CHG] Controller 7C:B2:7D:E1:C1:01 UUIDs: 0000110b-0000-1000-8000-00805f9b34fb
  [CHG] Controller 7C:B2:7D:E1:C1:01 UUIDs: 0000180a-0000-1000-8000-00805f9b34fb
  [CHG] Controller 7C:B2:7D:E1:C1:01 UUIDs: 00001800-0000-1000-8000-00805f9b34fb
  [CHG] Controller 7C:B2:7D:E1:C1:01 UUIDs: 00001108-0000-1000-8000-00805f9b34fb
  [CHG] Controller 7C:B2:7D:E1:C1:01 UUIDs: 0000110a-0000-1000-8000-00805f9b34fb
  [CHG] Controller 7C:B2:7D:E1:C1:01 UUIDs: 00001106-0000-1000-8000-00805f9b34fb
  [CHG] Controller 7C:B2:7D:E1:C1:01 Alias: ArchLinuxRui
  [CHG] Device 90:A8:22:74:05:EB Connected: no
```

RuiGuilherme, 2021-06-30:
I thought there might be a issue with the Pulseaudio, I switched to pipeware and I get the same issue.

navarobot, 2021-07-02:
Same problem.
Information:
- rtl8761b (ID 0bda:8771 Realtek Semiconductor Corp. Bluetooth Radio)
- ArchLinux
- Linux NAVAROBOT 5.12.14-zen1-1-zen #1 ZEN SMP PREEMPT Thu, 01 Jul 2021 03:35:38 +0000 x86_64 GNU/Linux
- Bluez 5.59-2
- Pipewire, pipewire-pulse 1:0.3.31-1

Journalctl:
```
bluetoothd[5750]: profiles/audio/avdtp.c:caps_to_list() Invalid capability data in getcap resp
kernel: bluetoothd[5750]: segfault at 3 ip 00007fbdff469248 sp 00007fff14e4e800 error 4 in libdbus-1.so.3.19.13[7fbdff444000+2f000]
bluetoothd[5750]: profiles/audio/avdtp.c:caps_to_list() Invalid capability data in getcap resp
bluetoothd[5750]: profiles/audio/a2dp.c:a2dp_select_capabilities() Unable to select SEP
bluetoothd[5750]: src/service.c:btd_service_connect() a2dp-source profile connect failed for F4:4E:FD:5B:FD:1E: Device or resource busy
systemd-coredump[6134]: [🡕] Process 5750 (bluetoothd) of user 0 dumped core.
systemd[1]: bluetooth.service: Main process exited, code=dumped, status=11/SEGV
```
Files:
[bluetoothctl_connect.txt](https://github.com/bluez/bluez/files/6755937/bluetoothctl_connect.txt)

[rtl8761b_connecting_to_Fiero_FR500.pcapng.gz](https://github.com/bluez/bluez/files/6755908/rtl8761b_connecting_to_Fiero_FR500.pcapng.gz)

EdwarDu, 2021-07-02:
Could be the same as https://github.com/bluez/bluez/issues/57
you can check the `dmesg` output, if there is a segfault on bluetoothd then probably you have the same problem as me (also with echo dot)
fixed by: compile and install bluez from master branch.

RuiGuilherme, 2021-07-02:
> fixed by: compile and install bluez from master branch.

This actually worked for me. Thanks so much for the quick reply! 

Quick tip in case someone lands here: Connecting to your computer through the web interface Alexa will only be input.

Vudentz, 2021-07-09:
This should have been fixed in 5.60

closed 2021-07-09 by Vudentz