bluez/bluez #167: Failed to set mode: Failed (0x03) for 0a12:0001 Cambridge Silicon Radio, Ltd Bluetooth Dongle (HCI mode)

alidasomi, 2021-07-01:
I have bougth an CSR 4.0 bluthooth dongle.It cant find any device.
My ubuntu version is 20.14.
my kernel version is 5.8.0-59-generic.

`sudo service bluetooth status`:

```
    Loaded: loaded (/lib/systemd/system/bluetooth.service; enabled; vendor preset: enabled)
     Active: active (running) since Fri 2021-07-02 03:15:57 +0430; 4min 14s ago
       Docs: man:bluetoothd(8)
   Main PID: 11949 (bluetoothd)
     Status: "Running"
      Tasks: 1 (limit: 19003)
     Memory: 920.0K
     CGroup: /system.slice/bluetooth.service
             └─11949 /usr/lib/bluetooth/bluetoothd

جولای 02 03:15:57 heydar-ThinkStation-P300 systemd[1]: Starting Bluetooth service...
جولای 02 03:15:57 heydar-ThinkStation-P300 bluetoothd[11949]: Bluetooth daemon 5.53
جولای 02 03:15:57 heydar-ThinkStation-P300 systemd[1]: Started Bluetooth service.
جولای 02 03:15:57 heydar-ThinkStation-P300 bluetoothd[11949]: Starting SDP server
جولای 02 03:15:57 heydar-ThinkStation-P300 bluetoothd[11949]: Bluetooth management interface 1.17 initialized
جولای 02 03:15:57 heydar-ThinkStation-P300 bluetoothd[11949]: Endpoint registered: sender=:1.75 path=/MediaEndpo>
جولای 02 03:15:57 heydar-ThinkStation-P300 bluetoothd[11949]: Endpoint registered: sender=:1.75 path=/MediaEndpo>
جولای 02 03:15:59 heydar-ThinkStation-P300 bluetoothd[11949]: Loading LTKs timed out for hci0
جولای 02 03:16:07 heydar-ThinkStation-P300 bluetoothd[11949]: Failed to set mode: Failed (0x03)
```

Vudentz, 2021-07-06:
@alainmic can you collect the HCI trace, perhaps some HCI command not supported.

alidasomi, 2021-07-10:
```
root@heydar-ThinkStation-P300:~# hcidump 
HCI sniffer - Bluetooth packet analyzer ver 5.53
device: hci0 snap_len: 1500 filter: 0xffffffffffffffff

< HCI Command: Reset (0x03|0x0003) plen 0
```

after restart bluetooth it log `HCI Command: Reset (0x03|0x0003) plen 0` and stay in this stage.

Vudentz, 2021-09-10:
So there is no Command Complete after that?

closed 2025-08-27 by github-actions[bot]