bluez/bluez #168: [Mesh] When provisioning a node with Input/Output OOB, OOB will not be used

FBNH, 2021-07-04:
When trying to provision a new node to the network even if the node supports input/output OOB, the provisioner will always choose no OOB instead.
This isn't the case for static OOB.

Instead, there should be a prompt message asking the user which OOB type to use in such case.

bgix, 2021-07-06:
Hi @FBNH -- As of release v5.59, this should no longer be the case. By default, the mesh-cfgclient in v5.58 and earlier only supported:
static-oob
out-numeric
in-numeric

But as of 5.59, it supports:
static-oob
push
twist
blink
beep
vibrate
public-oob
out-alpha
in-alpha
out-numeric
in-numeric

These capabilities can be modified in tools/mesh-cfgclient.c by modifying the contents of:
static const char * const caps[] = {"static-oob",
				"push",
				"twist",
				"blink",
				"beep",
				"vibrate",
				"public-oob",
				"out-alpha",
				"in-alpha",
				"out-numeric",
				"in-numeric"};

FBNH, 2021-07-07:
@bgix thanks for the answer!
I hope we'll update to the new version soon

closed 2025-08-27 by github-actions[bot]