bluez/bluez #170: Scanning sensors with low RSSI levels

edflahive, 2021-07-06:
Hi,

We are scanning BLE sensors using BlueZ versions 5.43 and 5.55 on an embedded Linux platform with a Panasonic 1326b module. These sensors advertise at a rate of 4s and are at fixed distances from the Linux device.

Basically, we are using C code to perform the scans similar to hcitool lescan –duplicates. We consider that this will give us better performance than using the DBus interface, even though these tools are deprecated?

We don’t need to capture every advertisement from the sensors and capturing one per minute is adequate. However, intermittently we fail to capture an advertisement for a number of minutes for sensors positioned towards the edge of the usable range i.e. we capture advertisements roughly every 10s for 90% of the time, but then we can fail to get any for a couple of minutes. We don’t detect any difference between the two BlueZ versions in use.

The root cause is likely to be low or varying RSSI levels, but we were wondering if there is anything that can be applied on the BlueZ to improve performance? For instance, does BlueZ start filtering out advertisements from MAC addresses for a certain time if it sees some badly formed messages?

Any suggestions on improving performance are most welcome. We have limited scope to change the hardware that is deployed.

Vudentz, 2021-07-06:
@edflahive if you are doing HCI commands directly that is pretty much interfacing with the controller directly, that said the Kernel would use extended scanning if the controller supports it but perhaps your problem is with actual parameter you are using, or perhaps you would have to fine tune the parameters in order to have better results.

We normally don't recommend using HCI commands directly since you had probably get in trouble if there were other application interacting with the controller, also you can have some filters over D-Bus e.g. bluetoothctl> scan.le on

edflahive, 2021-07-06:
Since it's an embedded system, we have full control over the use of the controller and using HCI is a little safer for us. However, I should have mentioned that we are using the same scan interval and window which should give us continuous scanning.

I'll try out the filter suggestion with both HCI and bluetoothctl to see if they perform better, but I was nervous about using bluetoothctl as I understand that duplicate packets would be filtered out.

closed 2021-09-01 by Vudentz