bluez/bluez #171: I can't connect my headphones

Toadfield, 2021-07-06:
I tried to connect it via the kde bluetooth menu.
When I try to connect my bluetooth headphones,then I see [this](https://preview.redd.it/snzdfons5f971.png?width=502&format=png&auto=webp&s=33f51169d13d82d814c304eb5f9a705d7e4f5b65).
And kde shows that they are connected for a second,but then they disconnect automatically.
I use [these headphones](https://www.amazon.com/dp/B07F62W3TQ?psc=1).
[This is my system info](https://preview.redd.it/zaajsq126f971.png?width=503&format=png&auto=webp&s=4da705bfd4ad151bdfc06964b7c816080bd0aef6).
I asked on reddit and the people said that this is probably a bluez bug.

Vudentz, 2021-07-06:
Please have a look at: https://github.com/bluez/bluez/issues/157

Toadfield, 2021-07-06:
When will the bug be fixed?

Vudentz, 2021-07-06:
@Toadfield the bug has been fixed, we just didn't have a release yet.

Toadfield, 2021-07-06:
What is the best workaround right now?

closed 2021-07-06 by Toadfield