bluez/bluez #172: Bluez can't connect permanently to a Bluetooth LE remote “Function not implemented (38)”

besi, 2021-07-09:
Since I was not yet able to resolve my issue I am cross-posting this issue here.
You'll find the original post [here](https://stackoverflow.com/questions/68202067/bluez-cant-connect-permanently-to-a-bluetooth-le-remote-function-not-implement) 

I have a new RaspberryPi buster image:

> Linux stereo 5.10.17+ #1414 Fri Apr 30 13:16:27 BST 2021 armv6l GNU/Linux

I did update bluez to it to `bluez 5.52`. What is already interesting is the fact that I have the following entry regarding `5.50` in my `/var/log/syslog`:

> Jun 30 23:15:09 stereopida bluetoothd[557]: Bluetooth daemon 5.50

That seems to indicate that the 5.50 version that was previously installed is still lingering around.

I did this same setup on other machines and there and then I could get it to work but this time there seems to be a problem.

Now using bluetoothctl I can see my device and pair it but the connection is instantly dropped:

```
[bluetooth]# trust 84:EB:18:0C:92:0F 
Changing 84:EB:18:0C:92:0F trust succeeded
[bluetooth]# pair 84:EB:18:0C:92:0F 
Attempting to pair with 84:EB:18:0C:92:0F
[CHG] Device 84:EB:18:0C:92:0F Connected: yes
Failed to pair: org.bluez.Error.AuthenticationFailed
[CHG] Device 84:EB:18:0C:92:0F Connected: no
[bluetooth]# connect 84:EB:18:0C:92:0F 
Attempting to connect to 84:EB:18:0C:92:0F
[CHG] Device 84:EB:18:0C:92:0F Connected: yes
Failed to connect: org.bluez.Error.Failed
[CHG] Device 84:EB:18:0C:92:0F Connected: no
```

When I check in the logs I do see the following:

```
Jun 30 23:15:53 stereopida bluetoothd[567]: src/device.c:bonding_request_new() Requesting bonding for 84:EB:18:0C:92:0F
Jun 30 23:15:53 stereopida bluetoothd[567]: src/agent.c:agent_ref() 0x7b62b0: ref=3
Jun 30 23:15:53 stereopida bluetoothd[567]: src/agent.c:agent_unref() 0x7b62b0: ref=2
Jun 30 23:15:53 stereopida bluetoothd[567]: src/adapter.c:suspend_discovery() 
Jun 30 23:15:53 stereopida bluetoothd[567]: src/adapter.c:adapter_bonding_attempt() hci0 bdaddr 84:EB:18:0C:92:0F type 1 io_cap 0x04
Jun 30 23:15:55 stereopida bluetoothd[567]: src/adapter.c:connected_callback() hci0 device 84:EB:18:0C:92:0F connected eir_len 13
Jun 30 23:15:55 stereopida bluetoothd[567]: src/adapter.c:pair_device_complete() Failed (0x03)
Jun 30 23:15:55 stereopida bluetoothd[567]: src/adapter.c:bonding_attempt_complete() hci0 bdaddr 84:EB:18:0C:92:0F type 1 status 0x3
Jun 30 23:15:55 stereopida bluetoothd[567]: src/device.c:device_bonding_complete() bonding 0x7ba428 status 0x03
Jun 30 23:15:55 stereopida bluetoothd[567]: src/device.c:device_bonding_failed() status 3
Jun 30 23:15:55 stereopida bluetoothd[567]: src/agent.c:agent_unref() 0x7b62b0: ref=1
Jun 30 23:15:55 stereopida bluetoothd[567]: src/adapter.c:resume_discovery() 
Jun 30 23:15:55 stereopida bluetoothd[567]: src/adapter.c:dev_disconnected() Device 84:EB:18:0C:92:0F disconnected, reason 0
Jun 30 23:15:55 stereopida bluetoothd[567]: src/adapter.c:adapter_remove_connection() 
Jun 30 23:15:55 stereopida bluetoothd[567]: plugins/policy.c:disconnect_cb() reason 0
Jun 30 23:15:55 stereopida bluetoothd[567]: src/adapter.c:bonding_attempt_complete() hci0 bdaddr 84:EB:18:0C:92:0F type 1 status 0xe
Jun 30 23:15:55 stereopida bluetoothd[567]: src/device.c:device_bonding_complete() bonding (nil) status 0x0e
Jun 30 23:15:55 stereopida bluetoothd[567]: src/device.c:device_bonding_failed() status 14
Jun 30 23:15:55 stereopida bluetoothd[567]: src/adapter.c:resume_discovery() 
Jun 30 23:16:04 stereopida bluetoothd[567]: src/device.c:device_connect_le() Connection attempt to: 84:EB:18:0C:92:0F
Jun 30 23:16:08 stereopida systemd[1]: systemd-hostnamed.service: Succeeded.
Jun 30 23:16:09 stereopida bluetoothd[567]: src/adapter.c:connected_callback() hci0 device 84:EB:18:0C:92:0F connected eir_len 13
Jun 30 23:16:09 stereopida bluetoothd[567]: src/device.c:att_connect_cb() connect error: Function not implemented (38)
Jun 30 23:16:09 stereopida bluetoothd[567]: src/adapter.c:dev_disconnected() Device 84:EB:18:0C:92:0F disconnected, reason 0
Jun 30 23:16:09 stereopida bluetoothd[567]: src/adapter.c:adapter_remove_connection() 
Jun 30 23:16:09 stereopida bluetoothd[567]: plugins/policy.c:disconnect_cb() reason 0
Jun 30 23:16:09 stereopida bluetoothd[567]: src/adapter.c:bonding_attempt_complete() hci0 bdaddr 84:EB:18:0C:92:0F type 1 status 0xe
Jun 30 23:16:09 stereopida bluetoothd[567]: src/device.c:device_bonding_complete() bonding (nil) status 0x0e
Jun 30 23:16:09 stereopida bluetoothd[567]: src/device.c:device_bonding_failed() status 14
Jun 30 23:16:09 stereopida bluetoothd[567]: src/adapter.c:resume_discovery() 
Jun 30 23:16:15 stereopida bluetoothd[567]: src/agent.c:agent_disconnect() Agent :1.20 disconnected
Jun 30 23:16:15 stereopida bluetoothd[567]: src/agent.c:agent_destroy() agent :1.20
Jun 30 23:16:15 stereopida bluetoothd[567]: src/agent.c:agent_unref() 0x7b62b0: ref=0

```
[... 326 lines cut ...]
    own_bdaddr_type 0 min_interval 24 max_interval 40
    latency 0 supervision_to 42 min_ce 0 max_ce 0
> HCI Event: Command Status (0x0f) plen 4
    LE Create Connection (0x08|0x000d) status 0x00 ncmd 1
> HCI Event: LE Meta Event (0x3e) plen 19
    LE Connection Complete
      status 0x00 handle 64, role master
      bdaddr 84:EB:18:0C:92:0F (Public)
< HCI Command: LE Read Remote Used Features (0x08|0x0016) plen 2
> HCI Event: Command Status (0x0f) plen 4
    LE Read Remote Used Features (0x08|0x0016) status 0x00 ncmd 1
> HCI Event: LE Meta Event (0x3e) plen 12
    LE Read Remote Used Features Complete
      status 0x3e handle 64
      Error: Connection Failed to be Established
< HCI Command: Disconnect (0x01|0x0006) plen 3
    handle 64 reason 0x13
    Reason: Remote User Terminated Connection
> HCI Event: Disconn Complete (0x05) plen 4
    status 0x00 handle 64 reason 0x3e
    Reason: Connection Failed to be Established
> HCI Event: Command Status (0x0f) plen 4
    Disconnect (0x01|0x0006) status 0x12 ncmd 1
    Error: Invalid HCI Command Parameters
```

Vudentz, 2021-07-09:
The connection is unstable it seems:

`Reason: Connection Failed to be Established (0x3e)`

This is down to the link-layer/controller so we can't do much if the connection keeps failing like this, so perhaps you will need to replace your controller or check if the problem is in the peripheral.

besi, 2021-07-10:
Whats a bit weird that I see the „Not implemented“ message.

closed 2021-08-05 by Vudentz

johnsto, 2022-01-03:
For the benefit of anyone else that ends up on this issue with the same symptoms, please heed @Vudentz's advice - in my case I had simply forgotten to re-attach those little wireless aerials to the back of my PC after a move! Please check your peripherals before wasting several hours stracing system calls like I did 🤣

xuanchibkd14, 2022-08-09:
anyones can solved it, i get it with raspberry pi 4

Vudentz, 2022-08-09:
> anyones can solved it, i get it with raspberry pi 4

Perhaps check if there is something with the antenna like the comment above.

grrowl, 2023-05-17:
@xuanchibkd14 wifi and bluetooth share a chip, try `ip link set wlan0 down` before you attempt pairing

david-halls, 2023-09-18:
@grrowl - this did actually fix for me, but the Pi should be able to support coexistence of BLE and Wi-Fi! I am seeing this issue on a specific kernel/BlueZ version against a BLE remote control, having never been a problem before. Are there some coexistence or connection interval settings that can be changed?