bluez/bluez #174: [5.60] Cannot connect to multiple devices simultaneously

jeffshee, 2021-07-16:
Hi, as described in the title, my system can't connect to multiple devices simultaneously.
Specifically, when there is already 1 device connected with the system, any attempts to connect other devices will fail.
Here is the status log:
```
 ❯ journalctl -fu bluetooth.service                                                                                                                                                                                                 [02:04:52]
-- Journal begins at Mon 2021-06-07 08:41:38 JST. --
Jul 17 01:34:56 ikaros bluetoothd[1079]: src/profile.c:ext_connect() Hands-Free Voice gateway failed connect to 00:1B:66:E5:5D:85: Connection refused (111)
Jul 17 01:34:56 ikaros bluetoothd[1079]: profiles/audio/avdtp.c:avdtp_connect_cb() connect to 00:1B:66:E5:5D:85: Connection refused (111)
Jul 17 01:34:56 ikaros bluetoothd[1079]: src/service.c:btd_service_connect() a2dp-source profile connect failed for 00:1B:66:E5:5D:85: Device or resource busy
Jul 17 01:34:58 ikaros bluetoothd[1079]: profiles/audio/avdtp.c:avdtp_connect_cb() connect to 00:1B:66:E5:5D:85: Connection refused (111)
Jul 17 01:35:16 ikaros bluetoothd[1079]: src/service.c:btd_service_connect() a2dp-source profile connect failed for 00:1B:66:E5:5D:85: Device or resource busy
Jul 17 01:35:36 ikaros bluetoothd[1079]: /org/bluez/hci0/dev_00_1B_66_E5_5D_85/sep3/fd0: fd(42) ready
Jul 17 01:51:22 ikaros bluetoothd[1079]: src/device.c:search_cb() 48:01:C5:0E:54:C8: error updating services: Connection timed out (110)
Jul 17 01:51:37 ikaros bluetoothd[1079]: profiles/audio/avdtp.c:avdtp_connect_cb() connect to 48:01:C5:0E:54:C8: Connection reset by peer (104)
Jul 17 01:52:20 ikaros bluetoothd[1079]: profiles/audio/avdtp.c:avdtp_connect_cb() connect to 48:01:C5:0E:54:C8: Connection refused (111)
Jul 17 02:00:35 ikaros bluetoothd[1079]: src/device.c:search_cb() 48:01:C5:0E:54:C8: error updating services: Connection timed out (110)
Jul 17 02:06:03 ikaros bluetoothd[1079]: src/device.c:search_cb() 48:01:C5:0E:54:C8: error updating services: Connection timed out (110)
Jul 17 02:06:19 ikaros bluetoothd[1079]: profiles/input/device.c:ioctl_is_connected() Can't get HIDP connection info
Jul 17 02:06:24 ikaros bluetoothd[1079]: profiles/input/device.c:control_connect_cb() connect to F4:93:9F:CE:D9:B6: Host is down (112)
```
For this log, first I had my earbuds (Sennheiser CX400BT TW) `00:1B:66:E5:5D:85` connected. Then I tried to pair my phone `48:01:C5:0E:54:C8`, the pairing was successful but it won't connect. I also tried to connect a PS4 controller `F4:93:9F:CE:D9:B6` after that and failed.

I'm not familiar with the Bluetooth stuff and hopefully, I had reported this issue to the right place.
Please let me know if you need anything else.

OS: Fedora 34
```
 ❯ rpm -qa | grep bluez                                                                                                                                                                                                             [02:23:17]
bluez-5.60-1.fc34.x86_64
bluez-libs-5.60-1.fc34.x86_64
bluez-obexd-5.60-1.fc34.x86_64
bluez-cups-5.60-1.fc34.x86_64
bluez-libs-5.60-1.fc34.i686
```

jeffshee, 2021-07-16:
Result of `sudo btmon -w hci.log`
With the earbuds connected, I recorded the traces while trying to connect the PS4 controller.
[hci.log](https://github.com/bluez/bluez/files/6832438/hci.log)

Vudentz, 2021-07-16:
@jeffshee It doesn't look like there is any connection attempt though, it start a discovery though so I assume it is scanning all the time which would also make the controller busy plus it seems you are playing as well as there is a lot of data coming thru, so try with bluetoothctl and make sure you are not scanning and not playing anything e.g. connect your phone first and only then your headset.

kakra, 2021-07-17:
Maybe the same issue I'm having with kdeconnectd running which results in permanent continuous scanning? Maybe we then need some QoS after all?

jeffshee, 2021-07-17:
@Vudentz Hi, this time I recorded the stack without playing the music.
Here is the procedure during the recording:
1. Connected to earbuds.
2. Tried to connect to the PS4 controller several times, failed.
3. Tried to connect to my phone several times, failed.
```
 ❯ journalctl -fu bluetooth.service                                                                         [01:53:21]
-- Journal begins at Mon 2021-06-07 08:41:38 JST. --
Jul 18 01:52:04 ikaros bluetoothd[1084]: Endpoint registered: sender=:1.73 path=/MediaEndpoint/A2DPSource/sbc
Jul 18 01:52:11 ikaros bluetoothd[1084]: src/profile.c:record_cb() Unable to get Hands-Free Voice gateway SDP record: Host is down
Jul 18 01:52:12 ikaros bluetoothd[1084]: Endpoint unregistered: sender=:1.73 path=/MediaEndpoint/A2DPSource/ldac
Jul 18 01:52:12 ikaros bluetoothd[1084]: Endpoint unregistered: sender=:1.73 path=/MediaEndpoint/A2DPSource/aac
Jul 18 01:52:12 ikaros bluetoothd[1084]: Endpoint unregistered: sender=:1.73 path=/MediaEndpoint/A2DPSink/sbc
Jul 18 01:52:12 ikaros bluetoothd[1084]: Endpoint unregistered: sender=:1.73 path=/MediaEndpoint/A2DPSource/sbc
Jul 18 01:52:16 ikaros bluetoothd[1084]: Endpoint registered: sender=:1.127 path=/MediaEndpoint/A2DPSource/ldac
Jul 18 01:52:16 ikaros bluetoothd[1084]: Endpoint registered: sender=:1.127 path=/MediaEndpoint/A2DPSource/aac
Jul 18 01:52:16 ikaros bluetoothd[1084]: Endpoint registered: sender=:1.127 path=/MediaEndpoint/A2DPSink/sbc
Jul 18 01:52:16 ikaros bluetoothd[1084]: Endpoint registered: sender=:1.127 path=/MediaEndpoint/A2DPSource/sbc
Jul 18 01:54:22 ikaros bluetoothd[1084]: src/service.c:btd_service_connect() a2dp-source profile connect failed for 00:1B:66:E5:5D:85: Device or resource busy
Jul 18 01:54:39 ikaros bluetoothd[1084]: profiles/input/device.c:ioctl_is_connected() Can't get HIDP connection info
Jul 18 01:54:44 ikaros bluetoothd[1084]: profiles/input/device.c:control_connect_cb() connect to F4:93:9F:CE:D9:B6: Host is down (112)
Jul 18 01:54:52 ikaros bluetoothd[1084]: profiles/input/device.c:ioctl_is_connected() Can't get HIDP connection info
Jul 18 01:55:16 ikaros bluetoothd[1084]: profiles/input/device.c:control_connect_cb() connect to F4:93:9F:CE:D9:B6: Connection timed out (110)
Jul 18 01:55:28 ikaros bluetoothd[1084]: profiles/input/device.c:ioctl_is_connected() Can't get HIDP connection info
Jul 18 01:56:10 ikaros bluetoothd[1084]: profiles/input/device.c:control_connect_cb() connect to F4:93:9F:CE:D9:B6: Connection refused (111)
Jul 18 01:56:49 ikaros bluetoothd[1084]: src/device.c:search_cb() 48:01:C5:0E:54:C8: error updating services: Device or resource busy (16)
Jul 18 01:57:31 ikaros bluetoothd[1084]: src/device.c:search_cb() 48:01:C5:0E:54:C8: error updating services: Connection timed out (110)
``` 
[hci.log](https://github.com/bluez/bluez/files/6835010/hci.log)

(The controller can be connected without any problem right after I disconnected the earbuds.)

jeffshee, 2021-07-17:
@kakra Hi, not sure about kdeconnectd, since it was not present in my system. I do use GSConnect though. But it shouldn't cause the Bluetooth issue since KDE Connect uses LAN/WLAN, not Bluetooth, right?

kakra, 2021-07-17:
kdeconnectd has a Bluetooth module which can be disabled at compile time. Since I'm running with the Bluetooth module disabled, the problem is gone. Previously, it caused a continuous scan due to triggering kio for bluetooth:// every 20 seconds. I think one user reported that GSConnect has similar behavior, you may try uninstalling it and see if it fixes the problem.

jeffshee, 2021-07-18:
@kakra Hi, thanks for your suggestion. I had tried to uninstall GSConnect and reboot, but it still didn't fix the problem. Could you point me to the bug report that you had mentioned regarding the GSConnect or KDE Connect?

kakra, 2021-07-18:
@jeffshee You may want to look here:
* https://github.com/bluez/bluez/issues/123

jeffshee, 2021-07-19:
@Vudentz Hi, here is the attempt with `bluetoothctl`, hope it is helpful.

```
 ❯ bluetoothctl                                                   [00:42:22]
Agent registered
[CHG] Controller C0:E4:34:FB:2C:B0 Pairable: yes
[bluetooth]# power off
Changing power off succeeded
[CHG] Controller C0:E4:34:FB:2C:B0 Powered: no
[CHG] Controller C0:E4:34:FB:2C:B0 Discovering: no
[CHG] Controller C0:E4:34:FB:2C:B0 Class: 0x00000000
[bluetooth]# power on
[CHG] Controller C0:E4:34:FB:2C:B0 Class: 0x007c0104
Changing power on succeeded
[CHG] Controller C0:E4:34:FB:2C:B0 Powered: yes
[bluetooth]# show
Controller C0:E4:34:FB:2C:B0 (public)
	Name: Ikaros Type-α
	Alias: Ikaros Type-α
	Class: 0x007c0104
	Powered: yes
	Discoverable: no
	DiscoverableTimeout: 0x000000b4
	Pairable: yes
	UUID: Message Notification Se.. (00001133-0000-1000-8000-00805f9b34fb)
	UUID: A/V Remote Control        (0000110e-0000-1000-8000-00805f9b34fb)
	UUID: OBEX Object Push          (00001105-0000-1000-8000-00805f9b34fb)
	UUID: Message Access Server     (00001132-0000-1000-8000-00805f9b34fb)
	UUID: PnP Information           (00001200-0000-1000-8000-00805f9b34fb)
	UUID: IrMC Sync                 (00001104-0000-1000-8000-00805f9b34fb)
	UUID: Headset                   (00001108-0000-1000-8000-00805f9b34fb)
	UUID: A/V Remote Control Target (0000110c-0000-1000-8000-00805f9b34fb)
	UUID: Generic Attribute Profile (00001801-0000-1000-8000-00805f9b34fb)
	UUID: Phonebook Access Server   (0000112f-0000-1000-8000-00805f9b34fb)
	UUID: Audio Sink                (0000110b-0000-1000-8000-00805f9b34fb)
	UUID: Device Information        (0000180a-0000-1000-8000-00805f9b34fb)
	UUID: Generic Access Profile    (00001800-0000-1000-8000-00805f9b34fb)
	UUID: Handsfree Audio Gateway   (0000111f-0000-1000-8000-00805f9b34fb)
	UUID: Audio Source              (0000110a-0000-1000-8000-00805f9b34fb)
	UUID: OBEX File Transfer        (00001106-0000-1000-8000-00805f9b34fb)
	Modalias: usb:v1D6Bp0246d053C
	Discovering: no
	Roles: central
	Roles: peripheral
Advertising Features:
	ActiveInstances: 0x00 (0)
	SupportedInstances: 0x05 (5)
	SupportedIncludes: tx-power
	SupportedIncludes: appearance
	SupportedIncludes: local-name
[bluetooth]# devices
Device F4:93:9F:CE:D9:B6 Wireless Controller
Device 48:01:C5:0E:54:C8 OnePlus 7
Device 00:1B:66:E5:5D:85 CX 400BT TW
[bluetooth]# connect 00:1B:66:E5:5D:85 
Attempting to connect to 00:1B:66:E5:5D:85
[CHG] Device 00:1B:66:E5:5D:85 Connected: yes
Connection successful
[CHG] Device 00:1B:66:E5:5D:85 ServicesResolved: yes
[CHG] Device F4:93:9F:CE:D9:B6 Connected: yes
[CX 400BT TW]# connect F4:93:9F:CE:D9:B6 
Attempting to connect to F4:93:9F:CE:D9:B6
Failed to connect: org.bluez.Error.Failed
[CHG] Device F4:93:9F:CE:D9:B6 Connected: no
[CX 400BT TW]# connect F4:93:9F:CE:D9:B6 
Attempting to connect to F4:93:9F:CE:D9:B6
Failed to connect: org.bluez.Error.Failed
[CHG] Device F4:93:9F:CE:D9:B6 Connected: yes
[CHG] Device F4:93:9F:CE:D9:B6 Connected: no
[CX 400BT TW]# 
```

Service output:
```
Jul 20 00:42:55 ikaros bluetoothd[1077]: /org/bluez/hci0/dev_00_1B_66_E5_5D_85/sep3/fd6: fd(32) ready
Jul 20 00:43:08 ikaros bluetoothd[1077]: profiles/input/device.c:ioctl_is_connected() Can't get HIDP connection info
Jul 20 00:43:48 ikaros bluetoothd[1077]: profiles/input/device.c:control_connect_cb() connect to F4:93:9F:CE:D9:B6: Connection refused (111)
Jul 20 00:44:03 ikaros bluetoothd[1077]: profiles/input/device.c:ioctl_is_connected() Can't get HIDP connection info
Jul 20 00:44:08 ikaros bluetoothd[1077]: profiles/input/device.c:control_connect_cb() connect to F4:93:9F:CE:D9:B6: Host is down (112)
```

Traces:
[hci.log](https://github.com/bluez/bluez/files/6842270/hci.log)

Additional info:
IIRC, it worked fine with Fedora 33, under the same Bluetooth devices and system.

pgborges, 2021-12-02:
I don't know what your bluetooth controller is, but I have the exact same issue here with an RTL8723BE. I have tried other kernel versions, unloading the wifi module and btcoexist and tweaking some module configurations, but nothing changes. With my headset connect, the mouse connection fails. If I disconnect th headset, then the mouse connects, but then the headset cannot connect while the mouse is connected.
I tested the very same OS (just installed my ssd into another machine and booted) and it worked flawlessly with an Intel controller. I guess it could be a problem with the realtek driver itself or maybe there's some issue between Bluez and the driver.

systemctl status bluetooth returns this:

`dez 01 11:06:48 agnus bluetoothd[845]: Endpoint registered: sender=:1.53 path=/MediaEndpoint/A2DPSource/sbc_xq_512`
`dez 01 11:06:48 agnus bluetoothd[845]: Endpoint registered: sender=:1.53 path=/MediaEndpoint/A2DPSink/sbc_xq_552`
`dez 01 11:06:48 agnus bluetoothd[845]: Endpoint registered: sender=:1.53 path=/MediaEndpoint/A2DPSource/sbc_xq_552`
`dez 01 11:07:38 agnus bluetoothd[845]: src/device.c:search_cb() 34:88:5D:58:E7:9E: error updating services: Host is down (112)`
`dez 01 11:08:06 agnus bluetoothd[845]: profiles/input/device.c:ioctl_is_connected() Can't get HIDP connection info`
`dez 01 11:41:18 agnus bluetoothd[845]: src/profile.c:ext_connect() Hands-Free Voice gateway failed connect to 9D:EE:5D:2B:3A:DD: Connection refused (111)`
`dez 01 11:41:19 agnus bluetoothd[845]: profiles/audio/avdtp.c:avdtp_connect_cb() connect to 9D:EE:5D:2B:3A:DD: Connection refused (111)`
`dez 01 11:41:19 agnus bluetoothd[845]: profiles/input/device.c:ioctl_is_connected() Can't get HIDP connection info`
`dez 01 11:41:20 agnus bluetoothd[845]: profiles/input/device.c:control_connect_cb() connect to 9D:EE:5D:2B:3A:DD: Connection refused (111)`
`dez 01 11:41:23 agnus bluetoothd[845]: profiles/audio/avdtp.c:avdtp_connect_cb() connect to 9D:EE:5D:2B:3A:DD: Connection refused (111)`

I guess its a limitation in the Realtek driver for Linux or some issue with Bluez and Realtek cards (I had a notebook with an Intel Wireless Card and the bluetooth connection worked fine. I could use my bluetooth headset and mouse simultaneously without any issue - I used Manjaro with linux514 in that notebook).

Any help is appreciated.
Thanks.

Inxi output:

`System:    Kernel: 5.15.2-2-MANJARO x86_64 bits: 64 compiler: gcc v: 11.1.0 Desktop: KDE Plasma 5.23.3 Distro: Manjaro Linux`
`Network:   Device-2: Realtek RTL8723BE PCIe Wireless Network Adapter vendor: Hewlett-Packard driver: rtl8723be v: kernel`
`Bluetooth: Device-1: Realtek Bluetooth Radio type: USB driver: btusb v: 0.8 bus-ID: 1-7:4`

goffyara, 2023-04-10:
Same issue Dualshock 4 and Mi Buds 4 Pro
```
02:00.0 Network controller [0280]: Realtek Semiconductor Co., Ltd. RTL8822BE 802.11a/b/g/n/ac WiFi adapter [10ec:b822]
```

```
апр 11 01:03:42 hp bluetoothd[522]: src/profile.c:record_cb() Unable to get Hands-Free Voice gateway SDP record: Host is down
апр 11 01:04:13 hp bluetoothd[522]: profiles/input/device.c:ioctl_is_connected() Can't get HIDP connection info
апр 11 01:04:55 hp bluetoothd[522]: profiles/input/device.c:control_connect_cb() connect to A4:AE:12:9D:1A:34: Connection refused (111)
апр 11 01:05:18 hp bluetoothd[522]: /org/bluez/hci0/dev_6C_D3_EE_58_75_63/fd0: fd(41) ready
апр 11 01:05:43 hp bluetoothd[522]: profiles/input/device.c:ioctl_is_connected() Can't get HIDP connection info
апр 11 01:06:44 hp bluetoothd[522]: src/profile.c:ext_connect() Hands-Free Voice gateway failed connect to 6C:D3:EE:58:75:63: Connection reset by peer (104)
апр 11 01:06:55 hp bluetoothd[522]: profiles/audio/avdtp.c:avdtp_connect_cb() connect to 6C:D3:EE:58:75:63: Connection reset by peer (104)
апр 11 01:07:09 hp bluetoothd[522]: profiles/audio/avdtp.c:avdtp_connect_cb() connect to 6C:D3:EE:58:75:63: Connection reset by peer (104)
```
```
bluez-5.66-1
```

SuperSandro2000, 2025-06-27:
.

SuperSandro2000, 2025-07-10:
Do I need to post more to get rid of the stale label?

SuperSandro2000, 2025-07-11:
sudo btmon -w hci.log [hci.log](https://github.com/user-attachments/files/21189734/hci.log)

viitahenri, 2025-09-21:
Encountering this same issue with a DualSense controller and usb keyboards. While DualSense is connected, I can't make a new connection to keyboards. Using ASUS USB-BT540 on Debian 13.

closed 2025-10-29 by github-actions[bot]

SuperSandro2000, 2025-10-29:
The stale bot is really getting on my nerves