bluez/bluez #175: [5.60] Fake PS3 controller disconnect after connect

Cooky-12, 2021-07-18:
## Archlinux 
#### pacman -Q | grep bluez
```
bluez 5.60-1
bluez-hid2hci 5.60-1
bluez-libs 5.60-1
bluez-plugins 5.60-1
bluez-tools 0.2.0-5
bluez-utils 5.60-1
```
#### uname -a
`5.12.15-arch1-1 #1 SMP PREEMPT Wed, 07 Jul 2021 23:35:29 +0000 x86_64 GNU/Linux`

#### dmesg
```
[ 7530.807836] usb 3-1.4: USB disconnect, device number 11
[ 7535.672637] usb 3-1.4: new full-speed USB device number 12 using xhci_hcd
[ 7535.822000] usb 3-1.4: New USB device found, idVendor=054c, idProduct=0268, bcdDevice= 0.01
[ 7535.822026] usb 3-1.4: New USB device strings: Mfr=0, Product=2, SerialNumber=0
[ 7535.822037] usb 3-1.4: Product: PS3 Controller
[ 7535.829282] usbhid 3-1.4:1.0: can't add hid device: -32
[ 7535.829315] usbhid: probe of 3-1.4:1.0 failed with error -32
[ 7535.928022] usb 3-1.4: USB disconnect, device number 12
[ 7536.695964] usb 3-1.4: new full-speed USB device number 13 using xhci_hcd
[ 7536.848103] usb 3-1.4: New USB device found, idVendor=054c, idProduct=0268, bcdDevice= 5.72
[ 7536.848133] usb 3-1.4: New USB device strings: Mfr=0, Product=2, SerialNumber=0
[ 7536.848146] usb 3-1.4: Product: PS3 Controller
[ 7536.865567] input: PS3 Controller Motion Sensors as /devices/pci0000:00/0000:00:14.0/usb3/3-1/3-1.4/3-1.4:1.0/0003:054C:0268.0008/input/input27
[ 7536.922989] input: PS3 Controller as /devices/pci0000:00/0000:00:14.0/usb3/3-1/3-1.4/3-1.4:1.0/0003:054C:0268.0008/input/input26
[ 7536.924058] sony 0003:054C:0268.0008: input,hiddev96,hidraw3: USB HID v81.11 Joystick [PS3 Controller] on usb-0000:00:14.0-1.4/input0
```
bluetoothctl didn't show pair or authentication info, so I use sixpair to write address to the controller

#### bluetoothctl
```
[bluetooth]# agent on
Agent is already registered
[bluetooth]# default-agent 
Default agent request successful
[bluetooth]# power on
[CHG] Controller 00:1A:7D:DA:71:13 Class: 0x006c010c
Changing power on succeeded
[CHG] Controller 00:1A:7D:DA:71:13 Powered: yes
[bluetooth]# pairable on
Changing pairable on succeeded
[bluetooth]# discoverable on
Changing discoverable on succeeded
[CHG] Controller 00:1A:7D:DA:71:13 Discoverable: yes
[CHG] Device A0:5A:5C:A6:15:5D Connected: yes
[CHG] Device A0:5A:5C:A6:15:5D Connected: no

[bluetooth]# info A0:5A:5C:A6:15:5D 
Device A0:5A:5C:A6:15:5D (public)
	Name: Sony PLAYSTATION(R)3 Controller
	Alias: Sony PLAYSTATION(R)3 Controller
	Class: 0x00002504
	Icon: input-gaming
	Paired: no
	Trusted: yes
	Blocked: no
	Connected: no
	LegacyPairing: no
```
#### bluetoothd
```
bluetoothd[7375]: src/agent.c:add_default_agent() Default agent set to :1.115 /org/bluez/agent
bluetoothd[7375]: src/adapter.c:set_mode() sending set mode command for index 0
bluetoothd[7375]: src/agent.c:agent_ref() 0x558e7de0c3f0: ref=1
bluetoothd[7375]: src/agent.c:register_agent() agent :1.115
bluetoothd[7375]: src/adapter.c:new_settings_callback() Settings: 0x00000ad0
bluetoothd[7375]: src/adapter.c:settings_changed() Changed settings: 0x00000010
bluetoothd[7375]: src/adapter.c:settings_changed() Pending settings: 0x00000000
bluetoothd[7375]: src/adapter.c:property_set_mode() sending Set Powered command for index 0
bluetoothd[7375]: src/adapter.c:dev_class_changed_callback() Class: 0x6c010c
bluetoothd[7375]: src/adapter.c:property_set_mode_complete() Success (0x00)
bluetoothd[7375]: src/adapter.c:new_settings_callback() Settings: 0x00000ad1
bluetoothd[7375]: src/adapter.c:settings_changed() Changed settings: 0x00000001
bluetoothd[7375]: src/adapter.c:settings_changed() Pending settings: 0x00000000
bluetoothd[7375]: src/adapter.c:adapter_start() adapter /org/bluez/hci0 has been enabled
bluetoothd[7375]: src/adapter.c:trigger_passive_scanning() 
bluetoothd[7375]: src/adapter.c:set_mode() sending set mode command for index 0
bluetoothd[7375]: src/adapter.c:property_set_mode() sending Set Discoverable command for index 0
bluetoothd[7375]: src/adapter.c:new_settings_callback() Settings: 0x00000ad3
bluetoothd[7375]: src/adapter.c:settings_changed() Changed settings: 0x00000002
bluetoothd[7375]: src/adapter.c:settings_changed() Pending settings: 0x00000008
bluetoothd[7375]: src/adapter.c:property_set_mode_complete() Success (0x00)
bluetoothd[7375]: src/adapter.c:new_settings_callback() Settings: 0x00000adb
bluetoothd[7375]: src/adapter.c:settings_changed() Changed settings: 0x00000008
bluetoothd[7375]: src/adapter.c:settings_changed() Pending settings: 0x00000000

bluetoothd[7375]: src/adapter.c:connected_callback() hci0 device A0:5A:5C:A6:15:5D connected eir_len 38
bluetoothd[7375]: profiles/input/server.c:connect_event_cb() Incoming connection from A0:5A:5C:A6:15:5D on PSM 17
bluetoothd[7375]: profiles/input/device.c:input_device_set_channel() idev (nil) psm 17
bluetoothd[7375]: profiles/input/server.c:connect_event_cb() Refusing input device connect: No such file or directory (2)
bluetoothd[7375]: src/adapter.c:dev_disconnected() Device A0:5A:5C:A6:15:5D disconnected, reason 1
bluetoothd[7375]: src/adapter.c:adapter_remove_connection() 
bluetoothd[7375]: plugins/policy.c:disconnect_cb() reason 1
bluetoothd[7375]: src/adapter.c:bonding_attempt_complete() hci0 bdaddr A0:5A:5C:A6:15:5D type 0 status 0xe
bluetoothd[7375]: src/device.c:device_bonding_complete() bonding (nil) status 0x0e
bluetoothd[7375]: src/device.c:device_bonding_failed() status 14
bluetoothd[7375]: src/adapter.c:resume_discovery() 
bluetoothd[7375]: src/adapter.c:new_settings_callback() Settings: 0x00000ad3
bluetoothd[7375]: src/adapter.c:settings_changed() Changed settings: 0x00000008
bluetoothd[7375]: src/adapter.c:settings_changed() Pending settings: 0x00000000
```

Do I need something others to make it work ?
Or try sixad / xboxdrv ?

sjanc, 2021-08-03:
Do you have sixaxis plugin enabled?

Cooky-12, 2021-08-04:
> Do you have sixaxis plugin enabled?
```
pacman -Ql bluez-plugins
bluez-plugins /usr/
bluez-plugins /usr/lib/
bluez-plugins /usr/lib/bluetooth/
bluez-plugins /usr/lib/bluetooth/plugins/
bluez-plugins /usr/lib/bluetooth/plugins/sixaxis.so
```
I installed sixaxis, but don't know whether it enabled or not.
I see it loaded by " lsof | grep sixaxis.so ".
Is it need some configuration ?

sjanc, 2021-08-04:
You should see loaded plugins in bluetoothd debug log

also, if you use sixaxis plugin, you should not use sixpair

Cooky-12, 2021-08-05:
> You should see loaded plugins in bluetoothd debug log
> 
> also, if you use sixaxis plugin, you should not use sixpair
```

bluetoothd[3574]: profiles/input/suspend-none.c:suspend_init() 
bluetoothd[3574]: profiles/network/manager.c:read_config() Config options: Security=true
bluetoothd[3574]: plugins/hostname.c:read_dmi_fallback() chassis: laptop
bluetoothd[3574]: plugins/hostname.c:read_dmi_fallback() major: 0x01 minor: 0x03
bluetoothd[3574]: plugins/sixaxis.c:sixaxis_init() 
bluetoothd[3574]: src/main.c:main() Entering main loop
bluetoothd[3574]: src/rfkill.c:rfkill_event() RFKILL event idx 0 type 2 op 0 soft 0 hard 0
bluetoothd[3574]: Bluetooth management interface 1.20 initialized
bluetoothd[3574]: src/adapter.c:read_version_complete() sending read supported commands command
bluetoothd[3574]: src/adapter.c:read_version_complete() sending read index list command
```
This may means sixaxis loaded, but it didn't write address to PS3 controller.
I use sixpair to write address to PS3 controller.

Cooky-12, 2021-08-31:
I find a dirty solution. reported in 
https://github.com/bluez/bluez/issues/198