bluez/bluez #178: a2dp profile not available

sviande, 2021-07-22:
Hi,

I have a problem with my bluetooth setup, i can't select the a2dp profile for my headset.
I tried the patch from https://patches.linaro.org/patch/285268/ and it works.

You can find the logs of both case with and without patch:
Logs with bluez build from master from [arch package](https://aur.archlinux.org/cgit/aur.git/tree/PKGBUILD?h=bluez-git) 

```
./configure \
    --prefix=/usr \
    --mandir=/usr/share/man \
    --sysconfdir=/etc \
    --localstatedir=/var \
    --libexecdir=/usr/lib \
    --with-dbusconfdir=/usr/share \
    --enable-external-ell \
    --enable-btpclient \
    --enable-midi \
    --enable-sixaxis \
    --disable-mesh \
    --enable-hid2hci \
    --enable-experimental \
    --enable-manpages \
    --enable-library # this is deprecated
```

Logs from master without patch
```
bluetoothd[88622]: Bluetooth daemon 5.60
systemd[1]: Started Bluetooth service.
bluetoothd[88622]: Starting SDP server
bluetoothd[88622]: Bluetooth management interface 1.20 initialized
bluetoothd[88622]: Endpoint registered: sender=:1.17 path=/MediaEndpoint/A2DPSource/ldac
bluetoothd[88622]: Endpoint registered: sender=:1.17 path=/MediaEndpoint/A2DPSink/aptx_hd
bluetoothd[88622]: Endpoint registered: sender=:1.17 path=/MediaEndpoint/A2DPSource/aptx_hd
bluetoothd[88622]: Endpoint registered: sender=:1.17 path=/MediaEndpoint/A2DPSink/aptx
bluetoothd[88622]: Endpoint registered: sender=:1.17 path=/MediaEndpoint/A2DPSource/aptx
bluetoothd[88622]: Endpoint registered: sender=:1.17 path=/MediaEndpoint/A2DPSource/aac
bluetoothd[88622]: Endpoint registered: sender=:1.17 path=/MediaEndpoint/A2DPSink/sbc
bluetoothd[88622]: Endpoint registered: sender=:1.17 path=/MediaEndpoint/A2DPSource/sbc
bluetoothd[88622]: profiles/audio/avdtp.c:avdtp_parse_rej() SET_CONFIGURATION request rejected: Configuration not supported (41)
bluetoothd[88622]: profiles/audio/avdtp.c:avdtp_parse_rej() SET_CONFIGURATION request rejected: Configuration not supported (41)
```

Logs from master with the patch
```
bluetoothd[88622]: Bluetooth daemon 5.60
systemd[1]: Started Bluetooth service.
bluetoothd[3322]: Starting SDP server
bluetoothd[3322]: Bluetooth management interface 1.20 initialized
bluetoothd[3322]: Battery Provider Manager created
bluetoothd[3322]: src/advertising.c:read_adv_features_callback() Failed to read advertising features: Not Supported (0x0c)
bluetoothd[3322]: Adv Monitor Manager created with supported features:0x00000000, enabled features:0x00000000, max number of supported monitors:32, max number of supported patterns:16
bluetoothd[3322]: Endpoint registered: sender=:1.17 path=/MediaEndpoint/A2DPSource/ldac
bluetoothd[3322]: Endpoint registered: sender=:1.17 path=/MediaEndpoint/A2DPSink/aptx_hd
bluetoothd[3322]: Endpoint registered: sender=:1.17 path=/MediaEndpoint/A2DPSource/aptx_hd
bluetoothd[3322]: Endpoint registered: sender=:1.17 path=/MediaEndpoint/A2DPSink/aptx
bluetoothd[3322]: Endpoint registered: sender=:1.17 path=/MediaEndpoint/A2DPSource/aptx
bluetoothd[3322]: Endpoint registered: sender=:1.17 path=/MediaEndpoint/A2DPSource/aac
bluetoothd[3322]: Endpoint registered: sender=:1.17 path=/MediaEndpoint/A2DPSink/sbc
bluetoothd[3322]: Endpoint registered: sender=:1.17 path=/MediaEndpoint/A2DPSource/sbc
bluetoothd[3322]: Path / reserved for Adv Monitor app :1.31
bluetoothd[3322]: profiles/audio/avdtp.c:avdtp_parse_rej() SET_CONFIGURATION request rejected: Configuration not supported (41)
bluetoothd[3322]: Adv Monitor app :1.31 disconnected from D-Bus
bluetoothd[3322]: /org/bluez/hci0/dev_04_52_C7_33_E8_F7/fd0: fd(32) ready
bluetoothd[3322]: src/profile.c:ext_io_disconnected() Unable to get io data for Hands-Free Voice gateway: getpeername: Transport endpoint is not connected (107)
bluetoothd[3322]: /org/bluez/hci0/dev_04_52_C7_33_E8_F7/fd1: fd(30) ready
```

MarijnS95, 2021-07-23:
@sviande Would you mind to recreate the logs (bluetoothd, not btmon) with the daemon running in debug mode: `/usr/lib/bluetooth/bluetoothd -d`? That'll help us follow the decision-making logic, as supposedly the "working" configuration only runs `SET_CONFIGURATION` once and fails, but never tries it again. Is `btmonOK.txt` possibly truncated?

sviande, 2021-07-23:
@MarijnS95 please view attached logs, no audio playing in both cases.

[bluezLogOk.txt](https://github.com/bluez/bluez/files/6867989/bluezLogOk.txt)
[bluezLogNotOk.txt](https://github.com/bluez/bluez/files/6867990/bluezLogNotOk.txt)

MarijnS95, 2021-07-23:
It appears the peer changes the configuration with `SetConfiguration` later to AAC on its own regard, but not without the patch because the disconnect timer has already fired before it. I'll leave it up to @Vudentz to confirm that assessment and come up with a more appropriate patch that does not regress like mine linked above.

sviande, 2021-07-23:
It's happening on PipeWire.

Vudentz, 2021-07-26:
@sviande It looks like the headset is not compliant:

```
< ACL Data TX: Handle 256 flags 0x00 dlen 18              #237 [hci0] 14.015541
      Channel: 3203 len 14 [PSM 25 mode Basic (0x00)] {chan 2}
      AVDTP: Set Configuration (0x03) Command (0x00) type 0x00 label 1 nosp 0
        ACP SEID: 3
        INT SEID: 8
        Service Category: Media Transport (0x01)
        Service Category: Media Codec (0x07)
          Media Type: Audio (0x00)
          Media Codec: SBC (0x00)
            Frequency: 48000 (0x10)
            Channel Mode: Joint Stereo (0x01)
            Block Length: 16 (0x10)
            Subbands: 8 (0x04)
            Allocation Method: Loudness (0x01)
            Minimum Bitpool: 2
            Maximum Bitpool: 53
> ACL Data RX: Handle 256 flags 0x02 dlen 8               #247 [hci0] 14.047879
      Channel: 66 len 4 [PSM 25 mode Basic (0x00)] {chan 2}
      AVDTP: Set Configuration (0x03) Response Reject (0x03) type 0x00 label 1 nosp 0
        Service Category: Media Codec (0x07)
        Error code: UNSUPPORTED_CONFIGURATION (0x29)
```

That configuration is mandatory for SBC, except perhaps if there are multiple SBC endpoints and in case of 0x03 it doesn't support such setting, you can use avinfo to capture what are the supported settings for it, it could be that we have to retry or something since the SBC endpoints seems to be given in different order it might be that 0x01 is the SBC indented to cover the mandatory parameters. That said we do send the exact capabilities of the endpoint on the SelectCapabilities so I wonder if PipeWire is ignoring that and parsing the capabilities on the endpoint objects thinking that it must be for another endpoint.

MarijnS95, 2021-07-26:
Unfortunately the device replies with quite a few SEIDs which have all been cached, no `GetCapabilities` to confirm the underlying supported configuration. Having had problems with this previously on a WH-1000XM3 the solution was simply to forget and re-pair the device to get rid of whatever was cached and carried over from an older BlueZ and things magically started working - though I believe that has already been fixed too.

@Vudentz It is indeed quite possible that ACP SEID 1 and 3 support the same SBC configuration, but the device only "allows" to use one of the two SEPs for it. It'd be great to see a recursive dbus introspection of `/dev/bluez/hci0/dev_THE_MAC_ADDRESS`.

It's worth trying PulseAudio from `master`. I don't remember it trying multiple configurations if one fails, but it might sort them in a different order and use the appropriate one first.

Either way it's strange that the device rejects every `SetConfiguration` call thus far, and - with the patch that prevents disconnection  giving it enough time - the peer being the one that sends `SetConfiguration`. Unfortunately this is truncated from `btmonOK.txt`. Maybe this sink wants to be in charge of the configuration, even if the a2dp source sends something valid?

Vudentz, 2021-07-26:
avinfo shouldn't care about the cache though, the problem is not the cache itself since we do validate then at reconnection but the fact there are multiple endpoints for the same codec with different capabilities might be confusing as the daemon is not able to select what is best to use, it just pick the first in the list. Anyway if the device really wants to take over and wouldn't accept any configuration from our side that would indeed be problematic since we would never be able to disconnect waiting the remote side to take over which can potentially just sit idle consuming battery.

Btw, what model is this?

sviande, 2021-07-27:
The headset is a Bose QC35.
I've just created a new log with btmon with bluez patched [btmon.txt](https://github.com/bluez/bluez/files/6883437/btmon.txt)

sviande, 2021-07-27:
And here the result of avinfo:
```
Stream End-Point #3: Audio Sink
	Media Codec: MPEG24
		Object Types: MPEG-2 AAC LC MPEG-4 AAC LC
		Frequencies: 8kHz 11.025kHz 12kHz 16kHz 22.05kHz 24kHz 32kHz 44.1kHz 48kHz
		Channels: 1 2
		Bitrate: 320000
		VBR: Yes
	Content Protection: SCMS-T
Stream End-Point #1: Audio Sink
	Media Codec: SBC
		Channel Modes: Mono DualChannel Stereo JointStereo
		Frequencies: 16Khz 32Khz 44.1Khz 48Khz
		Subbands: 4 8
		Blocks: 4 8 12 16
		Bitpool Range: 2-53
	Content Protection: SCMS-T
Stream End-Point #51: Audio Source
	Media Codec: Vendor Specific A2DP Codec
		Vendor ID 0x0000000a
		Vendor Specific Codec ID 0x0104
		Vendor Specific Data: 0x07 0x08 0x00 0x02 0xc0 0xff 0x8c 0x84 0xe2 0x00
	Content Protection: SCMS-T
Stream End-Point #19: Audio Sink
	Media Codec: Vendor Specific A2DP Codec
		Vendor ID 0x0000000a
		Vendor Specific Codec ID 0x0104
		Vendor Specific Data: 0x07 0x08 0x00 0x02 0xc0 0xff 0x8c 0x84 0xe2 0x00
	Content Protection: SCMS-T
Stream End-Point #49: Audio Source
	Media Codec: Vendor Specific A2DP Codec
		Vendor ID 0x0000000a
		Vendor Specific Codec ID 0x0103
		Vendor Specific Data: 0x07 0x06 0x00 0x00 0xff 0xff 0x02 0x35
	Content Protection: SCMS-T
Stream End-Point #17: Audio Sink
	Media Codec: Vendor Specific A2DP Codec
		Vendor ID 0x0000000a
		Vendor Specific Codec ID 0x0103
		Vendor Specific Data: 0x07 0x06 0x00 0x00 0xff 0xff 0x02 0x35
	Content Protection: SCMS-T
Stream End-Point #33: Audio Source
	Media Codec: SBC
		Channel Modes: Mono DualChannel Stereo JointStereo
		Frequencies: 16Khz 32Khz 44.1Khz 48Khz
		Subbands: 4 8
		Blocks: 4 8 12 16
		Bitpool Range: 2-53
	Content Protection: SCMS-T
```

MarijnS95, 2021-07-27:
@sviande That `dbus-send` call returns introspection data (XML) detailing the various interfaces and properties that exist, now you need a tool that retrieves and prints those interfaces and properties. I personally use something like `gdbus introspect -y -d org.bluez -o /org/bluez/hci0 -r` to recursively grab introspection data, read all the interfaces and properties, and print them.

However the output from `avinfo` shows us enough. Btmon traces clearly show PipeWire trying to set up an SBC configuration on Stream End-Point 3, which is actually an AAC endpoint according to `avinfo`. Seems like a PipeWire bug, please try PulseAudio to confirm.
(DBus introspection is still really interesting in the rare case that BlueZ is exposing the wrong codec on the bus, though that seems really unlikely)

sviande, 2021-07-27:
thanks for the info: 
the result of `gdbus introspect -y -d org.bluez -o /org/bluez/hci0 -r`
```
node /org/bluez/hci0 {
  interface org.freedesktop.DBus.Introspectable {
    methods:
      Introspect(out s xml);
    signals:
    properties:
  };
  interface org.bluez.Adapter1 {
    methods:
      StartDiscovery();
      SetDiscoveryFilter(in  a{sv} properties);
      StopDiscovery();
      RemoveDevice(in  o device);
      GetDiscoveryFilters(out as filters);
      ConnectDevice(in  a{sv} properties);
    signals:
    properties:
      readonly s Address = 'E0:D4:E8:19:BA:17';
      readonly s AddressType = 'public';
      readonly s Name = 'Tux';
      readwrite s Alias = 'Tux';
      readonly u Class = 7078156;
      readwrite b Powered = true;
      readwrite b Discoverable = false;
      readwrite u DiscoverableTimeout = 180;
      readwrite b Pairable = false;
      readwrite u PairableTimeout = 0;
      readonly b Discovering = false;
      readonly as UUIDs = ['0000110e-0000-1000-8000-00805f9b34fb', '0000111f-0000-1000-8000-00805f9b34fb', '00001200-0000-1000-8000-00805f9b34fb', '0000110b-0000-1000-8000-00805f9b34fb', '00001108-0000-1000-8000-00805f9b34fb', '0000110c-0000-1000-8000-00805f9b34fb', '00001800-0000-1000-8000-00805f9b34fb', '0000110a-0000-1000-8000-00805f9b34fb', '00001801-0000-1000-8000-00805f9b34fb', '0000180a-0000-1000-8000-00805f9b34fb'];
      readonly s Modalias = 'usb:v1D6Bp0246d053C';
      readonly as Roles = ['central'];
  };
  interface org.freedesktop.DBus.Properties {
    methods:
      Get(in  s interface,
          in  s name,
          out v value);
      Set(in  s interface,
          in  s name,
          in  v value);
      GetAll(in  s interface,
             out a{sv} properties);
    signals:
      PropertiesChanged(s interface,
                        a{sv} changed_properties,
                        as invalidated_properties);
    properties:
  };
  interface org.bluez.GattManager1 {
    methods:
      RegisterApplication(in  o application,
                          in  a{sv} options);
      UnregisterApplication(in  o application);
    signals:
    properties:
  };
  interface org.bluez.LEAdvertisingManager1 {
    methods:
      RegisterAdvertisement(in  o advertisement,
                            in  a{sv} options);
      UnregisterAdvertisement(in  o service);
    signals:
    properties:
      readonly y ActiveInstances = 0x00;
      readonly y SupportedInstances = 0x00;
      readonly as SupportedIncludes = ['local-name'];
      readonly as SupportedSecondaryChannels;
[... 450 lines cut ...]
          readonly y Codec = 0x00;
          readonly ay Capabilities = [0xff, 0xff, 0x02, 0x35];
          readonly o Device = '/org/bluez/hci0/dev_04_52_C7_33_E8_F7';
          readonly b DelayReporting = false;
      };
      interface org.freedesktop.DBus.Properties {
        methods:
          Get(in  s interface,
              in  s name,
              out v value);
          Set(in  s interface,
              in  s name,
              in  v value);
          GetAll(in  s interface,
                 out a{sv} properties);
        signals:
          PropertiesChanged(s interface,
                            a{sv} changed_properties,
                            as invalidated_properties);
        properties:
      };
    };
  };
};
```

MarijnS95, 2021-07-27:
@sviande Looks like BlueZ is going bonkers, not PW. We clearly have `Media Codec: MPEG24` from `avinfo` on SEP 3 which corresponds to `#define A2DP_CODEC_MPEG24		0x02`, but according to DBus `sep3` is on:

```c
    node /org/bluez/hci0/dev_04_52_C7_33_E8_F7/sep3 {
      ...
      interface org.bluez.MediaEndpoint1 {
        ...
        properties:
          readonly s UUID = '0000110b-0000-1000-8000-00805f9b34fb';
          readonly y Codec = 0x00;
          readonly ay Capabilities = [0xff, 0xff, 0x02, 0x35];
          readonly o Device = '/org/bluez/hci0/dev_04_52_C7_33_E8_F7';
          readonly b DelayReporting = false;
      };
```

`Codec = 0x00` here corresponds to SBC.

MarijnS95, 2021-07-27:
It's quite surprising too that both the codec and capabilities buffer are identical for each of the 7 SEPs. Maybe the DBus iterator is repeating old data?

Vudentz, 2021-07-27:
That is weird, maybe the cache is corrupted, try removing the cache file:

/var/lib/blueooth/<adapter bdaddr>/<device bdaddr>

sviande, 2021-07-28:
Removing the cache /var/lib/bluetooth/ seems to work.

Here you can find `avinfo` result 
```
Stream End-Point #3: Audio Sink
	Media Codec: MPEG24
		Object Types: MPEG-2 AAC LC MPEG-4 AAC LC
		Frequencies: 8kHz 11.025kHz 12kHz 16kHz 22.05kHz 24kHz 32kHz 44.1kHz 48kHz
		Channels: 1 2
		Bitrate: 320000
		VBR: Yes
	Content Protection: SCMS-T
Stream End-Point #1: Audio Sink
	Media Codec: SBC
		Channel Modes: Mono DualChannel Stereo JointStereo
		Frequencies: 16Khz 32Khz 44.1Khz 48Khz
		Subbands: 4 8
		Blocks: 4 8 12 16
		Bitpool Range: 2-53
	Content Protection: SCMS-T
Stream End-Point #51: Audio Source
	Media Codec: Vendor Specific A2DP Codec
		Vendor ID 0x0000000a
		Vendor Specific Codec ID 0x0104
		Vendor Specific Data: 0x07 0x08 0x00 0x02 0xc0 0xff 0x8c 0x84 0xe2 0x00
	Content Protection: SCMS-T
Stream End-Point #19: Audio Sink
	Media Codec: Vendor Specific A2DP Codec
		Vendor ID 0x0000000a
		Vendor Specific Codec ID 0x0104
		Vendor Specific Data: 0x07 0x08 0x00 0x02 0xc0 0xff 0x8c 0x84 0xe2 0x00
	Content Protection: SCMS-T
Stream End-Point #49: Audio Source
	Media Codec: Vendor Specific A2DP Codec
		Vendor ID 0x0000000a
		Vendor Specific Codec ID 0x0103
		Vendor Specific Data: 0x07 0x06 0x00 0x00 0xff 0xff 0x02 0x35
	Content Protection: SCMS-T
Stream End-Point #17: Audio Sink
	Media Codec: Vendor Specific A2DP Codec
		Vendor ID 0x0000000a
		Vendor Specific Codec ID 0x0103
		Vendor Specific Data: 0x07 0x06 0x00 0x00 0xff 0xff 0x02 0x35
	Content Protection: SCMS-T
Stream End-Point #33: Audio Source
	Media Codec: SBC
		Channel Modes: Mono DualChannel Stereo JointStereo
		Frequencies: 16Khz 32Khz 44.1Khz 48Khz
		Subbands: 4 8
		Blocks: 4 8 12 16
		Bitpool Range: 2-53
	Content Protection: SCMS-T
```

and `dbus instrospect result`
```
node /org/bluez/hci0 {
  interface org.freedesktop.DBus.Introspectable {
    methods:
      Introspect(out s xml);
    signals:
    properties:
  };
  interface org.bluez.Adapter1 {
    methods:
      StartDiscovery();
      SetDiscoveryFilter(in  a{sv} properties);
      StopDiscovery();
      RemoveDevice(in  o device);
      GetDiscoveryFilters(out as filters);
[... 506 lines cut ...]
          readonly b DelayReporting = true;
      };
      interface org.freedesktop.DBus.Properties {
        methods:
          Get(in  s interface,
              in  s name,
              out v value);
          Set(in  s interface,
              in  s name,
              in  v value);
          GetAll(in  s interface,
                 out a{sv} properties);
        signals:
          PropertiesChanged(s interface,
                            a{sv} changed_properties,
                            as invalidated_properties);
        properties:
      };
    };
  };
};
```

Thanks @MarijnS95 and @Vudentz for your help and support  👏
I'll be happy to help for creating/testing a patch when the cache seems corrupt.

MarijnS95, 2021-07-28:
@sviande Thanks for confirming! The original patch was merely an impartial solution to another problem that helped as a band-aid here, glad it is resolved properly.

Did you hold on to the broken cache file so that we can have a look inside? It's probably obvious what we'll find, but interesting for future patch testing nevertheless.

sviande, 2021-07-28:
Sorry i remove the cache and think to move it when it was removed :/

closed 2021-08-04 by BluezTestBot