bluez/bluez #179: [5.60] Cannot Connect consistently with BLE Device - Connection Refused

alfredoant3, 2021-07-23:
Hello,

I am using bluez 5.60 on my Linux computer to connect with a BLE remote device. When the remote device is ready to connect, it produces two advertising reports: one connectable advertisement and one ibeacon broadcast (non-connectable).

The following is the sequence I use to connect with the device:

- SetDiscoveryFilter to filter for the the service UUID.
- StartDiscovery to discover my device.
- Once my device is discover and I verify it has the correct service UUID (from the connectable advertisement), I call  StopDiscovery and then call the Connect method on the device object.

`res = g_dbus_proxy_method_call(proxy, "Connect", NULL, proxy_connect_reply, proxy, NULL);`

In a wireshark capture, I see that a successful Connect method produces the following sequence:

- LE Set Extended Scan Parameters
- LE Set Extended Scan Enable (true) (Passive filter policy)
- During the scanning it finds a LE Meta Advertising Report (connectable advertisement)
- LE Set Extended Scan Enable (false)
- LE  Extended Create Connection

When this fails, the following is the sequence of events captured in wireshark:

- LE Set Extended Scan Parameters
- LE Set Extended Scan Enable (true) (Passive filter policy)
- During the scanning it finds a LE Meta which is an iBeacon (Non-connectable, Legacy)
- 40 seconds pass (which I assume is some connection attempt process occurring)
- LE Set Extended Scan Enable (false) (possibly from a time out)
- And then it never performs the LE Create Connection

From the Dbus monitor this error comes up:

```
error time=1625773548.249392 sender=:1.3 -> destination=:1.137 error_name=org.bluez.Error.Failed reply_serial=191
   string "Software caused connection abort"
```

From bluetoothd debug syslog, this is the error I see:

```
src/device.c:device_connect_le() Connection attempt to: XX:XX:XX:XX:XX:XX
src/adapter.c:stop_discovery_complete() status 0x00
src/adapter.c:discovery_remove() owner :1.692
src/device.c:device_set_rssi_with_delta() rssi 0
src/adapter.c:trigger_passive_scanning()
src/adapter.c:discovering_callback() hci0 type 6 discovering 0 method 0
src/device.c:att_connect_cb() connect to XX:XX:XX:XX:XX:XX: Connection refused (111)
```

My suspicion is that since my remote device is producing a connectable advertisement and an ibeacon, the bluez stack is attempting to connect using the non-connectable advertisement instead of the connectable one.

I tried to examine the device.c and adapter.c apis but I could not figure out where is the connection performed. I got to the l2cap_connect function call at which point I am unsure where this goes. I was curious of how the filter (service UUID) I setup using SetDiscoveryFilter could be carried over into the connection process. I am only getting advertisements from my target device, but really I would only care about the connectable advertisement, not the ibeacon broadcast. I also read something about kernel passive scanning, is this possibly something I need to configure in some kernel configuration?

Is this a linux/bluez limitation? or how can I filter through the advertisements even during connection?

Thank you

Vudentz, 2021-07-23:
Ok, so you have a device that is advertising as connetable and non-connectable at same time using the same address? That perhaps is the problem since if we couldn't find a connectable advertisement it won't attempt to connect, so are you sure it does continue to advertise as connectable after that or it is rotating the advertising periods of connectable and non-connectable? If that is over 40 seconds to rotate that would explain why we cannot connect, this is why non-connectable beacons are recommended to use a different random address since the central has no idea that it could be doing multiple advertising at same time.

alfredoant3, 2021-07-27:
The device is indeed rotating between connectable and non-connectable within 10-20 seconds. So in theory, the connect should be able to see the connectable adv after the beacon, before the scan times out. But what I have noticed is every time the connect scan receives the beacon, it will not receive anything else during the 40 seconds of scan (as if it was stuck in some operation).

This is why I would like to know if there is a way to configure the connect method to filter by the specific service data UUID? This way the beacon could be ignored, and we would only connect against the true connectable adv. Especially if assigning different address to the adv and beacon is not feasible.

Vudentz, 2021-07-28:
@alfredoant3 Have you tried changing this on main.conf:

https://git.kernel.org/pub/scm/bluetooth/bluez.git/tree/src/main.conf#n155

alfredoant3, 2021-07-29:
I see that this has 
# LE scanning parameters used for connection establishment.
#ScanIntervalConnect=
#ScanWindowConnect=

I tried modifying those but did not see any impact on my wireshark capture. The LE Set Scan Parameters command still uses the same values. Is it as simple as removing '#' and adding the values and re-build bluez?

Ultimately I do not think my issue is related to interval or window, and more about filtering the connectable adv. In the Core Spec I see that during a  directed connection establishment, " The Host shall set the initiator filter policy to ‘ignore the Filter Accept List and
process connectable advertising packets from a specific single device specified by the Host’." page 1323

Isn't there a way for the connection process to ignore the beacon which is non-connectable, and only accept connectable advertising packets as the Core Spec says?

Vudentz, 2021-08-06:
@alfredoant3 Afaik we already do that, we program the device on the white list:

https://git.kernel.org/pub/scm/linux/kernel/git/bluetooth/bluetooth-next.git/commit/?id=9a9373ffc7338377c3837ce0ccd40f5c4402c9d1

You can see the filter is using the white list already, but your controller needs to support it as well, also make sure your kernel does have this change since it affects the window.

closed 2021-10-04 by alfredoant3