bluez/bluez #181: BLE device no more discoverable after pairing DBUS error

Vishal8523, 2021-08-06:
While pairing to a BLE device with the PIN, the below DBUS error is thrown:

`DBusError: Did not receive a reply. Possible causes include: the remote application did not send a reply, the message bus security policy blocked the reply, the reply timeout expired, or the network connection was broken.`

After this error, the device is no more discoverable until the system is rebooted.
Where as the BLE device is discoverable in other applications like nRF connect.

BlueZ version: 5.54

Appreciate help in this regards.

Vudentz, 2021-08-06:
@Vishal8523 has this been tested with 5.60? We don't actually support versions that old, also it is possible that the daemon is crashing so yet another reason to update.

Vishal8523, 2021-08-09:
@Vudentz : Ok sure. Let me try with Bluez 5.60+

Vishal8523, 2021-08-09:
The daemon is running and not killed. By the way, other devices can be paired / connected. Only the device which got the above message cannot be seen in the BLE scan.

Vishal8523, 2021-08-09:
It may be due to - "the message bus security policy blocked the reply"

Vishal8523, 2021-08-10:
Tested with Bluez 5.60 as well. The issue persists.

Vishal8523, 2022-11-22:
Any update on this issue ?

Vudentz, 2022-11-22:
> The daemon is running and not killed. By the way, other devices can be paired / connected. Only the device which got the above message cannot be seen in the BLE scan.
> 
> It may be due to - "the message bus security policy blocked the reply"

Perhaps there is something blocking the message, or lack of D-Bus rule for the agent, perhaps you should report to your distro this is happening, have your tried pairing with the likes of bluetoothctl?