bluez/bluez #183: Proper way to  create Custom LE Profile using LE COC over Bluez DBUS API

RoyAnupam, 2021-08-06:
Hi,
 I intend to create & use  *Custom LE profiles (Client& server both)* on top of LE Connection oriented channel (LE CoC)
What would be the proper way to register & use such type of profile as Bluez DBUS API user?

Can following API's be used for this purpose?
-  void RegisterProfile(object profile, string uuid, dict options)  [https://github.com/bluez/bluez/blob/master/doc/profile-api.txt](url)
-  void ConnectProfile(string uuid)   [https://github.com/bluez/bluez/blob/master/doc/device-api.txt](url)

Also, is there any example in [https://github.com/bluez/bluez](url)  which can be referred?

Vudentz, 2021-08-06:
@RoyAnupam LE will need a GATT service to expose the PSM you would want to use, so you will need to register that instead of Profile, the Profile API is more of a BR/EDR thing since that registers a SDP records as opposed to LE which uses GATT Services. We could however do allow GATT Services to be registered as profiles if the intend is to have bluetoothd authorize connections to the PSM, but that will need some updating in order to properly distinguish that is for LE not for BR/EDR.

RoyAnupam, 2021-08-07:
Hello @Vudentz , 
> LE will need a GATT service to expose the PSM you would want to use, so you will need to register that instead of Profile,

Yes, I would want to register custom PSM for the LE profile. However, I am wondering, how to expose the custom PSM to the  daemon via GATT DBUS API? Could you please guide.

As per doc/gatt-api.txt,  daemon accepts ***UUID(s) for service/char/desc (GATT Server Role)*** & GattProfile1 Interface needs ***Remote GATT Service UUID's (GATT Client Role) for auto connect***. There seems to be no provision for PSM, similar to BR/EDR RegisterProfile. Please help me understand, if I am missing something. Thank You~

RoyAnupam, 2021-08-13:
@Vudentz ,
 I got it now. Thank you for sharing your opinion~
 I am curious to know,  you have plan to provide support for it ?

wlynxg, 2023-09-25:
Is this feature implemented now?

Vudentz, 2023-09-25:
@wlynxg is this a feature request? It might need to be marked as such, also currently Im quite busy with LE Audio activities so I would appreciate if someone else takes over implementing the suggested changes.

closed 2025-08-26 by github-actions[bot]